realsenseai/librealsense #13363: Efficient conversion of point cloud model into finite element accurate mesh model

zhanglepy, 2024-09-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello,
By using a depth camera to obtain the surface point cloud model of an object (in ply format), and then stitching it together from multiple perspectives to form a complete point cloud model, what processing is needed to efficiently convert it into a finite element accurate mesh model suitable for SOFA deformation simulation. I previously tried using meshlab software to convert point clouds into models, then using FreeCAD to convert the models into solids, and finally using Gmsh to partition 2D and 3D meshes. However, the grid model established in this way is relatively rough, with a low success rate and a complex process.
I look forward to your help!

MartyG-RealSense, 2024-09-20:
Hi @zhanglepy  If you are stitching together .ply files then it may be worth next trying the free open-source **CloudCompare** tool - https://www.danielgm.net/cc/ - for merging them, as advised at https://github.com/IntelRealSense/librealsense/issues/12611#issuecomment-1911880790

If you only need to convert the point cloud to mesh then CloudCompare also has support for **poisson reconstruction** to perform the conversion.

zhanglepy, 2024-09-25:
Thanks for your reply！

closed 2024-10-10 by MartyG-RealSense