realsenseai/librealsense #13364: FW UPDATE failed

super-fsy, 2024-09-20:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2024-09-20:
Hi @super-fsy  This error usually happens when trying to play back a .bag file recording of camera data in the RealSense Viewer.  Are you attempting .bag file playback, please?

super-fsy, 2024-09-20:
The situation I encountered when I first connected the Realsense camera during the firmware upgrade, on a Windows platform, with the D455i camera.
[image]
Image (not the original, same issue).

super-fsy, 2024-09-20:
> The situation I encountered when I first connected the Realsense camera during the firmware upgrade, on a Windows platform, with the D455i camera. [...]

closed 2024-09-20 by super-fsy