realsenseai/librealsense #13365: "Frames didn’t arrive within 5 seconds" – Intermittent Issue with RealSense Camera on Linux

nurietarc, 2024-09-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | [D455](https://www.intelrealsense.com/depth-camera-d455/) | 
| Firmware Version                   | 2.55.1 | 
| Operating System & Version |   Linux (Ubuntu 22.04.5)  | 
| Kernel Version (Linux Only)    |  6.5.0-44-generic / 6.8.-44  (see explanation)                                         | 
| Platform                                 | PC (Thinkpad L390) |
| SDK Version                            |  Intel® RealSense™ SDK 2.0                           | 
| Language                            |  ROS2 Humble                          | 
| Segment			|  Robotics                         | 

## Context
I am using the Intel RealSense camera for robotics applications and trying to implement the ROS Wrapper for Intel (link [here](https://github.com/IntelRealSense/realsense-ros)) but having issues getting it to work.

## Steps followed and errors (and how to reproduce them)
### Attempt 1:
- Installed the SDK 2.0 library using the second option (Option 2: Install librealsense2 without graphical tools and examples from ROS servers for Foxy EOL distro), which does not include the RealSense Viewer.
- Followed Step 3 to install the Intel® RealSense™ ROS2 wrapper from source (Option 2: Install from source).
- Launched the node using ros2 launch realsense2_camera rs_launch.py and encountered the following issue:
```
[INFO] [launch]: All log files can be found below /home/convidat-cdei/.ros/log/2024-09-20-16-24-46-496575-PERAFITA-63105
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [63106]
[realsense2_camera_node-1] [INFO] [1726842286.902101578] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1726842286.902349126] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1726842286.902399445] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1]  20/09 16:24:46,904 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  20/09 16:24:46,905 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  20/09 16:24:46,905 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.9.auto
[realsense2_camera_node-1]  20/09 16:24:46,905 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
[realsense2_camera_node-1]  20/09 16:24:46,906 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.7.auto
[realsense2_camera_node-1]  20/09 16:24:46,906 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.13.auto
[realsense2_camera_node-1]  20/09 16:24:46,906 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.6.auto
[realsense2_camera_node-1]  20/09 16:24:46,906 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.12.auto
[realsense2_camera_node-1]  20/09 16:24:46,907 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1]  20/09 16:24:46,907 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.11.auto
[realsense2_camera_node-1]  20/09 16:24:46,943 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto
[realsense2_camera_node-1]  20/09 16:24:46,943 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
[realsense2_camera_node-1]  20/09 16:24:46,943 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.9.auto
[realsense2_camera_node-1]  20/09 16:24:46,943 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
[realsense2_camera_node-1]  20/09 16:24:46,943 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.7.auto
[realsense2_camera_node-1]  20/09 16:24:46,944 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.13.auto
[realsense2_camera_node-1]  20/09 16:24:46,944 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.6.auto
[realsense2_camera_node-1]  20/09 16:24:46,944 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.12.auto
[realsense2_camera_node-1]  20/09 16:24:46,944 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
[realsense2_camera_node-1]  20/09 16:24:46,944 WARNING [128721463395904] (backend-hid.cpp:1280) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.11.auto
[realsense2_camera_node-1] [INFO] [1726842286.945833372] [camera.camera]: Device with serial number 234322305657 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1726842286.945906876] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2 was found.
[realsense2_camera_node-1] [INFO] [1726842286.945932934] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1726842286.946288765] [camera.camera]: Device with port number 1-3 was found.
[realsense2_camera_node-1] [INFO] [1726842286.946323495] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1726842286.946344467] [camera.camera]: Device 234322305657 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1726842286.946440699] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1726842286.946865400] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1726842286.946899268] [camera.camera]: Device Name: Intel RealSense D455
[realsense2_camera_node-1] [INFO] [1726842286.946922714] [camera.camera]: Device Serial No: 234322305657
[realsense2_camera_node-1] [INFO] [1726842286.946943827] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2
[realsense2_camera_node-1] [INFO] [1726842286.946964415] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1726842286.946982611] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1726842286.947001497] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1726842287.084208180] [camera.camera]: Set ROS param depth_module.depth_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1726842287.086078637] [camera.camera]: Set ROS param depth_module.infra_profile to default: 640x480x15
[... cut ...]
* Tried launching the node with the` initial_reset` parameter.

## The problem
Even though the camera works on some machines, we cannot pinpoint the root cause of the issue or why the error occurs intermittently between different computers, and even on the same machines, where it may "magically" stop working after a few days without any apparent reason. This makes the camera unreliable for robotics applications, as we never know when the next unexpected error might appear. Any suggestions?

Many thanks!

nurietarc, 2024-09-25:
Hi @MartyG-RealSense 

Thank you for your help. First, these are not long-run sessions. We are still testing in the lab, and the tests last about 10 minutes to ensure everything works well. The camera shouldn’t be overheating or anything like that.

Regarding the error you mentioned, I removed any power-saving configurations on the laptop. I upgraded the kernel to 6.8 again (as mentioned, downgrading to 6.5 didn’t allow me to install the RealSense library, and since the library works fine on other machines with kernel 6.8, I gave it another try). I purged the previous installation and reinstalled the library using Option 1.

Upon opening the RealSense Viewer, I got the following error:
```
1: /etc/udev/rules.d/99-realsense-libusb.rules  
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules  
Make sure to remove redundancies!  
```

Additionally, I got this warning:
```
09:55:27 [Warn] /home/convidat-cdei/librealsense/src/linux/backend-hid.cpp:1280 - Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.4.auto`
```

Despite these errors and warning and a "frames didn't arrive" error at first, I was able to connect and view the camera in RealSense Viewer after some fiddling around, even with the errors still showing.

Then I proceed to removed the redundancies as indicated by the error (although I’m not entirely sure if I did this step correctly, I may need some guidance...), and just to be sure, I purged and reinstalled again. The redundancy and busnum/devnum errors persist and now I can't vew the camera RGB signal in RealSense Viewer anymore. Below is the complete log:
```
10:02:39 [Info]  - librealsense version: 2.55.1

10:02:39 [Warn] /home/convidat-cdei/librealsense/common/notifications.cpp:511 - Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!
10:02:39 [Info] /home/convidat-cdei/librealsense/src/rs.cpp:3103 - Framebuffer size changed to 1663 x 862
10:02:39 [Info] /home/convidat-cdei/librealsense/src/rs.cpp:3103 - Window size changed to 1663 x 862
10:02:39 [Info] /home/convidat-cdei/librealsense/src/rs.cpp:3103 - Scale Factor is now 1
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video0
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video1
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video2
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video3
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video4
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video5
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.3/video4linux/video6
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.3/video4linux/video7
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video0
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video1
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video2
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video3
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video4
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/video4linux/video5
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.3/video4linux/video6
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.3/video4linux/video7
10:02:39 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video0
[... cut ...]
10:02:39 [Warn] /home/convidat-cdei/librealsense/src/linux/backend-hid.cpp:1280 - Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.7.auto
10:02:39 [Warn] /home/convidat-cdei/librealsense/src/linux/backend-hid.cpp:1280 - Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.13.auto
10:02:39 [Warn] /home/convidat-cdei/librealsense/src/linux/backend-hid.cpp:1280 - Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.6.auto
```

Thanks again!

MartyG-RealSense, 2024-09-25:
You are very welcome!

Multiple udev rules (/lib and /etc) occur when librealsense has been installed from packages and from source code on the same computer.  One method or the other should be used, otherwise you could have two versions of librealsense on the same computer that conflict with each other.

The rule that should always be removed is `/etc/udev/rules.d/99-realsense-libusb.rules`, as removing the /lib rule breaks linkages to the packages.

As the above log shows that the multiple udev rules error is still present, please try deleting the /etc rule to see whether any of the errors that follow afterwards disappear.  The udev rule is responsible for handling RealSense camera devices in Ubuntu.

nurietarc, 2024-09-27:
Hello,

I have removed the rule doing `sudo rm /etc/udev/rules.d/99-realsense-libusb.rules` and rebooting the system. The redundancies error isn't showing now in the Realsense viewer. Hoewever, i still get the "Frames didn't arrive". Complete log of the Realsense viewer:

```
11:02:13 [Info]  - librealsense version: 2.55.1

11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video0
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video1
11:02:13 [Info] /home/convidat-cdei/librealsense/src/rs.cpp:3103 - Framebuffer size changed to 1663 x 862
11:02:13 [Info] /home/convidat-cdei/librealsense/src/rs.cpp:3103 - Window size changed to 1663 x 862
11:02:13 [Info] /home/convidat-cdei/librealsense/src/rs.cpp:3103 - Scale Factor is now 1
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video3
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video4
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video5
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video6
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video7
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video0
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video1
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video3
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video4
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video5
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video6
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video7
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video0
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video1
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video3
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video4
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video5
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video6
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video7
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
11:02:13 [Info] /home/convidat-cdei/librealsense/src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
[... cut ...]
11:03:42 [Info] /home/convidat-cdei/librealsense/src/core/notification.cpp:22 - Frames didn't arrived within 5 seconds
11:03:47 [Warn] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:1649 - Frames didn't arrived within 5 seconds
11:03:47 [Info] /home/convidat-cdei/librealsense/src/core/notification.cpp:22 - Frames didn't arrived within 5 seconds
11:03:52 [Info]  - Stopping streaming
11:03:52 [Info] /home/convidat-cdei/librealsense/src/linux/backend-v4l2.cpp:1427 - V4L stream is closed
```
Thank you!

nurietarc, 2024-09-27:
Hi Marty,

I am aware of the `Device 234322305657 is connected using a 2.1 port. Reduced performance is expected` error. I get this error as well in other computers (the ones listed at first comment of the issue) in which the camera works properly. That is a problem that I had in mind to attack whenever I solve the one of this issue (Frames didn't arrive error in some computers), since the cable is supposed to be high-quality and the USB ports of the computer should be 3.1 as well.  In any case, I am not using any USB hub.

Since you have said that the laptop itself and its connectors could be the problem behind, I have tried to open the RealSense viewer in another laptop which has exactly the same specifications (ThinkPad L390). I get the Frames didn't arrive error but no the busnum/devnum. Here is the log:

```
14:43:42 [Info]  - librealsense version: 2.55.1

14:43:42 [Info] ./src/rs.cpp:3103 - Framebuffer size changed to 1344 x 756
14:43:42 [Info] ./src/rs.cpp:3103 - Scale Factor is now 1
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video0
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video1
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video3
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video4
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video5
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video6
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video7
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video0
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video1
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video3
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video4
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video5
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video6
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video7
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video0
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video1
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video3
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video4
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video5
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video6 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video6
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video7 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.3/video4linux/video7
14:43:42 [Warn] ./common/device-model.cpp:442 - Exception caught trying to detect presets: could not open directory: /home/convidat-cdei/Documents/librealsense2/presets/
14:43:42 [Info]  - Intel RealSense D455 was selected as a default device
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video0
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/video4linux/video1
14:43:42 [Info] ./src/linux/backend-v4l2.cpp:753 - Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/video4linux/video2
[... cut ...]
14:44:08 [Warn] ./src/linux/backend-v4l2.cpp:1649 - Frames didn't arrived within 5 seconds
14:44:08 [Info] ./src/core/notification.cpp:22 - Frames didn't arrived within 5 seconds
14:44:12 [Info]  - Stopping streaming
14:44:12 [Info] ./src/linux/backend-v4l2.cpp:1427 - V4L stream is closed
14:44:28 [Info] ./src/rs.cpp:3103 - Framebuffer size changed to 1850 x 1016
14:44:28 [Info] ./src/rs.cpp:3103 - Window size changed to 1850 x 1016
```
I am even more lost than before... haha Thank yoy again very much!

MartyG-RealSense, 2024-09-27:
It is difficult to mitigate against random events.  You might be able to reduce the likelihood of them occurring though if you build the librealsense SDK from source code with the **libuvc backend** or the **RSUSB backend** installation methods though (if you are not using them already).  A build of librealsense created with these methods bypasses the Linux kernel and so can avoid unpredictable errors caused by conflicts between librealsense and the kernel.

nurietarc, 2024-09-30:
I'll try to build the librealsense with RSUSB backend once again, thank you. However, would this solve the "frames didn't arrive" error? Isn't it a power-related error?

Thank you!

MartyG-RealSense, 2024-09-30:
It would not typically be a power-related error.  It could occur because of causes such as:

- The computer's CPU processor chip not being able to keep up with the rate that new frames are arriving from the camera.  Reducing FPs speed could test whether this is the cause;
 
- The frame pipeline becoming "jammed up" with frames because old frames are not being released fast enough and so new frames cannot enter the pipeline.  The pipeline, which is like a continuously moving conveyor belt where new frames enter and old frames drop off the end, can only hold 16 frames at a time from each stream type by default;

MartyG-RealSense, 2024-10-20:
Hi @nurietarc  Do you require further assistance with this case, please?  Thanks!

nurietarc, 2024-10-21:
Hello! Thank you for following up. 

It is working on the robot machine, which is our main concern, but it’s still not working on other laptops. This could potentially be an issue for others, and I still believe these errors seem too random to ignore. It may be okay to mark this as resolved for now, but it would be important to monitor and address these errors properly to avoid future issues.

Thanks!

MartyG-RealSense, 2024-10-21:
Thanks very much for the update.  I'm pleased to hear that it is working on your robot machine!

Whilst the RealSense SDK is continuously improved with each new release, RealSense cameras are used with such a wide range of computers / computing devices and software / hardware configurations that issues may occur that require individual support on this GitHub forum.

closed 2024-10-28 by MartyG-RealSense