realsenseai/librealsense #13370: depth stream start failure /Left MIPI/ REC error

Artii24, 2024-09-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 05.13.00.55/ 05.16.01| 
| Operating System & Version |   Win 11 / Linux Ubuntu 20.04.6 LTS  | 
| Kernel Version (Linux Only)    |  5.15.0-122-generic                                       | 
| Platform                                 | PC/ NVIDIA Jetson  |
| SDK Version                            |  2.53/ 2.56.1                          | 
| Language                            |  C++                         | 
| Segment			|  Robot                         | 

### Issue Description
Hello, @MartyG-RealSense
I'll tell the story first, and at the end, what I've already done and what I can't do.

Together with the robot from the manufacturer came a D435i camera and a Jetson Orin Nano with a pre-installed algorithm. The camera did not work at once and my colleagues connected it to RealSenseViewer on a Linux computer. The camera produced a few depth frames, after which the Left MIPI and REC errors appeared. The camera lay for a month and then I took it home (where the climate is a little different) and ran RealSenseViewer on Windows and Linux (spec are above). I got lucky and at that moment there was an update 5.15.1(the camera was older). I installed it and left the camera running for a couple of hours and then the streams appeared (with all types of wires and on all computers). I gave the camera to a colleague and it worked with Jetson Orin Nano and Linux. At some moment plugging the camera into the Jetson Orin Nano the stream disappeared almost immediately. I updated to 5.16.01 -- didn't help. I repeated the experiment and rolled the camera back to 5.13.0.55 to upgrade to 5.15.1, but I hit the update limit. 

I ended up going through all your replies on these problems and did:
- "gold" reset
- using not official but very good type-c cable with USB 3.2.
- run all variants of update-fw tools 
- run calibration tools
- restore recovery mode
- calibrate data table
- checked different combinations of fps and frame size
- restart the depth device from the Windows Device manager

I can't :
- return to the supplier  (I don't have bills or other stuff to talk with intel, the robot distributor will send the replacement)
- test another camera
- update camera firmware because of update limits

My main concern is preventing problems with the second camera that will be shipped, or figuring out how I can fix them. Could you please help me?
thanks for your time,
Artem

[image]

MartyG-RealSense, 2024-09-24:
Hi @Artii24  If there will be periods of time when the camera may be unused for a week or a month then I strongly recommend not storing it in a closed space such as a desk, drawer, suitcase, etc.  In rare cases this can cause the camera hardware to develop a depth stream failure.  Whilst there is not a known cause for this issue, it is possible that condensation in the environmental conditions in which the camera is stored might contribute to it.  

The only known possibility for correcting this issue is to place the camera in a warm place for a few weeks or to keep attempting to use the camera in its powered-on state for a number of days until it starts working correctly.  More often though, this kind of storage-induced hardware failure is not fixable.

When using RealSense cameras with Jetson Nano boards, Intel strongly recommend enabling the **barrel jack** connector for extra power using the instructions at the link below if that Nano board has a barrel jack.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

The firmware update limit only applies to installing older versions than the firmware that is currently installed.  When a new firmware is released after 5.16.0.1 you should be able to upgrade to that, but you will not be able to install old firmware versions any more.

Artii24, 2024-09-24:
Thank you for your help. 

Is it possible to send it in for repair? 

"the camera in its powered-on state for a number of day" -- Does the camera need to stream everything or just be plugged into PC Usb3.2 port?

If the climate is cold and wet, what is the best way to store the camera? on a table out in the open? or in a dehydrating bag?

Doesn't the new firmware reset the counter? Do you know how long to wait for a new update? I don't understand, why not give an opportunity to update to the latest version and then block the opportunity....

MartyG-RealSense, 2024-09-24:
If your D435i came from the robot manufacturer then the manufacturer would be the appropriate people to talk to for support.  

There is not a repair service for RealSense cameras but it is possible to perform repair yourself by purchasing a replacement camera circuit board and swapping the new one in the camera for the original one.  RealSense cameras have two circuit boards, the Depth Module (the one that the sensors are on) and a Vision Processor D4 board.  The D4 board, at around 45 Euro, is much cheaper to replace than the Depth Module.

********

Yes, I would recommend having one of the still-working streams such as RGB activated during power-on sessions so that the camera is drawing power and heating up internally.

*********

There is not advice available about storing the camera in particular environments when it is not operating.  In the data sheet document for the 400 Series cameras though, it is stated that it should be stored in 'still air' and that 'controlled conditions should be used for long-term storage of [the] product'.  In regards to humidity level, it recommends 'Temperature / RH: 40 degrees C / 90%'.

********

The firmware update counter cannot be reset, even if the firmware is updated to a new one.  Once it reaches the 20 downgrade limit then the ability to downgrade to an older firmware is permanently locked for the lifetime of the camera.  Since the firmware is stored on the Vision Processor D4 board, replacing the camera's D4 board should reset the counter to zero.

******

New firmware versions typically arrive at the same time that a new librealsense SDK version is released.  The current latest 2.56.1 did not have a new firmware because it is an 'interim beta' release and so continued to use 5.16.0.1 but the next librealsense release will likely be accompanied by a new firmware.

Each librealsense version typically has its own specific recommended firmware, as using an older firmware with a newer librealsense release could mean that the firmware does not contain necessary functions for that latest librealsense to operate correctly.

Artii24, 2024-09-24:
thanks for the reply.
The manufacturer just sent out another one, I was wondering what to do if you are an individual -- buy a replacement board, thanks.
How do I know what needs to be replaced? It seems as if the D4 board is working properly.

MartyG-RealSense, 2024-09-24:
There is not information about which of the two boards is affected.  But if depth fails but RGB works normally, this suggests to me that the control board (the Vision Processor D4) is unaffected and the failure occurs on the depth module.

Artii24, 2024-09-24:
Motion module is a separate board, isn't it?

MartyG-RealSense, 2024-09-24:
35 degrees C is the official recommended maximum internal operating temperature.  But in real world conditions it could be around 42 degrees C before problems start to occur in the depth image above that level.  At 60 degrees the firmware driver shuts off the projector's laser to try to reduce the camera's temperature.

The motion module (IMU) is integrated on the Vision Processor D4 board on versions of it that have an IMU.  There are 5 different versions of the D4 board.  A D435i uses the **V3** version of the D4 board.

Artii24, 2024-09-25:
Does the depth module have different versions? which one is preferred for  D435i?

MartyG-RealSense, 2024-09-25:
Yes, there are a range of different depth modules.  The D435i uses the **D430** depth module.

A **Left IC2 Config error** can be an indication - though not always - of a problem with the interconnecting 'interposer' cable that connects the Vision Processor D4 board and the Depth Module board together, such as the cable not being fully seated in the cable's connector on one of the boards.

**Asic Temperature value is not valid** warnings are typically not serious.  They can simply mean that the camera was not ready to provide an Asic temperature reading in the instant that it was requested.  The Asic refers to the Vision Processor D4 board.

Artii24, 2024-09-27:
Is it possible to check that this "interposer" cable is connected well enough?

MartyG-RealSense, 2024-09-28:
There is no way to check without opening the camera casing.  If you purchased the camera less than a year ago then I would not recommend this as you would very likely invalidate the camera's sales warranty.  

If you do not mind about not being able to return the camera under warranty then you can open the casing to check.  If you unplug the USB cable from the side of the D435i then there are two small screws.

MartyG-RealSense, 2024-10-03:
Hi @Artii24  Do you require further assistance with this case, please?  Thanks!

Artii24, 2024-10-04:
You have helped me a lot, but it didn't solve my problem. I will try to check the connection between the boards later. The issue can be closed. Thanks again(:

closed 2024-10-04 by Artii24

MartyG-RealSense, 2024-10-05:
You are very welcome.  Good luck!