realsenseai/librealsense #13373: how to install wrapper python on raspberry pi 5

lam26817938, 2024-09-24:
| Camera Model                       | D400 | 
| Operating System & Version |   Linux raspberrypi 6.6.47+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux  |                                  
| Platform                                 | Raspberry Pi 5  |
| SDK Version                            |  2   | 
| Language                            |  python   |            

### Issue Description

I want to use Realsense with OpenCV on Raspberry Pi 5. I have tried some guides for the old Raspberry Pi version, but they didn't work for Pi 5. Any solution or suggestions? 

This is where I stuck when trying to install from my build

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/lian/librealsense/wrappers/python
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/home/lian/librealsense/wrappers/python/setup.py", line 5, in <module>
          from pyrealsense2._version import __version__
        File "/home/lian/librealsense/wrappers/python/pyrealsense2/__init__.py", line 2, in <module>
          from .pyrealsense2 import *
      ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.
```

Thanks

MartyG-RealSense, 2024-09-24:
Hi @lam26817938  Whilst the librealsense SDK does have support for installing the pyrealsense2 wrapper from Python PyPi wheel packages on Arm archtecture devices with **pip install pyrealsense2**, this method does not work on Raspberry Pi boards unfortunately and the pyrealsense2 wrapper should be built on Pi from source code instead.

The easiest approach may be to install librealsense and pyrealsense2 together at the same time using a pre-made build script provided by the **libuvc backend** installation method, which works well on Pi.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

To install pyrealsense2 at the same time as librelsense, line 46 of the 'libuvc_installation.sh' build script should be changed from this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

to this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS:bool=true`

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

lam26817938, 2024-09-24:
thanks for your response

what should I do if I got this

```
+ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
++ uname -r
+ echo 'Kernel version 6.6.47+rpt-rpi-2712'
Kernel version 6.6.47+rpt-rpi-2712
+ sudo apt-get update
Hit:1 http://deb.debian.org/debian bookworm InRelease
Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]                                                                                                           
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]                                                                                                                     
Hit:4 https://download.docker.com/linux/debian bookworm InRelease                                                                                                                           
Get:5 http://archive.raspberrypi.com/debian bookworm InRelease [39.0 kB]                                                                                             
Err:6 https://librealsense.intel.com/Debian/apt-repo bullseye InRelease
  403  Forbidden [IP: 2600:9000:21dd:9e00:1a:eb13:66c0:93a1 443]
Get:7 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [477 kB]
Get:8 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [506 kB]
Reading package lists... Done     
E: Failed to fetch https://librealsense.intel.com/Debian/apt-repo/dists/bullseye/InRelease  403  Forbidden [IP: 2600:9000:21dd:9e00:1a:eb13:66c0:93a1 443]
E: The repository 'https://librealsense.intel.com/Debian/apt-repo bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

which raspberry OS do you recommend, seem my current one is too new.

MartyG-RealSense, 2024-09-25:
Buster is the last known Raspberry Pi OS version to work well with RealSense cameras.

The next OS after that, Bullseye, had problems with RealSense.

The link below has an installation guide for Bookworm and librealsense on Raspberry Pi, though I am not aware of anyone who has tried the guide to confirm if it works.

https://couka.de/2024/02/11/librealsense-on-raspberry-pi-2-raspberry-os-bookworm-and-python-3-11/

lam26817938, 2024-09-26:
It is installed and can be imported, but it seems it can only be read as a module. When I try to use it, I get this error: ‘module ‘pyrealsense2’ has no attribute ‘XXX.’
I have tried some solutions from other issues, but I still don’t understand the cause of this problem. I’m not an expert on this. Thank you for your help.

lam26817938, 2024-09-26:
Also, I got this message when I run `./vcpkg integrate install`
"Make projects should use "-dcmake_toolchain_file=/home/lian/vcpkg/scripts/buildsystems/vcpkg.cmake"

MartyG-RealSense, 2024-09-26:
There is limited information about installing pyrealsense2 for use with Raspberry Pi OS, unfortunately as it is much more common to install it for Ubuntu on Pi, using the libuvc_installation.sh build script mentioned earlier in this discussion. 

Using vcpkg is not usually necessary in order to install librealsense on a Raspberry Pi.  And installing librealsense from vcpkg and then installing it a second time from source code using the instructions under the **build pyrealsense Python bindings** heading could lead to two separate librealsense installations on the same computer that could conflict with each other.

So my advice would be to skip trying to install librealsense with vcpkg, as it would be unnecessary when you have installed it with git clone.

I would also recommend adding the CMake build flags below to your CMake build instruction.

**-DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release**

So it would look like this:

`cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

closed 2024-10-10 by MartyG-RealSense