realsenseai/librealsense #13374: Librealsense v2.55.1,JetsonPack 6.0

2805651606, 2024-09-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (5.16.0.1) | 
| Operating System & Version |  (Ubuntu22,Jetsonpack6.0) | 
| Kernel Version (Linux Only)    |  (5.15.136-rt-tegra)                                         | 
| Platform                                 | NVIDIA Jestion Orin Nx  |
| SDK Version                            |  { 2.55.1}                          | 
| Language                            |  {python/C }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
使用下面命令进行安装的
$ wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
$ chmod +x ./libuvc_installation.sh
$ ./libuvc_installation.sh
安装完以后，rs-enumerate-devices 可以输入相机的信息，但是运行realsense-viewer
[image]
[image]

MartyG-RealSense, 2024-09-25:
Hello, please try editing line 46 of the `libuvc_installation.sh` script to include the flags **-DBUILD_EXAMPLES=TRUE** and **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** and then running the script to build librealsense with support for graphics and the graphical tools and examples including realsense-viewer.

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

*******

您好，请尝试编辑 `libuvc_installation.sh` 脚本的第 46 行，以包含标志 **-DBUILD_EXAMPLES=TRUE** 和 **-DBUILD_GRAPHICAL_EXAMPLES=TRUE**，然后运行脚本以构建 librealsense，支持图形和图形工具以及包括 realsense-viewer 在内的示例。

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

MartyG-RealSense, 2024-10-03:
Do you require further assistance with this case, please?  Thanks!

您需要进一步协助处理此案吗？谢谢！

2805651606, 2024-10-10:
[image]
会显示No Frame Reviewed

MartyG-RealSense, 2024-10-10:
If you have built librealsense with libuvc_installation.sh with the example programs included as suggested above at https://github.com/IntelRealSense/librealsense/issues/13374#issuecomment-2374342809 - does the IMU also not display accel and gyro if you run the example program **rs-motion** in the 'usr/local/bin' folder of Ubuntu?

2805651606, 2024-10-11:
> [如果您已经使用 libuvc_installation.sh 构建了 librealsense，其中包含了上面#13374（评论）](https://github.com/IntelRealSense/librealsense/issues/13374#issuecomment-2374342809)中建议的示例程序- 如果您在 Ubuntu 的“usr/local/bin”文件夹中运行示例程序**rs-motion**，IMU 是否也不会显示加速度计和陀螺仪？

[image]
[image]
这算是不显示？

MartyG-RealSense, 2024-10-11:
If the graphic of the IMU on the screen in rs-motion moves when you move the camera then gyro and accel are detected and working.  This program does not display the accel and gyro values though.

如果移动相机时 rs-motion 屏幕上的 IMU 图形移动，则表示陀螺仪和加速度计已检测到并正常工作。但此程序不显示加速度计和陀螺仪值。

2805651606, 2024-10-12:
我进行了测试，屏幕上的imu图形没有跟着相机移动，一直保持不动，我这接下来应该怎么办呢

MartyG-RealSense, 2024-10-12:
The only remaining option that I can see is to install the RealSense MIPI driver at the link below that provides IMU support for JetPack 6 when using the camera on a USB cable connection.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

Questions about installing the driver can be posted at the link below.

https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues

*********

我能看到的唯一剩余选项是安装下面链接中的 RealSense MIPI 驱动程序，该驱动程序在通过 USB 电缆连接使用相机时为 JetPack 6 提供 IMU 支持。

https://github.com/IntelRealSense/realsense_mipi_platform_driver

有关安装驱动程序的问题可以在下面的链接中发布。

https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues

MartyG-RealSense, 2024-10-28:
Case closed due to no further comments received.

由于未收到进一步评论，案件结案。

closed 2024-10-28 by MartyG-RealSense