realsenseai/librealsense #13375: How to read multi D435i RGB

HorizonRy, 2024-09-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version | Ubuntu 20.04 LTS                                   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC                                                          |
| SDK Version                           |  { legacy / 2.<?>.<?> }                           | 
| Language                               |  Python                                                  | 
| Segment			        |   Robot                                                   |

MartyG-RealSense, 2024-09-25:
Hi @HorizonRy  Multiple RealSense cameras can be distinguished by their serial number, which can be accessed by a program.  A program can either automatically build a list of all attached cameras or use the serial number to access one specific camera.

There is an example of a RealSense Python multiple camera program at the link below.

https://github.com/ivomarvan/samples_and_experiments/tree/master/Multiple_realsense_cameras

Alternatively, a script at https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-1264583048 takes the approach of creating a separate configuration for each of two cameras.

MartyG-RealSense, 2024-10-03:
Hi @HorizonRy  Do you require further assistance with this case, please?  Thanks!

HorizonRy, 2024-10-06:
> Hi @HorizonRy Do you require further assistance with this case, please? Thanks!

It's helpful, Thanks!

closed 2024-10-06 by HorizonRy