realsenseai/librealsense #13382: d455 not getting recognised by realsense viewer

RISHI-VEMPATI, 2024-09-27:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|

Please provide a description of the problem
I am using a intel realsense depthcamera D455 with jetson orin nano. Camera is not being detected by the intelrealsense viewer sdk 2.0. but it is getting detected by jetson orin nano . i have checked it using "lsusb" command
[image]
[image]

MartyG-RealSense, 2024-09-27:
Hi @RISHI-VEMPATI  Are you using JetPack 6 on your Orin Nano, please?  If you are, JetPack 6 support was not introduced into the SDK until the current latest SDK version 2.56.1.  It is possible to use 2.55.1 with a Jetson with JetPack 6 if the SDK is built from source code with a build script provided by the **libuvc backend** procedure at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

If you need to use the RealSense Viewer tool then it can be included in the libuvc build of the SDK by changing line 46 of the 'libuvc_installation.sh' build script to the line below.

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_WITH_CUDA=on -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

RISHI-VEMPATI, 2024-09-28:
Yea, i am using jetpack 6.0 . I am working on building an open source AMR for a project using ROS2 humble hawksbill for which I need Ubuntu 22.02. so , I am using jetpack 6.0.

MartyG-RealSense, 2024-09-28:
JetPack 6 does not contain a function called **hiddraw** that RealSense cameras equipped with an IMU component (such as D455) depend upon.

If the 'MIPI driver' at the link below is installed then it can provide IMU support for RealSense cameras that are on a USB cable connection.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

Due to the complexity of installing this driver though, I strongly recommend the much easier option of building the librealsense SDK with the libuvc backend or RSUSB backend method mentioned above.

RISHI-VEMPATI, 2024-10-01:
[image]
i have followed the instructions for building the librealsense SDK with the libuvc backend. i have also changed the line 46 of the mentioned script. but when i am running 'rs-enumerate-devices' in terminal i am getting that no devices detected even though it is plugged in, i have also attached the screenshot

MartyG-RealSense, 2024-10-01:
Thank you for the update.  Does it make a difference if you launch rs-enumerate-devices in **sudo** admin permissions mode with the command  `sudo rs-enumerate-devices`

RISHI-VEMPATI, 2024-10-01:
[image]
no the output didn't change

MartyG-RealSense, 2024-10-01:
=Is the camera still undetected if you unplug the micro-sized end of the USB cable from the base of the camera, turn the connector around the other way and re-insert it (USB-C cables are two-way insertion at the micro-sized end).

RISHI-VEMPATI, 2024-10-01:
i am using a type-A to type-C cable. with type-A port connected to jetson orin nano and type-C port to camera. one thing i would like to add is that i came to know that this not D455 camera, instead it is a D457 camera.

MartyG-RealSense, 2024-10-01:
D457 cameras are designed to be used with a high speed **GMSL** cable by default and are not detected when plugged into a USB port.  To use the camera on a USB connection, you need to take off a small removable cap on top of the camera and move a physical switch under the cover sideways into the 'USB' position.  The D457 should then be detected as a D455.

Instructions for performing this mode change can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

RISHI-VEMPATI, 2024-10-01:
i have already changed the physical switch. the camera is being detected on other PC's with USB port.

MartyG-RealSense, 2024-10-01:
It could then be an issue with insufficient power being provided by the USB port on the Orin Nano - just enough for lsusb to read its details but not enough for a program like rs-enumerate-devices to access it.

Do you have the option to test the camera with a mains electricity powered USB hub instead of plugging it directly into the Orin Nano in order to confirm whether it is a power related issue?

RISHI-VEMPATI, 2024-10-01:
i have connected the camera to my PC, it is working which implies that it is getting power properly. So, inorder to check the power out by USB Port i have used a multimeter and esp32. i have measured volatge between Vin and Gnd. it turnsout it is 4.78V, 4.9V in orin nano USB Port, My PC's USB Port respectively.

MartyG-RealSense, 2024-10-01:
Although USB is a standard, not all USB ports are equal and the in-built ports on one machine (especially single-board computers) may have problems that a desktop / laptop PC's ports do not.  

Sometimes though it is not related to insufficient power but instead due to the brand of **USB controller** that a manufacturer has chosen to handle their in-built ports (some USB controller brands work better with RealSense cameras than others).

However, other Orin Nano users have been able to successfully use their RealSense cameras with this board, which makes it less likely that your issue is due to a problem in the Orin Nano brand of Jetson board.

There have been rare cases for Jetson users where the libuvc backend method has not worked but the **RSUSB backend** method has.  The libuvc and RSUSB methods both work in a similar way (bypassing the kernel) so there should not be a difference between them, but apparently there sometimes is.

The easiest way to test this may be to use the libuvc_installation.sh script again, but this time change line 46's -DFORCE_LIBUVC=true flag to **-DFORCE_RSUSB_BACKEND=true**

RISHI-VEMPATI, 2024-10-01:
[image]
Still getting the same output

MartyG-RealSense, 2024-10-01:
Have you tested with other programs, such as **rs-hello-realsense**

RISHI-VEMPATI, 2024-10-03:
Tried this now. It still shows no devices detected.
[image]

MartyG-RealSense, 2024-10-03:
I examined this case again thoroughly from the beginning, considering every aspect.

- The camera works on PCs but not on Jetson.
- The camera is detected by lsusb but not by RealSense programs such as rs-enumerate-devices and rs-hello-realsense.
- Launching programs with sudo admin permissions does not make a difference.
- Building the librealsense SDK with the libuvc and RSUSB backends does not make a difference.  
- Although JetPack 6 is being used with a camera model that has an IMU (D457), which can cause non-detection, libuvc and RSUSB can solve this problem.  But not apparently in this case.
- The D457's mode switch is confirmed to be moved sideways into the 'USB position.  The camera would not work on other computers if the switch was not in USB mode.

This does not leave many remaining possibilities to explore unfortunately, except for testing the camera on a USB hub to test whether your particular Jetson board has been equipped with a USB controller brand by the manufacturer that is not fully compatible with RealSense cameras.  A hub uses its own separate USB controller and so plugging a camera into the hub instead of a computer's built-in port can enable the camera to work on a hub port when it will not detect on a built-in port.

RISHI-VEMPATI, 2024-10-04:
I removed the binary installation of the package. Now, the camera is being detected properly. Thank you for your swift responses for the issue. I am grateful for your help.

closed 2024-10-04 by RISHI-VEMPATI

MartyG-RealSense, 2024-10-04:
You are very welcome.  It's excellent news that you succeeded.  Thanks very much for the update!

joonhyeok1231, 2025-05-15:
> I removed the binary installation of the package. Now, the camera is being detected properly. Thank you for your swift responses for the issue. I am grateful for your help.

Hello, I'm also connecting the realsense d455 camera to the jetson orin nano board, and the same thing happened to you. The lsusb command shows the camera in the list, but the rs-enumerate-devices command gives me the same response as you. Can you tell me in detail how you solved it?

RISHI-VEMPATI, 2025-05-15:
remove the the binary installation, install it from source. But, still
going forward you will face many issues. I would suggest to use a d435i
camera instead of d455.