realsenseai/librealsense #13383: Issues with Frame Drop and Synchronization Using Multiple Intel RealSense D455F Cameras

absargana, 2024-09-27:
Hi everyone,
I'm working on a project involving head pose estimation and eye blink detection using three Intel RealSense D455F cameras in a recording room. The goal is to capture synchronized RGB and depth streams. Here’s my setup:
•	Camera Model: D455F (3 cameras)
•	Firmware Version: 2.55.16.486
•	Operating System: Windows 11
•	SDK Version: 2.55.16.486
•	Programming Language: Python
•	Recording Tool: RealSense Viewer
•	Connection: All cameras are connected via a USB hub to my computer.
I'm saving the recordings as .bag files. However, I’m facing a couple of challenges:
1.	Frame Drop: When using all three cameras simultaneously, I experience up to a 25% frame drop, which significantly affects my data quality.
2.	Synchronization: How can I achieve proper synchronization across the three cameras to ensure aligned data streams?
3.	File Format: Is it better to continue saving the output in .bag files or save the RGB and depth streams directly for easier processing?
Any suggestions or best practices on handling these issues would be greatly appreciated. Thank you!

MartyG-RealSense, 2024-09-27:
Hi @absargana  Thanks very much for your questions.

1.  What is the specification of the computer that you are using with the cameras, please?  Each camera that is enabled simultaneously consumes more of the computer's resources.  It is recommendable to use a computer with at least an Intel Core i5 processor or equivalent for handling three simultaneous cameras.

2.  Whilst depth can be synchronized between multiple cameras using hardware sync (joining the cameras together with wiring linking to their sync pins (under a small removable cap on top of the camera), there is ony one mode for depth + rgb synchronization with hardware sync.  Called **genlock hardware sync**, the RGB sync did not work well though and genlock sync is no longer supported by Intel (though is still accessible in the Viewer and SDK).

It may be better simply to start all cameras simultaneously using Python code, like the multiple camera project at the link below.

https://github.com/ivomarvan/samples_and_experiments/tree/master/Multiple_realsense_cameras

3.  If you need to access the recorded data and interact with it in real-time then .bag files are the best choice.  If you only need to play back the files to view them but not interact with the data then recording to a non-interactive video format such as .avi or .mpg should be sufficient.

MartyG-RealSense, 2024-10-03:
Hi @absargana  Do you require further assistance with this case, please?  Thanks!

absargana, 2024-10-03:
Hi MartyG-RealSense,

Thank you very much for your detailed response. I have implemented the multi-camera code to address the synchronization issues, and it seems to be working well
[image]
 as all cameras now start recording simultaneously. However, I am still experiencing significant frame drops, as indicated in the attached snapshot. Could you please guide me on how to resolve this issue?

Additionally, is there a way to start all cameras simultaneously using the RealSense Viewer? This feature would be very convenient for both recording and monitoring the recordings.

I look forward to your guidance.

Best Regards,

Allah Bux

MartyG-RealSense, 2024-10-03:
On each camera in the Viewer side-panel, please try expanding open its RGB settings and disabling an option called **Auto-Exposure Priority** that is enabled by default.  

When auto-exposure is enabled and Auto-Exposure Priority is disabled, RealSense programs such as the Viewer will attempt to enfoece a constant FPS rate for both depth and RGB instead of allowing the FPS to vary.

********

It is not possible to enable all cameras simultaneously in the Viewer, unfortunately.

********

RealSense applications can take advantage of an Nvidia GPU if the RealSense SDK is built from source code with CMake with the flag **-DBUILD_WITH_CUDA=TRUE** included in the CMake build instruction.  However, it only provides improved performance for three specific types of operation - pointclouds, depth-color alignment and YUY to RGB color conversion - so it may not be worth enabling support for your Nvidia GPU if your project will not be using pointclouds or alignment.

absargana, 2024-10-04:
Hi,

I've disabled the Auto-Exposure Priority option, but I'm still encountering frame drop issues. By using two laptops for recording, the frame drop rate has significantly decreased to approximately 5%.  Then , issue is how to synchronize the recording of multiple cameras effectively using two computers. 

Currently, I am recording at 30 FPS with a resolution of 1280 x 720 for the color stream and 848 x 480 for the depth streams. Is it possible at all  to achieve 0% frame dropout, and if so, how can this be accomplished?

Any suggestions or assistance would be greatly appreciated.

Best regards,
AB Sargana

MartyG-RealSense, 2024-10-04:
Capture and synchronization with multiple RealSense 400 Series cameras and multiple computers, with one camera per computer, has been demonstrated by Intel before in 2018.  There is not a step-by-step guide to how it was set up though.

https://www.intelrealsense.com/intel-realsense-volumetric-capture/

***********

Intel have previously provided a multiple camera hardware sync mode for depth + RGB.  It was called **genlock**.  Intel no longer support genlock modes of hardware sync but genlock is still accessible in RealSense programs such as the Viewer.  It is enabled by setting the **Inter Cam Sync Mode** option on each camera to a value of '3'.  The cameras also have to be physically connected together with a 'hardware sync cable' that you must create yourself.

However, depth + RGB sync mode '3' was designed for the D415 camera model and may not work well, or at all, with a D455f.

THough the documentation for genlock sync is no longer available online, if you are interested in exploring it then you can access an archived PDF document version of the documentation at the link below.

[External Synchronization.pdf](https://github.com/IntelRealSense/librealsense/files/11382074/External.Synchronization.pdf)

********

In regard to achieving 0% frame drop, the likelihood of frame drops occurring can be reduced by adding **latency** to the pipeline by increasing the 'frame queue size', but the RealSense Viewer does not have the option to configure the queue size.

The Viewer's settings interface (accessed by the gear-wheel icon in the top corner of the Viewer window) does though offer the ability to configure whether compression is enabled during bag recording.  Compression can cause frame drops, so you can set compression to 'Never' to ensure that it can never occur.  If you make this change then click the 'Apply' button afterwards to confirm the change.

[image]

absargana, 2024-10-14:
Hi MartyG-RealSense,

Thank you for your support so far and for inquiring about the status of the issue. I have been trying a software-based solution because I do not want to go for hardware synchronization, as you mentioned that it may not work well with D455f cameras. Upon analysis, I have identified that the problem initially perceived as frame drops is an issue of frame delay. Intel RealSense viewer appeared to show frame drops, but it turned out that the frames were queued and subsequently saved to disk, with actual frame drops being less than 1%.
However, synchronization remains a significant challenge. Despite starting and stopping the cameras simultaneously, there is a misalignment between the RGB and depth streams of the same camera. For example, if depth frames drop while RGB frames continue recording, it disrupts synchronization and worsens over time. Additionally, synchronization among the three cameras presents further complications.
I have experimented with several solutions, such as dropping corresponding RGB frames when a depth frame drops to maintain synchronization for a single camera. This method has shown some improvement, but issues across multiple cameras persist. Any further help or guidance will be highly appreciated.

MartyG-RealSense, 2024-10-14:
If delay is being caused by frames being saved to disk then you can store frames in the computer's memory instead using an instruction called **Keep()**.  

With Keep(), all the frames are stored in memory during streaming and then when streaming is stopped, they can be written to file in a single batch action.  Processing such as post-processing filters and depth-color alignment can also be batch-applied to a set of frames like this.  https://github.com/IntelRealSense/librealsense/issues/6146 has information about using Keep() in a Python script.

closed 2024-10-28 by MartyG-RealSense