realsenseai/librealsense #13390: realsense-viewer has display bugs in ubuntu 20.04

YiKang-Liu, 2024-10-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

my camera model is d455, and my system version is ubuntu 20.04 and kernel version is 5.15.0-122-generic.
Now I follow the guide in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md to install the SDK as source code. And finall I can start and run realsense-viewer successfully

However ,I run this SDK in my labtop and I have an external display. SDK works fine on an external display, but not on the laptop's own screen.The display on the laptop's own screen looks as shown here
[image]

I can't do anything on this screen, this screen just shows a plus sign that maximises this screen much more and shows one more add. can't do anything beyond that.What's wrong with it and how can I fix it?
I was hoping someone could teach me how to solve this problem

MartyG-RealSense, 2024-10-01:
Hi @YiKang-Liu  This is a rare display problem that has occurred a small number of times in the past.  

If pressing the F8 key on the keyboard to toggle the Viewer's full screen display mode on and off does not make a difference, further advice about the problem can be found at https://github.com/IntelRealSense/librealsense/issues/5511

YiKang-Liu, 2024-10-01:
I've used all the methods mentioned in this reply https://github.com/IntelRealSense/librealsense/issues/5511 and very sadly none of them work.Using F8 results in an even bigger, full-screen “Add”.
I then tried to reinstall the SDK without using an external monitor, but it still didn't work, and it still displayed like this in the laptop's built-in monitor
[image]
Then, I try to add scale_factor = 1.f; before imgui_impl_glfw.cpp:243.
The situation has finally changed, but it is what it has become, and it does not solve the problem
[image]
The most difficult thing for me to understand is that there is an animation when launching realsense-viewer, which is displayed normally and at the same resolution as my laptop's built-in monitor. But when the animation finishes loading and the functional interface is launched, it goes back to the previous situation
I really don't understand why realsense-viewer shows up and works fine in an external monitor, but in a laptop's monitor this problem occurs!

MartyG-RealSense, 2024-10-01:
Does the problem still occur if the external monitor is unplugged from the laptop before launching realsense-viewer?

YiKang-Liu, 2024-10-01:
yes...I even tried uninstalling and reinstalling the SDK without an external monitor connected, but the result remains the same. It does seem very confusing to me.
But I haven't tried installing the SDK using the debian package method though, I'll try that later and hopefully that will be useful

YiKang-Liu, 2024-10-02:
I'm sorry to say that trying to install the SDK from Debian's repositories didn't work, and the program still has the wrong resolution in my computer's own display.
I even went as far as deleting the existing version of the SDK (2.55.1) and reinstalling an older version (2.50.0), but the problem still occurs.
Even when using the 2.50.0 version of the SDK, it keeps writing to /var/log/uvcdynctrl-udev.log until it fills up the entire root /
Maybe focusing on ros usage and ignoring realsense-viewer is a better option for me

YiKang-Liu, 2024-10-02:
Yes, only in the built-in display of the laptop, realsense-viewer has a resolution error. The interface of the other programs in the built-in display of the laptop and in the external display is normal

MartyG-RealSense, 2024-10-02:
Does your camera have the **5.16.0.1** version of the camera firmware driver installed, as this is the version recommended for use with 2.55.1.

YiKang-Liu, 2024-10-03:
My camera model is d455 and the current firmware version is 05.13.00.50. Also the SDK version used now is 2.50.0.
Previously my camera's firmware was 5.16.0.1 while the SDK version was 2.55.1 and there was a problem with the SDK resolution. Now I lowered the firmware version of the camera and the SDK version at the same time, and the problem is still not solved!
If this problem is really hard to solve, let's put it aside and hopefully I won't have too much trouble with the camera's ros wrapper

MartyG-RealSense, 2024-10-03:
I researched the problem of realsense-config.json being unable to be found.  On Windows 11 the Appdata folder is hidden by default.  You can make it visible in your user account folder by going to the top of the file navigator panel and selecting the menu option **View** and then the sub-options **Show > Hidden items**.

[image]

Once viewing of hidden items is enabled, the AppData folder appears in the user account folder and realsense-config.json can be found in AppData > Roaming.

landert-elon, 2024-10-08:
I meet the same bug.Perhaps this is the reason why Intel's market value is now less than Jensen Huang's personal wealth and less than half of AMD's.
[image]

MartyG-RealSense, 2024-10-08:
Hi @YiKang-Liu  Were you able to find the realsense-config.json file using the instructions at https://github.com/IntelRealSense/librealsense/issues/13390#issuecomment-2391304072

YiKang-Liu, 2024-10-08:
ennnn...
I tried it and the SDK is working fine in my windows 11 and then this config file can be found. But I need to use ubuntu for the project, and then there is no way to solve this problem in ubuntu, so I'm shelving this for now. 
The good thing is that realsense's ros package is working fine (although there are some bugs too), but at least it works

MartyG-RealSense, 2024-10-08:
On Ubuntu realsense-config.json should be able to be deleted with the command below to reset the configuration:of realsense-viewer to its defaults:

`rm ~/.realsense-config.json`

YiKang-Liu, 2024-10-09:
I've tried that, but it doesn't work.
Here is the default content of .realsense-config.json
{
  “viewer_model.sdk_version": ‘25000’,
  “window.position_x": ‘155’,
  “window.position_y": ‘147’,
  “window.saved_pos”: “1”
}
Is the content correct?

MartyG-RealSense, 2024-10-09:
I do not have access to the Ubuntu version of realsense-config.json but I have attached the version provided for Windows so you can open it as a text file to view its contents.  In my one, window.position_x is 0 and window.position_y 34.

[realsense-config.json](https://github.com/user-attachments/files/17308435/realsense-config.json)

YiKang-Liu, 2024-10-10:
I use the realsense-config.json you provied, but it still doesn't work

MartyG-RealSense, 2024-10-10:
I researched this issue again from the beginning but could not find any solutions other than those that have already been discussed, unfortunately.

MartyG-RealSense, 2024-10-11:
You are very welcome.  Thanks very much for your kind patience and understanding!

closed 2024-10-12 by YiKang-Liu