realsenseai/librealsense #13393: WIN11: RealSense SDK ERROR rs2_create_context_ex..

RoyAmoyal, 2024-10-02:
Hey, I download the SDK 2.55/2.56 and the newer to my windows 11. 
[image]

When I try to open to open the file after installing it I get this error. 
[image]

The realsense-config.json doesn't exist (C/users/[user]/AppData/Roaming) and I couldn't find it. 

How can I fix that problem?

MartyG-RealSense, 2024-10-02:
Hi @RoyAmoyal  Can you confirm if you have downloaded and run the stabdalone RealSense Viewer **Intel.RealSense.Viewer.exe** from the 'Assets' file list of 2.55.1's Release page, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

[image]

Once downloaded, the .exe file must be run to launch this version of the RealSense Viewer.  The shortcut launch icon that the full SDK installer places on the Windows desktop shold not be used to launch the standalone Viewer.

RoyAmoyal, 2024-10-02:
I got the same error, even after downloading the Intel.RealSense.Viewer.exe.
Do I need to download more files? Or just the SDK and the viewer exe files?

MartyG-RealSense, 2024-10-02:
The errno=42 error has been reported most often for version 2.55.1.  

You should not need to download anything other than the .exe when using the standalone Viewer.

Does the error still appear if you use the Viewer .exe for 2.54.2?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

closed 2024-10-20 by MartyG-RealSense