realsenseai/librealsense #13395: High-speed capture mode of Intel® RealSense™ Depth Camera D435

BertieBcs, 2024-10-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         | set high-speed capture mode  for IR camera |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  { v2.56.1 }                          | 
| Language                            |  Python | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello, I would like to use high-speed capture mode by using the 300fps monochrome image stream via the IR cameras; as described in the 2020 white-paper:
https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435

the following way:
pipeline_profile = cfg.resolve(pipe)
device = pipeline_profile.get_device()

for sensor in device.sensors:
    if sensor.name == 'Stereo Module':
        for profile in sensor.profiles:
            if profile.stream_type() == rs.stream.infrared:
                cfg.enable_stream(profile.stream_type(), 1, 848, 100,  rs.format.y8, 300)  

pipe.start(cfg)

I somehow cannot get above 60 FPS. I am using Python 3.11 and OpenCV

Thanks for any suggestions!

best regards,

Bertie

MartyG-RealSense, 2024-10-03:
Hi @BertieBcs  848x100 High Speed Capture mode is supported on D435 for the depth and infrared streams, so you should be able to access the infrared stream at 300 FPS with Python code.

Do you get a **Couldn't resolve requests** error if you try to use an FPS higher than 60?

Is 300 FPS provided if the simple test script below is used?

```
import pyrealsense2 as rs

pipeline = rs.pipeline()
cfg = rs.config()
cfg.enable_stream(rs.stream.infrared, 848, 100, rs.format.y8, 300)

pipe.start(cfg)
```

BertieBcs, 2024-10-04:
Dear MartyG,

no, I don't get a Couldn't resolve requests error.

The test code you provided, as well as this one below

import pyrealsense2 as rs
import numpy as np
import cv2

pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_stream(rs.stream.infrared, 848, 100, rs.format.y8, 300)
pipe.start(cfg)

while True:
    frame = pipe.wait_for_frames()
    ir_frame =  frame.get_infrared_frame()
    ir_image = np.asanyarray(ir_frame.get_data())
    cv2.imshow('ir',ir_image)

    if cv2.waitKey(1) == ord('q'):
        break

pipe.stop()
cv2.destroyAllWindows()

works properly and displays a 848*100 stream (just not with 300 FPS).

Best regards,

B

MartyG-RealSense, 2024-10-05:
How do you confirm what the FPS of the infrared stream provided by the Python script is, please?

Do you have access to the RealSense Viewer tool?  If you do, what is the displayed FPS for an 848x100 infrared stream set at 300 FPS?

This is from my own D435:

[image]

[image]

BertieBcs, 2024-10-05:
The hardware FPS is indeed 300 and the viewer FPS is around 60 - but,
because of my display specs, which don't allow me to set higher refresh
rate then 60 Hz, I guess. (Driver is uptodate)

Best regards,

B

MartyG-RealSense, 2024-10-05:
The 'Viewer FPS' value can be disregarded.  The FPS value that matters is the one displayed as an overlay at the top of the stream panel when the 'i' icon on the stream panel's toolbar is enabled, like in my image above.

BertieBcs, 2024-10-06:
It appears that the camera stream is fed in with that low frame frate, at least if I open it with Python:

https://github.com/BertieBcs/RealSense-D435.git

MartyG-RealSense, 2024-10-06:
There was a past Python case at https://github.com/IntelRealSense/librealsense/issues/6578#issuecomment-643811438 where the user was using 848x100 at 300 FPS and was not getting a **Couldn't resolve requests** error but felt that they were not receiving 300 FPS consistently.

You could check the real FPS with Python code by retrieving the value of a camera **metadata** parameter called Actual FPS.  The Python code for retrieving and printing the value would be something like this:

```
fps = rs.frame.get_frame_metadata(frame, rs.frame_metadata_value.actual_fps)  
print("FPS : ", fps)
```

BertieBcs, 2024-10-06:
> There was a past Python case at #6578 (comment)

I am getting this error message:

RuntimeError: Infrared frame does not support metadata "Actual Fps"

if only the IR camera is on.

If the color and/or the depth stream is enabled, than it prints the actual
FPS.

Regards,

B

MartyG-RealSense, 2024-10-07:
Assuming that if there was a problem with infrared FPS then it would also affect the depth FPS in the same way, what Actual FPS value is provided if depth is enabled?

MartyG-RealSense, 2024-10-08:
I generated the 'frame metadata' overlay in the RealSense Viewer with only Infrared enabled and the Actual FPS was updating in real-time and running at around 300.

Because 848x100 mode in the Viewer produces a stream panel of a short height, a scroll-bar at the side of the stream panel has to be used to scroll the metadata details overlay down to find the Actual FPS value.

[image]

BertieBcs, 2024-10-08:
Now it seems to have another issue: there is no IR stream at all. Python
throws an error:
pipe.start(cfg)
RuntimeError: Couldn't resolve requests
and in the viewer the option infrared stream does not even appear. Is the
camera gone or just some settings problem?

Regards,

B

MartyG-RealSense, 2024-10-09:
If you are on Windows then there is a rare problem where the infrared streams can disappear, and in the Stereo Module of the RealSense Viewer only the depth stream is listed.

The problem is usually related to an issue with the RealSense depth and RGB drivers in the Windows Device Manager.  It can be resolved by using a procedure at the link below to uninstall these drivers **completely**, leaving no trace of them on the computer, and then reinstalling them.

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

BertieBcs, 2024-10-11:
Hello and thanks for the support. I can confirm that my camera works fine
now.

Best regards,

B

MartyG-RealSense, 2024-10-11:
You are very welcome.  I'm pleased to hear that your camera is working now.  Thanks very much for the update!

MartyG-RealSense, 2024-10-20:
Hi @BertieBcs  Do you require further assistance with this case, please?  Thanks!

BertieBcs, 2024-10-20:
No, I am fine. Thank you again!

BB

MartyG-RealSense, 2024-10-21:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2024-10-21 by MartyG-RealSense