realsenseai/librealsense #13401: Test patterns with D435i?

p-ogrady, 2024-10-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Linux Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  5.15 | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.55.1 | 
| Language                            | python | 
| Segment			|  Robot | 

### Issue Description

Is there an API that can configure the D435 unit to produce a test video/depth image, instead of the actual camera image?  This is useful, for example, in our application's CI system, where we want to prove that we can connect and see video from the camera without having to control what the camera is actually looking at.  If the test mode always produces a specific pattern, then we can test that received data to be sure that we're seeing the expected pattern.  It's fairly common for cameras to have modes that produce e.g. [SMPTE test patterns](https://en.wikipedia.org/wiki/SMPTE_color_bars) -- is it possible to configure this unit in a similar way?

MartyG-RealSense, 2024-10-06:
Hi @p-ogrady  RealSense cameras do not have a test pattern mode.  However, the RealSense SDK does have a text-based testing tool called **rs-data-collect** that will check whether frames are being received from streams without producing an image.

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

p-ogrady, 2024-10-06:
Thanks @MartyG-RealSense.  I appreciate the rs-data-collect suggestion, but just proving that we're receiving data isn't the problem; the requirement here is proving that the received data buffer has the expected data in it.  Another important use case for a feature like this is in-system self-testing.  Do please let me know if a future firmware update adds a feature like this--thanks!

MartyG-RealSense, 2024-10-07:
If such a tool does become available in the future then I will let you know.

An existing visual test method that comes to mind is one where you point the camera at a high-speed LED panel product that operates at a frequency of several hundred Hz to see if cameras are capturing color and depth.  The cameras need to be set to capture at a very low exposure time (for example <3ms) for this to be visible, otherwise the whole panel will just be a blur. 

In the past, Intel have tested the brand of panel at the link below with RealSense 400 Series cameras for the purposes of confirming whether multiple cameras were hardware synced with each other.

https://www.image-engineering.de/products/equipment/measurement-devices/900-led-panel

[image]

MartyG-RealSense, 2024-10-20:
Hi @p-ogrady  Do you require further assistance with this case, please?  Thanks!

p-ogrady, 2024-10-21:
Thanks, @MartyG-RealSense -- I truly appreciate your inputs here; these are some very useful suggestions...  but these all require external equipment in a lighting-controlled environment, neither of which are practical for our CI lab or for a robot in a self-test mode.  For now, we'll just observe the video and depth data without any verification that the content delivered to the host is reasonable, and if, at a later time, we're able to configure the device to transmit a specific image (e.g. SMPTE test patterns; but any specific content would be fine) then we can add a test to check the actual received data. You're welcome to close this item as "won't do" or "no problem found"--or keep it open as an "add a test pattern" request--whatever you believe is best.  Thanks again for your valuable input!

MartyG-RealSense, 2024-10-21:
If you need a 'test card' image that is always the same and is not a video image of what the camera is looking at, the example program **rs-software-device** might be able to meet that need.  It allows the RealSense SDK to generate a 3D model generated from a .png image file instead of a live image.  By default, it uses an image file of the RealSense logo.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

It is a C++ program rather than Python.  If your installation of the RealSense SDK includes the official example programs though then you could check whether there is a pre-built executable version of rs-software-device in the **usr/local/bin** directory of Ubuntu.

p-ogrady, 2024-10-21:
Thanks, @MartyG-RealSense -- Does the rs-software-device example pass its test data from the camera over the USB cable?  The goal of this test mode is to prove that data originating in the camera device, sent over a relatively long string of I/O components (including user-reconfigurable USB cabling), arrives in the host memory as expected.  Please understand that this is a production test that would run on every system before deployment, proving that the system was built to specification.  This test would be even more beneficial if it ran on every deployed systems' power-on self test, to reduce the time that a user would spend diagnosing a problem--like a disconnected or broken USB cable.  If rs-software-device doesn't actually send the video data from the camera device, then it would only serve as a developer test, showing that the libraries were being used as expected--which is something the developer would discover during application development anyway.

MartyG-RealSense, 2024-10-21:
No, rs-software-device is a 'virtual camera' that can run a RealSense program script without the real camera being attached.  So whilst it can provide a consistent-every-time image by using a png file, it could not act as a test of how a camera is functioning and so on those grounds it might not be suited for your application after all.

The simplest camera test would be to use a USB cable for testing that has been cut open and a connection for a voltmeter or multimeter device spliced onto it.  When the camera is plugged in but inactive, there should be 5 volts detectable on the cable (like there is for any USB device that is plugged in but not currently active).  When the camera starts streaming, voltage should rise on the meter because the camera is drawing power for the stream.  If the stream was not working or the cable was broken then voltage would likely remain at 5 V.

MartyG-RealSense, 2024-10-29:
At this point I believe that it would be best to close the issue.  Thanks again for your kind patience!

closed 2024-10-29 by MartyG-RealSense