realsenseai/librealsense #13407: Low voltages when trying to hardware sync multiple D455 cameras

Nico-Sander, 2024-10-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 } | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version  |  Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.10.192-tegra   | 
| Platform                                 | NVIDIA Jetson Orin Dev Kit 64GB  |
| SDK Version                            |  2.55.1 | 
| Language                            |  python                         | 
| Segment			|  Robotics                  | 

-------------------------------------------------------

### Issue Description
Hi there, 
I am trying to set up Hardware Sync between 2 Intel RealSense D455 cameras. I am using Python to configure and start the cameras. According to the "Multi camera configurations" whitepaper (https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration)
Pins 5 (Signal) and Pins 9 (Ground) of both cameras need to be connected. 
In the python file I then used the following code to set on of my cameras to "Master":

device_right.first_depth_sensor().set_option(rs.option.inter_cam_sync_mode, 1)  # Master
device_left.first_depth_sensor().set_option(rs.option.inter_cam_sync_mode, 2)   #  Slave

Here's the problem:
When checking the signals sent between the two cameras on Pins 9 and 5, i get the correct frequency of about 30Hz (camera configured to 30fps), but the Peak-to-Peak Voltage is only about 100mV. From what I've read, a Peak-to-Peak Voltage of 1.8V is needed for the "slave"-camera to recognize the trigger signal. I have measured the Voltage directly at the Pins, so faulty connections are not the problem. 

[image]

Has anyone experienced a similar issue before. I hope someone can help me with known explanations / solutions for my problem.

Thanks

Nico-Sander, 2024-10-08:
Hello Marty, 
yes the Master camera is actively streaming. I am visualizing the depth and color output while measuring the sync signals. I like to add, that i get the same voltage readings when starting the camera with realsense-viewer. So the Inter Sync Mode seems to be set correctly.

MartyG-RealSense, 2024-10-08:
The value of 1.8 v is always referred to in regard to use of an **external** trigger signal.  It is never referred to in regard to triggering by a master camera.

A known characteristic of a trigger generated by a master camera is that it should have a pulse width of 50 microseconds, whilst an external trigger should have a pulse width of 100 microseconds or more.

Nico-Sander, 2024-10-08:
Hi Marty, 
thanks for the explanation. I will try to measure the pulse width tomorrow and will get back to you.

Nico-Sander, 2024-10-09:
Quick follow up on my issue. The problem is resolved and the sync seems to be working. I tried to measure the pulse-width today and was able to see the 50us pulse-width generated by the master camera.

[image]

You can also see that the voltage is now 1.74V, maybe a restart of my system was all that was necessary to get a better voltage.

I validated that the sync is working, by placing a millisecond clock in the overlapping field of view of the 2 cameras. Both cameras always show the exact same state of the clock, so the cameras capture the images within the same millisecond.

[image]

MartyG-RealSense, 2024-10-09:
That's great news that your setup is working correctly.  Thanks very much for the update!

MartyG-RealSense, 2024-10-20:
Hi @Nico-Sander  Do you require further assistance with this case, please?  Thanks!

Nico-Sander, 2024-10-20:
Hi @MartyG-RealSense,
thanks again for answering all my questions. Everything works fine now.
Feel free to close this Issue.

MartyG-RealSense, 2024-10-20:
You are very welcome, @Nico-Sander - I'm pleased to hear that it is working well.  As you suggested, I will close the issue.  Thanks again!

closed 2024-10-20 by MartyG-RealSense