realsenseai/librealsense #13409: Realsense D435i device not detected by librealsense Jetson Orin NX 16GB

Austen-G, 2024-10-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | Unknown | 
| Operating System & Version |   Ubuntu 22.04 (Jetpack 6.0) | 
| Kernel Version (Linux Only)    |  5.15.148-tegra                                       | 
| Platform                                 | Jetson Orin NX 16GB |
| SDK Version                            | 2.55.1  | 

### Issue Description
I'm attempting to get a D435i stereo camera integrated with a Jetson Orin NX development board for a robotics project, but have been unable to get the realsense-viewer application or rs-fw-update tool to recognize the camera to verify that it is working. I have used this particular camera in the past with another platform with no issues so I doubt it to be a camera issue. 

I can find the camera when connected using lsusb. Initial installation used the apt repository, but after finding documentation that it may not work correctly for Jetsons, I switch to an installation from source using the libuvc installation script. 

One note is that realsense-viewer complains about finding multiple realsense udev rules, but I've been unable to fix that issue. Could anyone offer any help to get things working?

MartyG-RealSense, 2024-10-09:
Hi @Austen-G  RealSense cameras equipped with an IMU can have problems with JetPack 6 such as non-detection because JP6 lacks a function called **hiddraw** that RealSense IMU-equipped cameras (which are 'HID' devices because of their IMU) rely upon.

Yes, I would recommend using the 'libuvc_installation.sh' build script, and adding some additional configuration flags to line 46 of the script:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release - DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA=on`

The additional flags install the full range of RealSense tools and examples (including realsense-viewer) and provides CUDA support for using the Jetson's graphics GPU for accelerating processing of pointclouds and depth-color alignment.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

*********

An alternative to using libuvc_installation.sh would be to downgrade the Jetson version from 6 to 5.13 if your project permits doing so.

Austen-G, 2024-10-09:
Ok, those were the steps I followed to install with libuvc, minus one or two of the arguments in the cmake file, but just to confirm it should work with JetPack 6 if done correctly? 

Now that I think about it, it's also possible that the camera has an older firmware, could that be an issue? I'll make sure to verify the version and what it should be with this version of the SDK regardless.

MartyG-RealSense, 2024-10-09:
The libuvc installation method builds a version of librealsense that bypasses the kernel, so it is not dependent on particular JetPack versions and can therefore work on a Jetson with JP6.

It is important to use a firmware version where possible that is the recommended one for a particular librealsense version.  For 2.55.1 the recommended firmware is 5.16.0.1.

Austen-G, 2024-10-10:
No dice after updating firmware version, I do get this error though when launching the realsense SDK:

Timestamp: 917203.993568
Severity: Warn
Description: Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!

The build script  used for the libuvc differed a bit from the one I sent though, I'm going to attempt a reinstall, but could conflicting udev rules cause an issue?

MartyG-RealSense, 2024-10-10:
I'm pleased to hear that you achieved a working installation!

When the multiple udev rules problem occurs (usually when a package installation and source code installation are performed on the same computer), it is the **/etc/udev/rules.d/99-realsense-libusb.rules** rule that should be deleted to resolve the problem.  It sounds as though you are past that issue though after the fresh install.

If the streams appear to be functioning normally despite the **control_transfer returned** warning generating repeatedly then you can safely ignore this warning.

Austen-G, 2024-10-10:
Awesome! I'm getting a few ASIC temperature invalid questions as well. 

One issue still is I cannot get the accel and gyro streams online, this isn't strictly necessary at the moment for our project, but might be in the future. Is this something that can be used on the Orins at the moment? Or does the lack of the hiddraw function make that impossible for the moment?

Austen-G, 2024-10-10:
Another note, is after following this by installing the Realsense ROS wrappers, the camera has once again stopped showing up, an uninstall of the wrappers does not solve this issue.

MartyG-RealSense, 2024-10-11:
Typically, the IMU can work with JetPack 6 if one of the approaches below is used:

- Build librealsense from source code with the 'libuvc backend' or the 'RSUSB backend' installation methods; or
- Install the RealSense MIPI driver at the link below, which provides support for 'HID' devices (RealSense cameras with an IMU) on JetPack 6.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

Usually I would recommend a source code build as the MIPI driver installation process is complex.  However, a libuvc or RSUSB build of librealsense has not been providing working IMU support for all Jetson users.

Austen-G, 2024-10-11:
I'll give the MIPI driver a go, but may need to reinstall the original SDK again to get the camera to show back up, nothing after that ROS2 wrapper install. 

And I attempted to install the wrappers from the apt repositories following install option 1 for ubuntu on their page, which might have been where I went wrong, if I can get the camera to come back I'll attempt an install from source after those MIPI drivers are added.

Austen-G, 2024-10-11:
I was able to get the camera back, it looks like when installing the RealSense ROS2 wrapper it installed the apt version of the SDK as a dependency, and did not remove it on the previous uninstall. Running an autoremove fixed the issue and the camera shows back up again.

I'm going to avoid working with the MIPI driver at the moment. I need to get the IMU working eventually, but for now stereo/RGB-D imagery will be enough, I have another IMU as well that I can integrate in the short-term. 

I'll try the wrapper from source to avoid the SDK install and hopefully it'll be working!

Austen-G, 2024-10-11:
Install from source fixed it, I'll close the issue for now but will likely have more questions in the future when attempting the MIPI driver installation, thanks for the help!

closed 2024-10-11 by Austen-G

MartyG-RealSense, 2024-10-11:
You are very welcome - I'm pleased to hear that you achieved a positive outcome.  

I recommend posting any questions about the MIPI driver on its own dedicated Issues page.

https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues

ojingo, 2025-01-24:
Nearly the same config. But I think my 435i is just dead.. not a single blip from it during boot up and doesn't see it on lsusb nor does the app see it after installing your kit.. you guys have a bad batch of cameras that dont boot?

MartyG-RealSense, 2025-01-25:
Hi @ojingo  Please create a new issue for your problem by visiting the front page of this support forum using the link below and clicking on the green **New Issue** button.  Also please include a link to this discussion in the message of your new issue.  Thanks!

https://github.com/IntelRealSense/librealsense/issues

MartyG-RealSense, 2025-01-28:
Hi @ojingo  I have replied on your new issue.  Thanks!