realsenseai/librealsense #13411: NRTL certificate of D455

GLW1215, 2024-10-09:
from the NRTL statement on datasheet, the NRTL is applicable to D45x modules:
[image]
However, when I check it in Nemko, there is no camera D45x in the NRTL certificate details:
[image]

So, where can I find the NRTL certificate for D455?

MartyG-RealSense, 2024-10-09:
Hi @GLW1215  All the available regulatory certificates for RealSense cameras are listed at the link below.

https://www.intelrealsense.com/regulatory-information/#depth-cameras

The US NRTL certificate does not have D455 on it.

The D455 model is covered by the CB certificates.

CB Certificate to IEC62368-1 (has the Nemko logo at the bottom of it)
CB Certificate to IEC60825-1 (has the UL logo at the bottom of it)

GLW1215, 2024-10-10:
We initially chose the D455 camera because your datasheet indicated it has NRTL certification. Since our product needs to be sold in the United States, it must also have NRTL certification. What solutions do you have for this situation? For example, can you obtain NRTL certification for the D455?

MartyG-RealSense, 2024-10-10:
I will consult with my Intel RealSense colleagues regarding the NRTL certificate on the regulatory site, which is dated 2019 and so pre-dates the 2020 release of the D455. Thanks very much for your patience!

MartyG-RealSense, 2024-10-11:
Hi again @GLW1215  I have received advice from my Intel RealSense colleagues about your NRTL question.  They said:

"Because RealSense cameras are not AC powered (do not connect directly to an outlet), NRTL is not required.  If end customer product needs an NRTL, it can obtain NRTL even when non NRTL marked components are used.  They should review this with their product certification test house they are using.”

closed 2024-10-28 by MartyG-RealSense