realsenseai/librealsense #13414: No RealSense devices were found after Reboot [Unitree A1]

Jellal-17, 2024-10-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 2.55 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.90                                       | 
| Platform                                 | Nvidia Jetson Xavier NX  |
| SDK Version                            | 2                        | 
| Language                            |  -                        | 
| Segment			|  Robot                    | 

### Issue Description
Hello, I have a similar problem to this [issue](https://github.com/IntelRealSense/librealsense/issues/11883), though it is a bit further along the line. Replying to the [comment](https://github.com/IntelRealSense/librealsense/issues/11883#issuecomment-1585669706), I have killed a process 'RobotVision', which is something the robot broadcasts through it's network, that was using the camera stream. It worked then. But after reboot, no matter what I do, the SDK doesn't detect the camera. I tried reinstalling the SDK, following some of the issues here and as the previous time, it worked before a reboot and the loop continues. So, the output of `rs-enumerate-devices` is `No device detected. Is it plugged in?`. 

I have also tried [this](https://github.com/IntelRealSense/librealsense/issues/11883#issuecomment-1582222466), for the installation of the ROS wrapper, I have used this [method 2](https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy).

Jellal-17, 2024-10-14:
Hello @MartyG-RealSense, Sorry for the late reply. 

Yes, I did switch it off from the power button and boot it. It was the same issue. 
There is a way to not start the other processes right from the boot, so I will try to do that and update here. But for now, it's the same.

Jellal-17, 2024-10-14:
@MartyG-RealSense,

I’ve removed the 'RobotVisionSystem' process from startup, and everything is working fine now. I will go ahead and close the issue.

For future reference, if you are developing on the Unitree A1 for low-level control and need to access the camera, follow these steps:

Navigate to the RobotVisionSystem directory:  `cd RobotVisionSystem`
Open the start.sh file: `gedit start.sh` and Comment out the last line. 

Restart the robot and check with:  `realsense-viewer`

Alternatively, if you only need to access the camera for the current session:
Find the process:  `ps -aux | grep RobotVisionSystem`
Kill the process: `kill -9 ####`

However, please note that killing the process might only work for a limited number of reboots. In my case, it stopped working after the first reboot, so I recommend the first method for a more permanent solution.

closed 2024-10-14 by Jellal-17

MartyG-RealSense, 2024-10-14:
It's great to hear that you achieved a solution, @Jellal-17 - thanks very much for the update and for sharing the details of your solution!