realsenseai/librealsense #13417: Realsense-viewer does not register camera unlesss I use a docker container

AnayGokhale, 2024-10-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version | Ubuntu 22 | 
| Kernel Version (Linux Only)    |  5.15.136-tegra  | 
| Platform                                 | NVIDIA Jetson Orin Nano 8gb Developer Kit |
| SDK Version                            |  2.0 I think I am not sure how to check | 
| Language                            |  python                          | 

### Issue Description
I have been able to run realsense-viewer through a docker container following these links:https://nvidia-isaac-ros.github.io/getting_started/dev_env_setup.html and https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/realsense_setup.html
However, When I try to run it normally, realsense-viewer does not register the camera. Similarly I tried the python script in the image below, and I got Runtime Error: No device connected. Please help me fix this.
[image]

MartyG-RealSense, 2024-10-14:
Hi @AnayGokhale  Is the camera detected outside of Docker if you launch the programs with **sudo** admin permissions, please?  For example:

`sudo realsense-viewer`

`sudo python3 test.py`

AnayGokhale, 2024-10-14:
@MartyG-RealSense The camera is not detected outside the Docker even if I run it with sudo permissions. 
The SDK version is v2.55.1

MartyG-RealSense, 2024-10-15:
What procedure did you use to install librealsense outside of a Docker container, please?  For example, installing from packages using the Jetson instructions at the link below.

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

AnayGokhale, 2024-10-15:
@MartyG-RealSense Hello, I followed the steps from that exact link. It does not work still. I did not do anything besides the instructions in that link.

AnayGokhale, 2024-10-15:
@MartyG-RealSense This is the output when I run realsense-viewer:  

>  15/10 18:43:46,825 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.0/video4linux/video0
 15/10 18:43:46,944 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.0/video4linux/video1
 15/10 18:43:46,945 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.0/video4linux/video2
 15/10 18:43:46,946 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.0/video4linux/video3
 15/10 18:43:46,947 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.3/video4linux/video4
 15/10 18:43:46,948 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.3/video4linux/video5
 15/10 18:43:47,449 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.0/video4linux/video0
 15/10 18:43:47,450 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.0/video4linux/video1
 15/10 18:43:47,451 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.0/video4linux/video2
 15/10 18:43:47,451 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.0/video4linux/video3
 15/10 18:43:47,452 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.3/video4linux/video4
 15/10 18:43:47,452 INFO [281473468672064] (backend-v4l2.cpp:753) Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-1/1-1:1.3/video4linux/video5
 15/10 18:43:47,460 INFO [281473468672064] (context.cpp:128) **Found 0 RealSense devices** (0xff requested & 0xff from device-mask in settings)

MartyG-RealSense, 2024-10-16:
As you are using Ubuntu 22 and a Jetson, are you using JetPack 6 please?  JP6 can cause RealSense cameras equipped with an IMU like D435i (known as HID devices) to not be detected, as JP6 lacks a function called 'hiddraw' that the IMU-equipped RealSense cameras rely upon.

There are some librealsense installation methods that can workaround the IMU issue though to enable the camera to work. These are:

1.  Install the RealSense MIPI driver at the link below, which adds support for IMU when using JetPack 6.  I do not recommend this method if possible to avoid it though due to the complexity of the installation procedure.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

Or:

2.  Build librealsense from source code with the 'libuvc backend' installation method, which has a build script called 'libuvc_installation.sh'

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

A libuvc build of librealsense bypasses the kernel and so is not dependent on a particular Linux version or kernel version and does not need to have a kernel patch applied to it.

Or:

3. Downgrade yuor JetPack version from 6 to 5.1.3, though this is not an option for many RealSense Jetson users.

AnayGokhale, 2024-10-19:
Hello,
I am using JetPack 6.
I tried method number 2 however it still does not work. `rs-enumerate-devices` returned `No device detected. Is it plugged in?` and the realsense-viewer UI did not register it either. I have also tried connecting it to the Jetson Orin Nano through the USB-A port and it still does not work.
I cannot do method number 3 as it Jetpack 5.1.3 is not useful to me.
Is there any other solution?

MartyG-RealSense, 2024-10-20:
Other than building librealsense beta version 2.56.2 from source code (as official JetPack 6 support was added in version 2.56.1), I cannot think of another method to try unfortunately.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.2

Would you be able to continue using librealsense from with a Docker container if a solution is not achieved for outside-container use?

MartyG-RealSense, 2024-10-28:
Hi @AnayGokhale  Do you require further assistance with this case, please?  Thanks!

AnayGokhale, 2024-11-02:
@MartyG-RealSense Hello, yes I do still need help with this. I have a few questions/ideas.
As per my understanding, the instructions I am using create a ROS docker container. What happens if I put ROS on my Jetson so that I do not have to use a docker? Is this possible and do you think the camera would work?
I might just have to try the first out of the three methods you recommended.

MartyG-RealSense, 2024-11-02:
Yes, you can install the librealsense SDK and ROS wrapper on a Jetson board without using a Docker container.

You could install librealsense on Jetson from packages or from source code by using one of the **JetsonHacks** installation scripts at the link below.  Use 'installLibrealsense.sh' for a package installation or 'buildLibrealsense.sh' for a source code installation.

https://github.com/JetsonHacksNano/installLibrealsense

Using one of these JetsonHacks build scripts shold have the same results as using Intel's official installation instructions.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Then build the ROS wrapper from source code once you have successfully installed and tested librealsense on the Jetson without a container.

MartyG-RealSense, 2024-11-15:
Case closed due to no further comments received.

closed 2024-11-15 by MartyG-RealSense

AnayGokhale, 2024-11-25:
@MartyG-RealSense 
Just to end the thread, I ended up downgrading to Jetpack 5.1.3 as I realized it would work for me. The camera now works without a docker and starts through the python script as well. 
Thank You for you help!

MartyG-RealSense, 2024-11-26:
You are very welcome, @AnayGokhale - it's great to hear that you were successful.  I'm pleased that I could help.  Thanks very much for the update!