realsenseai/librealsense #13420: After several operations to get VideoFrame from FrameSet in PreviewActivity, GLSurfaceView is not updated.

yorushiba, 2024-10-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | Android  |
| SDK Version                            |  2.54.2                          | 
| Language                            |  Java                          | 
| Segment			|  Smartphone                          | 

### Issue Description
Perform the following processing in onFrameset in PreviewActivity.java.

1. clone() frameSet to frameSet_tmp
2. In a separate thread, retrieve the VideoFrame from frameSet_tmp
3. close() frameSet_tmp and VideoFrame

After the above process is performed several times, the preview will no longer be updated.

	 public void onFrameset(final FrameSet frameSet) {
	     mStreamingStats.onFrameset(frameSet);
	     mGLSurfaceView.upload(frameSet);

	     if (!processing) {
	         processing = true;

	         frameset_tmp = frameSet.clone();
	         new Thread(() -> {
	             try {
	                 VideoFrame colorFrame = frameset_tmp.first(StreamType.COLOR).as(Extension.VIDEO_FRAME);
	                 ...Some kind of processing
	                 colorFrame.close();
	             } catch (Exception e) {
	                 throw new RuntimeException(e);
	             }
	             frameset_tmp.close();
	             processing = false;
	         }).start();

The following logs are output:

	2024-10-11 16:36:43.522  5329-5428  librs                   com.intel.realsense.camera           D  CallbackFinished,Depth,#1097,@1728632203522.970947, callback duration: 0.583008 ms
	2024-10-11 16:36:43.540  5329-5431  librs                   com.intel.realsense.camera           D  User didn't release frame resource.
	2024-10-11 16:36:43.540  5329-5431  librs                   com.intel.realsense.camera           D  publish(...) failed
	2024-10-11 16:36:43.554  5329-5427  librs                   com.intel.realsense.camera           D  Passing packet to user CB with size 1843448
	2024-10-11 16:36:43.555  5329-5428  librs                   com.intel.realsense.camera           D  FrameAccepted,Depth,Counter,1098,Index,0,BackEndTS,0.000000,SystemTime,1728632203555.250000 ,diff_ts[Sys-BE],1728632203555.250000,TS,1728632203537.076172,TS_Domain,Global Time,last_frame_number,1097,last_timestamp,1728632203503.482666
	2024-10-11 16:36:43.555  5329-5428  librs                   com.intel.realsense.camera           D  --> syncing [Depth/12 #1098 @1728632203537.08/Global Time]
	2024-10-11 16:36:43.555  5329-5428  librs                   com.intel.realsense.camera           D  <-- [Depth/12 #1098 @1728632203537.08/Global Time]  (TS: Depth/12 Infrared/13 Infrared/14 Color/15)
	2024-10-11 16:36:43.555  5329-5428  librs                   com.intel.realsense.camera           D  ... have [Depth/12 #1098 @1728632203537.08/Global Time]
	2024-10-11 16:36:43.556  5329-5428  librs                   com.intel.realsense.camera           D  <-- [[Depth/12 #1098 @1728632203537.08/Global Time]]  (CI: (TS: Depth/12 Infrared/13 Infrared/14 Color/15))
	2024-10-11 16:36:43.556  5329-5428  librs                   com.intel.realsense.camera           D  <-- queueing [[Depth/12 #1098 @1728632203537.08/Global Time]]
	2024-10-11 16:36:43.556  5329-5428  librs                   com.intel.realsense.camera           D  --> frame ready: [[Depth/12 #1098 @1728632203537.08/Global Time]]
	2024-10-11 16:36:43.556  5329-5428  librs                   com.intel.realsense.camera           D  CallbackFinished,Depth,#1098,@1728632203556.270996, callback duration: 0.572998 ms
	2024-10-11 16:36:43.566  5329-5431  librs                   com.intel.realsense.camera           D  recoverable_exception: Out of frame resources!
	2024-10-11 16:36:43.567  5329-5431  librs                   com.intel.realsense.camera           E  Exception was thrown during callback: Out of frame resources!
	2024-10-11 16:36:43.567  5329-5431  librs                   com.intel.realsense.camera           D  CallbackFinished,Color,#1078,@1728632203567.325928, callback duration: 73.783936 ms
	2024-10-11 16:36:43.567  5329-5431  librs                   com.intel.realsense.camera           I  Frame Callback Color #1078 overdue. (FPS: 30, max duration: 32.2581 ms)
	2024-10-11 16:36:43.567  5329-5431  librs                   com.intel.realsense.camera           D  FrameAccepted,Color,Counter,1081,Index,0,BackEndTS,0.000000,SystemTime,1728632203567.396973 ,diff_ts[Sys-BE],1728632203567.396973,TS,1728632202899.132080,TS_Domain,Global Time,last_frame_number,1078,last_timestamp,1728632202798.350098
	2024-10-11 16:36:43.571  5329-5430  librs                   com.intel.realsense.camera           D  Passing packet to user CB with size 1843448
	2024-10-11 16:36:43.588  5329-5427  librs                   com.intel.realsense.camera           D  Passing packet to user CB with size 1843448
	2024-10-11 16:36:43.589  5329-5428  librs                   com.intel.realsense.camera           D  FrameAccepted,Depth,Counter,1099,Index,0,BackEndTS,0.000000,SystemTime,1728632203588.927979 ,diff_ts[Sys-BE],1728632203588.927979,TS,1728632203570.669678,TS_Domain,Global Time,last_frame_number,1098,last_timestamp,1728632203537.076172
	2024-10-11 16:36:43.589  5329-5428  librs                   com.intel.realsense.camera           D  --> syncing [Depth/12 #1099 @1728632203570.67/Global Time]
	2024-10-11 16:36:43.589  5329-5428  librs                   com.intel.realsense.camera           D  <-- [Depth/12 #1099 @1728632203570.67/Global Time]  (TS: Depth/12 Infrared/13 Infrared/14 Color/15)
	2024-10-11 16:36:43.589  5329-5428  librs                   com.intel.realsense.camera           D  ... have [Depth/12 #1099 @1728632203570.67/Global Time]
	2024-10-11 16:36:43.589  5329-5428  librs                   com.intel.realsense.camera           D  <-- [[Depth/12 #1099 @1728632203570.67/Global Time]]  (CI: (TS: Depth/12 Infrared/13 Infrared/14 Color/15))
	2024-10-11 16:36:43.589  5329-5428  librs                   com.intel.realsense.camera           D  <-- queueing [[Depth/12 #1099 @1728632203570.67/Global Time]]
	2024-10-11 16:36:43.589  5329-5428  librs                   com.intel.realsense.camera           D  --> frame ready: [[Depth/12 #1099 @1728632203570.67/Global Time]]

From the logs, it seems that the problem is caused by the Frame resources not being released. Is there any other processing required besides the close() process?

MartyG-RealSense, 2024-10-15:
Hi @yorushiba  Have you got a **mGLSurfaceView.clear();** instruction in your script on the line before the pipeline is started, please, like in the example code below.

```
 private synchronized void start() {
        if(mIsStreaming)
            return;
        try{
            Log.d(TAG, "try start streaming");
            mGLSurfaceView.clear();
            mPipeline.start(mConfig);
            mIsStreaming = true;
            mHandler.post(mStreaming);
            Log.d(TAG, "streaming started successfully");
        } catch (Exception e) {
            Log.d(TAG, "failed to start streaming");
        }
    }

```

yorushiba, 2024-10-15:
They are as follows. Nothing has been changed here.

	try {
	    mGLSurfaceView.clear();
	    mStreamer.start();
	}
	catch (Exception e){
	    if(mStreamer != null)
		mStreamer.stop();
	    Log.e(TAG, e.getMessage());
	    Toast.makeText(this, "Failed to set streaming configuration ", Toast.LENGTH_LONG).show();
	    Intent intent = new Intent(PreviewActivity.this, SettingsActivity.class);
	    startActivity(intent);
	}

MartyG-RealSense, 2024-10-15:
I have not seen the message **User didn't release frame resource** for years.  The last time that it occurred though, I advised that it seems to be generated when the number of published frames exceeds the maximum frame queue size.  

The other warning in your log **Out of frame resources!** would seem to be evidence of that, as it can occur when the frame queue becomes 'jammed up' with frames because - as you said - old frames are not being released and incoming new frames therefore cannot enter the frame queue.

By default the frame queue can hold 16 frames from each type of stream.  Increasing the frame queue size means that the queue can hold more frames simultaneously.  I do not know of a code example for implementing a frame queue size change in Java though (but it is possible to access the SDK's C++ functions through a 'native' Android application though.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/native_example/

yorushiba, 2024-10-22:
I solved the problem by getting the Frame and VideoFrame in stages and close() each one.
Thanks for the advice!

         frameset_tmp = frameSet.clone();
         new Thread(() -> {
             try {
                 Frame frame = frameset_tmp.first(StreamType.COLOR);
                 VideoFrame colorFrame_tmp = frame.as(Extension.VIDEO_FRAME);
                 ...Some kind of processing
                 colorFrame.close();
                 frame.close();
             } catch (Exception e) {
                 throw new RuntimeException(e);
             }
             frameset_tmp.close();

closed 2024-10-22 by yorushiba

MartyG-RealSense, 2024-10-22:
You are very welcome, @yorushiba - it's great to hear that you achieved a solution.  Thanks very much for the update and sharing the details of your solution!