realsenseai/librealsense #13421: RuntimeError: get_xu(ctrl=1) failed! Last Error: Device or resource busy

ravil99, 2024-10-15:
| System Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | [5.16.0.1]| 
| Operating System & Version |   Ubuntu 22.04.5 LTS  | 
| Kernel Version (Linux Only)    |  6.8.0-45-generic | 
| Platform                                 | PC  |
| SDK Version                            |  2.56.1                     | 
| Language                            | Python                         | 
| Segment   |  CV                         | 

### Issue Description
Hello everyone. I met a problem with using librealsense. He are having an experimental setup with 4 D415 cameras connected to one Ubuntu machine. Each camera is plugged in to separate USB port via 5 meter active cable. We are using v4l USB backend.

Our project is written in Python and initially we try to load our custom .json camera preset sequentially. Here is a code example:

`preset_path: str = "path_to_desired_preset"  # just example of a path
des_preset: dict = json.load(preset_path)
advanced_mode = rs.rs400_advanced_mode(device)
advanced_mode.load_json(json.dumps(des_preset))`

This code is run in loop to cover all 4 cameras.
But sometimes executing `advanced_mode.load_json(json.dumps(des_preset))` resuts in the following error:

`RuntimeError: get_xu(ctrl=1) failed! Last Error: Device or resource busy`.

Error text suggests, that something is trying to access the camera and some conflict occurs. But it's not the case - only my code is trying to access camera.

And behavior of this error is not stable. Usually it appears during loading of preset to first or second camera. And it takes 4-5 code restarts to load all 4 presets at the same time.

Originally this error occurred with presets generated with `"fw version": "05.12.07.150"`. Then presets were regenerated with `"fw version": "5.16.0.1"`, but it didn't fixed the error.

Surprisingly, this error doesn't appear if before running my project all cameras' Depth Stream was opened in `realsense-viewer`.

Do you know what does this `Runtime Error` means and how to fix it?

I will be glad to any suggestion.

MartyG-RealSense, 2024-10-28:
Hi @ravil99  Do you require further assistance with this case, please?  Thanks!

ravil99, 2024-10-29:
Hello!

I've stopped researching on this problem for a few week, but I will start working on it again today or tomorrow.

So several questions may arise a bit later. Please, don't close this issue now.

MartyG-RealSense, 2024-10-29:
That's no problem at all to keep the issue open.  I look forward to your next update.  Good luck!

ravil99, 2024-10-30:
Hi again!

We are having rs.stream.depth with 15 FPS and rs.stream.color with 6 FPS. The problem disapeared after I've equalized FPS by putting 6 FPS to all streams.

MartyG-RealSense, 2024-10-31:
It's great to hear that you resolved your problem.  Thanks very much for the update!

ravil99, 2024-11-09:
No, it's done! Thank you.

MartyG-RealSense, 2024-11-09:
You are very welcome.  :)  As you have achieved a solution, I will close this case.  Thanks again!

closed 2024-11-09 by MartyG-RealSense