realsenseai/librealsense #13423: Wrong HID Device Path Parsing Assumption

pasandal, 2024-10-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Firmware Version                   | 2.55.1.6486 |               | 

### Issue Description

- We have identified an issue with the `parse_usb_path_multiple_interface` and `parse_usb_path_from_device_id`  function in your code assuming that HID device paths must contain a `vid_` prefix
- Our device path uses `ven_` (e.g., `ven_elas&dev_e551&subsys_17aa233c`), which is valid but not recognized by your parsing logic.
- This leads to errors being logged and delays in the camera's initialization.

### Impact:

- Unnecessary error logging.
- Increased initialization time for the Realsense camera.
- Potential compatibility issues with other devices using different but valid device path formats.

### Recommendation:

- Modify the device path parsing logic to be more flexible and accommodate different valid prefixes like `ven_` and `dev_`.
- Avoid hardcoding the expectation of `vid_` and `pid_` in the device path.

Link to code: https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/src/win/win-helpers.cpp#L106

MartyG-RealSense, 2024-10-15:
Hi @pasandal  You are welcome to modify the source code of the librealsense SDK and then build the SDK from the modified source code.

**Linux**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

**Windows**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

pasandal, 2024-10-15:
Hi @MartyG-RealSense ,

Thank you for your response. I understand that modifying the SDK locally is an option, but I wanted to clarify if this issue could be addressed in the official SDK itself.  Given that this affects initialization time and compatibility with valid device path formats like `ven_`, it seems like it could impact other users too.

MartyG-RealSense, 2024-10-15:
If yourself or another RealSense user is able to develop a solution then we would welcome it being submitted as a **Pull Request (PR)** for consideration by the RealSense team for inclusion in the official SDK.

https://github.com/IntelRealSense/librealsense/pulls

MartyG-RealSense, 2024-10-28:
Hi @pasandal  Bearing in mind the recommendation in the above comment that a solution should be developed and submitted for inclusion in the SDK by a RealSense user, do you require further assistance with this case please?  Thanks!

closed 2024-11-04 by MartyG-RealSense