realsenseai/librealsense #13424: Does anyone have an older version of the SDK (Intel RealSense SDK 2016 R2)?

cchengtien, 2024-10-15:
I tried the official link "http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe" but it doesn't work anymore and always fails during the download process
I want to use it to drive SR300 camera in Windows 10

MartyG-RealSense, 2024-10-15:
Hi @cchengtien  If the 2016 R2 link is not working for you then you can use the much more modern RealSense SDK 2.50.0 which dates from November 2021.  2.50.0 is the final version of this modern SDK that fully supports SR300.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

You can obtain the full SDK for Windows 10 by downloading and running the file **Intel.RealSense.SDK-WIN10** from the 'Assets' file list, or simply download and run the RealSense Viewer tool (**Intel.RealSense.Viewer.exe**) if you just want to access the camera's streams without doing any programming.

[image]

cchengtien, 2024-10-17:
In fact, I wanted to use its old internal 3D scanning feature and see how it worked.

MartyG-RealSense, 2024-10-17:
At my location at least, it is possible to launch the 2016 R2 download if you click the link to begin the download and then refresh the web-page that the download is taking place on.  The download should then begin.

MartyG-RealSense, 2024-10-17:
Whilst the original download page for the 2016 SDK is no longer online, an archived version of it is available on the Wayback Machine website.  In the 'Downloads' section at the bottom of the page there are download links for R2 and also the R3 version.

https://web.archive.org/web/20180719232831/https://software.intel.com/en-us/realsense-sdk-windows-eol

With the R3 version, instead of a 1.8 gb download it is a set of smaller downloads, with a core Essentials module that you install first and then optional modules, such as 3D Scan, so you can install only the features you need.  

The download links require the filling in of a form first before the download begins and I do not know if all the download links still work.

cchengtien, 2024-10-17:
Great! it works
I really appreciate your help

MartyG-RealSense, 2024-10-17:
You are very welcome.  It's great to hear that it worked for you.  :)

closed 2024-10-17 by cchengtien

Lapr87, 2025-11-25:
Here

https://web.archive.org/web/20210906081509/http://registrationcenter-download.intel.com/akdlm/irc_nas/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe