realsenseai/librealsense #13425: realsense viewer error

minseo104, 2024-10-15:
| | |
|---|---|
|**librealsense**|2.56.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25601, json_settings:{):
    failed to convert special folder: errno=42

```

MartyG-RealSense, 2024-10-16:
Hi @minseo104  Advice about the **errno=42** error can be found at https://github.com/IntelRealSense/librealsense/issues/13037

closed 2024-11-15 by MartyG-RealSense

MartyG-RealSense, 2024-12-04:
My Intel RealSense colleagues have provided some further advice about the errno=42 error.

Given that the problem seems to be related to certain charcters being used in a Windows user account name, I asked whether creating a secondary Windows user account with a name that has no special characters might help.  This was their response. 

You can try to run it as Administrator instead of switching the account.  On Windows 11, it also has a feature to run the app as a different user.  That option can be enabled with the instructions at the link below:
 
https://winaero.com/how-to-run-apps-as-different-user-in-windows-11/

*******************

我的英特尔 RealSense 同事提供了一些有关 errno=42 错误的进一步建议。

鉴于该问题似乎与 Windows 用户帐户名称中使用的某些字符有关，我询问创建一个名称中没有特殊字符的辅助 Windows 用户帐户是否有帮助。这是他们的回复。

您可以尝试以管理员身份运行它，而不是切换帐户。在 Windows 11 上，它还具有以其他用户身份运行应用程序的功能。可以按照以下链接中的说明启用该选项：

https://winaero.com/how-to-run-apps-as-different-user-in-windows-11/