realsenseai/librealsense #13427: Problem

Lynnn8, 2024-10-16:
| | |
|---|---|
|**librealsense**|2.25.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

MartyG-RealSense, 2024-10-16:
Hi @Lynnn8  Can you provide information about your problem with version 2.25.0 please?

This version is very old (August 2019), and modern camera firmware driver versions will not be compatible with it.  I would therefore not recommend using it.

Lynnn8, 2024-10-16:
May I ask where I should download the appropriate software? I followed the csdn blog to download it before. Now you can connect to the device but there is no image display, the screen is as follows
[image]
[image]

MartyG-RealSense, 2024-10-16:
Which RealSense camera model do you have, please? Some models are not supported in older RealSense software versions depending on when a model was released and so will not be detected in those old versions.

For anything in the RealSense 400 Series (D415, D435, D455, etc) I recommend installing version 2.55.1.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

If you have an L515 camera model, please use 2.51.1.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

If you have an SR300 camera model, please use 2.50.0.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

********

The camera's firmware driver of RealSense 400 Series cameras should also be updated to have the correct firmware version for a particular RealSense software version.

For 2.55.1: firmware 5.16.0.1
For 2.50.0 or 2.51.1: firmware 5.13.0.50

closed 2024-11-15 by MartyG-RealSense