realsenseai/librealsense #13436: Provide deb package source

ethanc8, 2024-10-18:
Hi, can you please provide the sources which are used to build the Debian packages, including the Debian metadata such as the `control` and `rules` files? Then it would be easier to build DKMS packages for Debian-based distributions other than Canonical b

MartyG-RealSense, 2024-10-18:
Hi @ethanc8  The source code version of librealsense includes a **package.xml** file for building DKMS packages from the source.  It is in the main 'root' folder of the source code.

https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/package.xml

[image]

The source can be downloaded as a zipped file from the 'Assets' file list at the bottom of the release information for each librealsense version.

https://github.com/intelrealsense/librealsense/releases

[image]

If possible though I would recommend building librealsense from source code with CMake with the **libuvc backend** or the  *RSUSB backend* installation methods, as these install methods bypass the Linux kernel and so are not dependent on a particular Linux type or kernel version and do not require the application of a patch script to the kernel.

ethanc8, 2024-10-18:
If I want to build deb packages, which took would I use to build the deb packages using the package.xml file?

MartyG-RealSense, 2024-10-18:
libuvc backend was the predecessor to RSUSB backend, but either method works and they both perform the same function of bypassing the kernel and so allowing any flavor of Linux to be used with a RealSense camera.

The libuvc backend method is easier to perform than RSUSB backend as it has a pre-made build script called 'libuvc_installation.sh'

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

*************

I will consult with my Intel RealSense colleagues regarding how the package.xml file is used to build a Debian package.

ethanc8, 2024-10-18:
Ok. For now I will try to work with the RSUSB backend. However, it would be good to know more about what each of the kernel patches do and whether they rely on any of the patches Canonical applied to the mainline kernel to make the Ubuntu kernel, as knowing that would allow using the patches on other kernels (I'm specifically interested in the [RK3588 BSP based on kernel 6.1](https://github.com/Joshua-Riek/linux-rockchip/tree/noble)).

MartyG-RealSense, 2024-10-18:
For installing librealsense on an RK (Rockchip) device it is recommended to build from source with libuvc or RSUSB backend.

ethanc8, 2024-10-18:
Ok, thanks.

MartyG-RealSense, 2024-10-28:
Hi @ethanc8  Do you require further assistance with this case, please?  Thanks!

ethanc8, 2024-10-28:
@MartyG-RealSense I don't need more assistance at this time, but I need more time in order to attempt the librealsense build (I have been quite busy recently).

MartyG-RealSense, 2024-10-29:
It's no problem at all.  I look forward to your next update when you are able to perform the build.  Good luck!

MartyG-RealSense, 2024-11-23:
Case closed due to no further comments received.  Please feel free to re-open it when you are ready to proceed.  Thanks!

closed 2024-11-23 by MartyG-RealSense