realsenseai/librealsense #13437: pipeline_profile = config.resolve(pipeline_wrapper) RuntimeError: No device connected on NVIDIA Jetson AGX ORIN 64G DevKit device realsense-viewer showing RGB feed

monajalal, 2024-10-18:
Please note I am trying to show side-by-side feed of ORBBEC Astra Embedded S and Intel RealSense D435 camera (both just color), I have used different name for everything that involves orbbec started with orbbec_

```
pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected
```

^^^

Also, realsense-viewer shows RGB feed and orbbec color-viewer.py runs successfully separately.

When I test rs camera:

```
$ cat test_rs_camera.py 
import pyrealsense2 as rs
pipe = rs.pipeline()
print(pipe)
print(dir(pipe))
pipe.start()
print(pipe.get_active_profile())
pipe.stop()
```

I get no error

```
$ python test_rs_camera.py 
<pyrealsense2.pipeline object at 0xffffbd0225b0>
['__class__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'get_active_profile', 'poll_for_frames', 'start', 'stop', 'try_wait_for_frames', 'wait_for_frames']
<pyrealsense2.pipeline_profile object at 0xffffbd0585b0>

```

Could you please help how to fix this issue?

```
$ python
Python 3.10.12 (main, Sep 11 2024, 15:47:36) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2
>>> pyrealsense2.__version__
'2.50.13'
```

```
mona@ubuntu:~$ uname -a
Linux ubuntu 5.15.136-tegra #1 SMP PREEMPT Mon May 6 09:56:39 PDT 2024 aarch64 aarch64 aarch64 GNU/Linux
mona@ubuntu:~$ lsb_release -a
LSB Version:    core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy
```

This is the full log:

```
[10/18 11:55:54.188790][info][7394][Context.cpp:68] Context created with config: default config!
[10/18 11:55:54.188856][info][7394][Context.cpp:73] Work directory=mydir, SDK version=v1.10.12-20240809-69b687f
[10/18 11:55:54.188900][info][7394][LinuxPal.cpp:32] createObPal: create LinuxPal!
[10/18 11:55:54.266899][warning][7394][OpenNIDeviceInfo.cpp:186] New openni device matched.
[10/18 11:55:54.267080][info][7394][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[10/18 11:55:54.267167][info][7394][DeviceManager.cpp:15] Current found device(s): (1)
[10/18 11:55:54.267187][info][7394][DeviceManager.cpp:24] 	- Name: SL1000S_U3, PID: 0x060b, SN/ID: , Connection: USB2.0
[10/18 11:55:54.267286][info][7394][Pipeline.cpp:15] Try to create pipeline with default device.
[10/18 11:55:54.269489][info][7394][OpenNIHostProtocol.cpp:567] Hardware versions: FW=5.8.23 (14), HW=0, Chip=6, Sensor=0, SYS=12
[10/18 11:55:54.271194][error][7394][OpenNIHostProtocol.cpp:584] Get usb core type failed!
[10/18 11:55:54.281078][info][7394][OpenNISensorFirmware.cpp:1185] Sensor serial number:AQAAC830368
[10/18 11:55:54.284591][info][7394][OpenNISensorFirmware.cpp:1213] Firmware version RD3010
[10/18 11:55:54.286069][info][7394][OpenNISensorFirmware.cpp:1219] Device frequency 50
[10/18 11:55:54.337552][error][7394][OpenNIHostProtocol.cpp:957] Host Protocol sub cmd not supported!
[10/18 11:55:54.337596][warning][7394][OpenNISensorFirmware.cpp:153] OpenNI2 camera don't support Watchdog function!
[10/18 11:55:54.337845][info][7394][AbstractDevice.cpp:121] 	- Firmware version: RD3010
[10/18 11:55:54.338172][info][7394][OpenNIDevice.cpp:37] OpenNI device created! PID: 0x060b, SN: AQAAC830368
[10/18 11:55:54.338215][info][7394][DeviceManager.cpp:150] Device created successfully! Name: SL1000S_U3, PID: 0x060b, SN/ID: 
[10/18 11:55:54.338278][warning][7394][ObException.cpp:5] Current device does not support frame sync!
[10/18 11:55:54.338448][warning][7394][Pipeline.cpp:45] Execute failure! A libobsensor_exception has occurred!
	 - where:45#Pipeline
	 - msg:Current device does not support frame sync!
	 - type:N11libobsensor31unsupported_operation_exceptionE
[10/18 11:55:54.338498][info][7394][Pipeline.cpp:47] Pipeline created with device: {name: SL1000S_U3, sn: AQAAC830368}, @0xAAAAF120D6B0
[10/18 11:55:54.398090][info][7394][OpenNIDevice.cpp:460] Color sensor has been created!
[10/18 11:55:54.398274][info][7394][Pipeline.cpp:188] Check and set config start!
[10/18 11:55:54.398307][info][7394][Pipeline.cpp:223] Check and set config done!
[10/18 11:55:54.398846][info][7394][OpenNIDevice.cpp:374] Depth sensor has been created!
[10/18 11:55:54.402416][info][7394][Pipeline.cpp:288] Try to start streams!
[10/18 11:55:54.402530][info][7394][VideoSensor.cpp:646] start OB_SENSOR_COLOR stream with profile: {type: OB_STREAM_COLOR, format: OB_FORMAT_RGB888, width: 640, height: 480, fps: 30}, Backend Format: OB_FORMAT_MJPG
[10/18 11:55:54.919431][info][7394][Pipeline.cpp:301] Start streams done!
[10/18 11:55:54.919488][info][7394][Pipeline.cpp:277] Pipeline start done!
[10/18 11:55:55.364816][warning][7453][Pipeline.cpp:341] Pipeline source frameset queue fulled, drop the oldest frame!
Traceback (most recent call last):
  File "myfile.py", line 179, in <module>
    pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected
[10/18 11:55:56.075824][info][7394][Pipeline.cpp:411] Try to stop pipeline!
[10/18 11:55:56.075948][info][7394][Pipeline.cpp:383] Try to stop streams!
[10/18 11:55:56.092993][info][7394][VideoSensor.cpp:734] Video sensor stopped, @OB_SENSOR_COLOR
[10/18 11:55:56.093013][info][7394][Pipeline.cpp:396] Sensor stream stopped, sensorType=OB_SENSOR_COLOR
[10/18 11:55:56.093021][info][7394][Pipeline.cpp:402] Stop streams done!
[10/18 11:55:56.094654][info][7394][Pipeline.cpp:443] Stop pipeline done!
[10/18 11:55:56.094678][info][7394][Pipeline.cpp:75] Pipeline destroyed! @0xAAAAF120D6B0
[10/18 11:55:56.094701][info][7394][OpenNIDevice.cpp:41] OpenNI device destroyed! PID: 0x060b, SN: AQAAC830368
[10/18 11:55:56.094721][info][7394][VideoSensor.cpp:303] VideoSensor destroyed, @OB_SENSOR_COLOR
[10/18 11:55:56.094736][info][7394][OpenNIVideoSensor.cpp:1014] OpenNIVideoSensor destroyed! sensorType=OB_SENSOR_DEPTH
[10/18 11:55:56.133929][info][7394][ObLibuvcDevicePort.cpp:70] uvc_close done.
[10/18 11:55:56.133980][info][7394][ObLibuvcDevicePort.cpp:71] ~ObLibuvcDevicePort done
[10/18 11:55:56.135605][info][7394][Context.cpp:84] Context destroyed
[10/18 11:55:56.135941][warning][7456][Pipeline.cpp:341] Pipeline source frameset queue fulled, drop the oldest frame! [**35 logs in 771ms, last: 11:55:56.067408**]

```

```
from pyorbbecsdk import Config
from pyorbbecsdk import OBError
from pyorbbecsdk import OBSensorType, OBFormat
from pyorbbecsdk import Pipeline, FrameSet
from pyorbbecsdk import VideoStreamProfile

pipeline = rs.pipeline()
config = rs.config()

orbbec_config = Config()
orbbec_pipeline = Pipeline()

# orbbec profile setup

try:
    orbbec_profile_list = orbbec_pipeline.get_stream_profile_list(OBSensorType.COLOR_SENSOR)
    try:
        orbbec_color_profile: VideoStreamProfile = orbbec_profile_list.get_video_stream_profile(640, 0, OBFormat.RGB, 30)
    except OBError as e:
        print(e)
        orbbec_color_profile = orbbec_profile_list.get_default_video_stream_profile()
        print("color profile: ", orbbec_color_profile)
    orbbec_config.enable_stream(orbbec_color_profile)
except Exception as e:
    print(e)

orbbec_pipeline.start(orbbec_config)

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

config.enable_stream(rs.stream.color, 1920, 1080, rs.format.bgr8, 30)

pipeline.start(config)

finally:
    pipeline.stop()
    orbbbec_pipeline.stop()
    cv2.destroyAllWindows()
```

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

D435                | 

### Issue Description
<Describe your issue / question / feature request / etc..>

monajalal, 2024-10-18:
this might not be a solution but somehow works

disconnected all usb devices, checked to see if realsense is connected via realsense-viewer and then commented this part and now it works

```
# ctx = rs.context()
# devices = ctx.query_devices()
# for dev in devices:
#     dev.hardware_reset()

``` 

please let me know what your workarounds are for resolving this issue if it keeps happened

MartyG-RealSense, 2024-10-19:
An alternative to resetting the camera is to reset the entire USB port, which does not require the camera to be detected.  

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/8274#issuecomment-770191698 had success using a Python reset script called PyUSB and said that they were able to reset the camera with it as often as they wanted.

A USB port can also be reset in Linux using a 'shell' (.sh) reset script, like the **usbreset** one at https://github.com/themancalledjakob/usbreset

MartyG-RealSense, 2024-10-28:
Hi @monajalal  Do you require further assistance with this case, please?  Thanks!

monajalal, 2024-10-29:
No for us unplugging and plugging back the camera worked.

closed 2024-10-29 by monajalal

MartyG-RealSense, 2024-10-29:
Okay, thanks very much for the update!