realsenseai/librealsense #13439: Inquiry Regarding License for Publishing Built pyrealsense2 for Python 3.12

Johnny-dev-01, 2024-10-19:
Hello,

I recently built the pyrealsense2 library from source using CMake, as there was no pre-built version available for Python 3.12. However, I have some concerns regarding the licensing and whether I am permitted to publish my built version on GitHub.

Could you please clarify if the license allows for the distribution of modified or built versions of the library? I want to ensure that I comply with any licensing requirements before sharing my work with the community.

Thank you for your assistance!

Best regards,

Johnny-Dev

MartyG-RealSense, 2024-10-19:
Hi @Johnny-dev-01  The pyrealsense2 wrapper is distributed under an **Apache License**.

https://en.wikipedia.org/wiki/Apache_License

The license "allows users to use the software for any purpose, to distribute it, to modify it, and to distribute modified versions of the software under the terms of the license, without concern for royalties".

The wrapper's Apache license is confirmed here:

https://pypi.org/project/pyrealsense2/

[image]

MartyG-RealSense, 2024-10-28:
Hi @Johnny-dev-01  Do you require further assistance with this case, please?  Thanks!

Johnny-dev-01, 2024-10-29:
> Hi @Johnny-dev-01 Do you require further assistance with this case, please? Thanks!

Hi,

Thank you very much for following up. Currently, I don't have any issues and don't need further assistance. If I have any questions or problems in the future, I will definitely let you know.

Thanks a lot!

MartyG-RealSense, 2024-10-29:
You are very welcome!  As you do not need further assistance, I will close this case.  Thanks again!

closed 2024-10-29 by MartyG-RealSense