realsenseai/librealsense #13450: Pyrealsense2 with python 3.12 Realsense D435

bavleym, 2024-10-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (5.16.0.1) | 
| Operating System & Version |   {Win 11}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC -> NVIDIA Jetson  |
| SDK Version                            |  {2.56.2.7411}                          | 
| Language                            |  {python}                          | 
| Segment			|  {Appliction }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello, 

I am using the latest version of VS Code with anaconda as my python 3.12 base.  I want to to be able to use pyrealsense2 but it won't let me pip install it

PS C:\Users\Bavley> pip install pyrealsense2  
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2

I have tried following instructions from other issue replies such as using cmake to build it but VS Code still does not recognize that pyrealsense is installed. Is there anyway to use it with python 3.12?

MartyG-RealSense, 2024-10-22:
Hi @bavleym  SDK version 2.56.2 is a beta version that does not have a .whl package for installing pyrealsense2 from packages with `pip install pyrealsense2`.  Also, the pip install method does not usually work with Jetson boards.

You can install the pyrealsense2 wrapper for Python 3.12 on PC and Jetson if the wrapper is built from source code instead.

The easiest way to do this on Windows may be to build the SDK from source code using the instructions at the link below and tick the **BUILD_PYTHON_BINDINGS** box on the CMake graphical interface so that the pyrelsense2 wrapper is built from source at the same time as the SDK.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

bavleym, 2024-10-22:
Hi @MartyG-RealSense  thanks for your response. Those instructions are unclear, where do I open the cmake file in order to be able to tick the BUILDs? I have the realsense sdk file downloaded but I do not know what file to open to reach that first window.

MartyG-RealSense, 2024-10-23:
The graphical CMake tool shown in the instructions where you can tick flags is called **cmake-gui** (gui is 'graphical user interface).  

If you go to the CMake downloads page at the link below and then go to its **Binary Distributions** section, if you download and extract the 'Windows x64 ZIP' file then you will have a runnable program called **cmake-gui.exe**

https://cmake.org/download/

Further information about configuring the launch of the tool can be found here:

https://cmake.org/cmake/help/latest/manual/cmake-gui.1.html#manual:cmake-gui(1)

bavleym, 2024-10-23:
Hey @MartyG-RealSense , my fault I should've been more clear. I have the Cmake gui installed already, but I can not figure out how to open the realsense source code on the gui, or have all the options for building show up. I tried looking through the source code I downloaded and there is no .cmake file. Is there a more clear process you can provide for m to build the wrapper from source code.

MartyG-RealSense, 2024-10-23:
There is not an alternative Windows source code installation guide available, unfortunately.

However, if your problem is not being able to open the RealSense source code by selecting it with the 'Browse Source' button on the cmake-gui interface then I wonder if you have yet downloaded the source as a zip file and extracted the source code folder within it.

This is done by the following procedure.

1.  Visit the librealsense **Releases** page at the link below.

https://github.com/intelrealsense/librealsense/releases

2.  At the bottom of the information listing for each librealsense version there is an **Assets** file list.  Click on 'Assets' if the list of files is not shown.

3.  Click on the 'Source code (zip' file to start a download of the zipped source code in your web browser and then extract the contents of the zip file once download has completed to obtain the librealsense source code folder inside.

[image]

4.  Use cmake-gui's 'Browse Source' button to select the 'librealsense' source code folder that you extracted from the zip.  The path to it should then be displayed on the cmake-gui interface.

5.  In regard to what to select for the 'Where to build the binaries' selector of cmake-gui: judging by the example image of the interface in the instructions, I believe that you should open the librealsense folder with the Windows file navigator (e.g This PC) and you should see a list of folders and files like the image below.

[image]

6.  In that directory, create a new empty folder called **build** and then select this 'librealsense > build path using the 'Browse Build' button on the cmake-gui interface.

7.  After configuring the tick boxes if they are now listed, click the 'Generate' button at the bottom of the cmake-gui window to build librealsense.

bavley-mobarak, 2024-11-04:
Hi Marty, 

I have followed the steps you provided for me and it it appears everything may have installed correctly. My question now is what do I do with this.....how do I use pyrealsense2 with VSCode? Do i need to pip install or how would I point something towards this installation?

MartyG-RealSense, 2024-11-04:
As far as I am aware there is not a special procedure for using pyrealsense2 in VSCode, and you just write a pyrealsense2 script in the VSCode interface in the same way that you would write an ordinary Python script - please see https://github.com/IntelRealSense/librealsense/issues/10859 for an example of this.

[image]

MartyG-RealSense, 2024-11-15:
Hi @bavleym  Do you require further assistance with this case, please?  Thanks!

bavley-mobarak, 2024-11-17:
Hi Marty, 

Yes heres where I am and have done till now.

1. Downloaded the latest release SDK source code zip file, and extracted it.
2. Ran the CMake Gui at this source C:/Users/Gaiaf/OneDrive - CSULB/Documents/CECS 490 B/librealsense-2.56.2
3. Set the "Where to build binaries:" location as C:/Users/Gaiaf/OneDrive - CSULB/Documents/CECS 490 B/librealsense-2.56.2/build
4. Pressed "Configure" and selected VIsual Studio 17 2022 as my current generator
5.  [image]
6. From the available options I followed the steps to install the wrapper from source for windows [here](https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python#building-from-source)
7. I then click generate and run into the issue of 
[image]

Here is the terminal output from cmake: 

>

>  CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
>   Compatibility with CMake < 3.10 will be removed from a future version of
>   CMake.
> 
>   Update the VERSION argument <min> value or use a ...<max> suffix to tell
>   CMake that the project does not need compatibility with older versions.
> 
> 
>   Selecting Windows SDK version 10.0.26100.0 to target Windows 10.0.22631.
>   Checking internet connection...
>   Internet connection identified
>   Info: REALSENSE_VERSION_STRING=2.56.2
>   Setting Windows configurations
>   CMake Deprecation Warning at CMake/windows_config.cmake:2 (cmake_minimum_required):
>   Compatibility with CMake < 3.10 will be removed from a future version of
>   CMake.
> 
>   Update the VERSION argument <min> value or use a ...<max> suffix to tell
>   CMake that the project does not need compatibility with older versions.
>   Call Stack (most recent call first):
>   CMake/include_os.cmake:4 (include)
>   CMakeLists.txt:37 (include)
> 
> 
>   libusb not found; using internal version (missing: LIBUSB_LIB LIBUSB_INC) 
>   CMake Deprecation Warning at CMake/external_pybind11.cmake:1 (cmake_minimum_required):
>   Compatibility with CMake < 3.10 will be removed from a future version of
>   CMake.
> 
>   Update the VERSION argument <min> value or use a ...<max> suffix to tell
>   CMake that the project does not need compatibility with older versions.
>   Call Stack (most recent call first):
>   CMake/global_config.cmake:68 (include)
>   CMakeLists.txt:45 (global_set_flags)
> 
> 
>   Fetching pybind11...
>   CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
>   Compatibility with CMake < 3.10 will be removed from a future version of
>   CMake.
> 
>   Update the VERSION argument <min> value or use a ...<max> suffix to tell
>   CMake that the project does not need compatibility with older versions.
> 
> 
> 
>   CMake Deprecation Warning at build/third-party/pybind11/CMakeLists.txt:8 (cmake_minimum_required):
>   Compatibility with CMake < 3.10 will be removed from a future version of
>   CMake.
> 
>   Update the VERSION argument <min> value or use a ...<max> suffix to tell
>   CMake that the project does not need compatibility with older versions.
> 
> 
>   pybind11 v2.10.3 
>   CMake Warning (dev) at build/third-party/pybind11/tools/FindPythonLibsNew.cmake:98 (find_package):
>   Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
>   are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
>   the cmake_policy command to set the policy and suppress this warning.
> 
>   Call Stack (most recent call first):
>   build/third-party/pybind11/tools/pybind11Tools.cmake:50 (find_package)
>   build/third-party/pybind11/tools/pybind11Common.cmake:180 (include)
>   build/third-party/pybind11/CMakeLists.txt:208 (include)
>   This warning is for project developers.  Use -Wno-dev to suppress it.
> 
>   CMake Error at build/third-party/pybind11/tools/FindPythonLibsNew.cmake:242 (message):
>   Python libraries not found
>   Call Stack (most recent call first):
>   build/third-party/pybind11/tools/pybind11Tools.cmake:50 (find_package)
>   build/third-party/pybind11/tools/pybind11Common.cmake:180 (include)
>   build/third-party/pybind11/CMakeLists.txt:208 (include)
> 
> 
>   Configuring incomplete, errors occurred!
> 
>

MartyG-RealSense, 2024-11-17:
Does the same compilation issue if using the source code of librealsense 2.55.1 instead of 2.56.2?  Although official Python 3.12 support was introduced in 2.56.1, the pyrealsense2 wrapper should still be compatible with Python 3.12 when it is built from source code in librealsense 2.55.1.

bavley-mobarak, 2024-11-18:
So first question, would I need to build from source for 2,55.1 when there is already an executable for the SDK [here](https://github.com/IntelRealSense/librealsense/releases/download/v2.55.1) and after installing it, it provides the wrappers for python 
[image]

But to answer the question, yes I still generate compilation issues and not too sure what to do next. 
[image]

MartyG-RealSense, 2024-11-18:
The **cp37-win** Python wrapper component provided by the RealSense SDK installer file for Windows (Intel.RealSense.SDK-WIN10) only supports Python 3.7, not 3.12.

In regard to the 'project files may be invalid' error, a detailed discussion at https://github.com/IntelRealSense/librealsense/issues/1948 about building the SDK with cmake-gui on Windows might provide some useful insights if you have not seen it already.

bavley-mobarak, 2024-11-20:
Ok so i figured out the issue, in the Cmake.txt file included with the SDK source i needed to change the python paths so I added this to the top of the txt files

set(Python3_ROOT_DIR "C:/Users/Gaiaf/anaconda3")
set(Python3_EXECUTABLE "C:/Users/Gaiaf/anaconda3/python.exe")
set(Python3_INCLUDE_DIR "C:/Users/Gaiaf/anaconda3/include")
set(Python3_LIBRARY "C:/Users/Gaiaf/anaconda3/libs/python312.lib")  # Adjust 'python312.lib' to match your Python version

Secondly I had some other version of python interfering with my VS Code so I had to go through the properties on VS code and make sure that all my python paths pointed to the anaconda installation. Everything moved forward smoothly from there

MartyG-RealSense, 2024-11-20:
It's great to hear that you were successful.  Thanks very much for the update and for sharing the details of your solution!

MartyG-RealSense, 2024-11-25:
Hi @bavley-mobarak  After achieving a working solution, do you require further assistance with this case please?  Thanks!

closed 2024-12-01 by MartyG-RealSense