realsenseai/librealsense #13453: Jetson Nano pyrelasense has not attribute pipeline

harshal-14, 2024-10-22:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|213622072204|
|**Firmware Version**|5.15.1.55|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|218523067486|
|**Firmware Update Id**|218523067486|
|**Dfu Device Path**||

Hi,
I built everything based on this method
> https://github.com/IntelRealSense/librealsense/issues/13445

, however I cannot find the pyrealsense to work with even a basic python script.

```
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)

try:
    while True:
        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        # Stack both images horizontally
        images = np.hstack((color_image, depth_colormap))

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)

finally:
    # Stop streaming
    pipeline.stop()
    ```
 I get this error
> AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

Upon digging I found this is a common issue on Jetson devices https://github.com/IntelRealSense/librealsense/issues/13080 but I don't find the *.so files for pyrealsense. What am I doing wrong?

MartyG-RealSense, 2024-10-23:
Hi @harshal-14  When pyrealsense2 is built from source code, after building of the wrapper is completed the .so files will usually be found in the **librealsense > build > wrappers > python** directory of the librealsense source code folder.

I note that you used the **libuvc backend** instructions, which use the 'libuvc_installation.sh' build script.  This script downloads the librealsense source code and places a folder called **librealsense-master** on the computer.  

If you modified line 46 of the libuvc script to include the flag `-DBUILD_PYTHON_BINDINGS:bool=true` so that pyrealsense2 is built from source code at the same time as librealsense, then I would recommend finding the 'librealsense-master' folder on your computer and then navigating to the sub-folder location 'build > wrappers > python' to see whether the .so files are there.

harshal-14, 2024-10-23:
[image]
Hey @MartyG-RealSense,
I don't see any .so files in here.

MartyG-RealSense, 2024-10-23:
Thanks for checking the wrappers > python folder.

For some RealSense users in the past, amending their 'import pyrealsense2 as rs' line to the one below has resolved the **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** error.

`import pyrealsense2.pyrealsense2 as rs`

harshal-14, 2024-10-23:
Hi Marty, thanks for your prompt reply, however this doesn't solve the problem.
>     import pyrealsense2.pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'

MartyG-RealSense, 2024-10-23:
Usually, installing the pyrealsense2 wrapper from packages instead of source code with the command **pip install pyrealsense2** does not work on Jetson boards even though pip install of the wrapper supports devices with Arm architecture like Jetson.  It would be worth trying just to eliminate the possibility of whether it will work or not though.

harshal-14, 2024-10-23:
If I installed through pip then it shows even if device is connected and detected by realsense-viewer, lsusb it gives the following error:
>     pipeline_profile = config.resolve(pipeline_wrapper)
>     RuntimeError: No device connected

MartyG-RealSense, 2024-10-24:
Is the camera also not detected if the test script below is used?

```
# First import the library
import pyrealsense2 as rs

# Create a context object. This object owns the handles to all connected realsense devices
pipeline = rs.pipeline()
pipeline.start()

try:
    while True:
        # Create a pipeline object. This object configures the streaming camera and owns it's handle
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        if not depth: continue

        # Print a simple text-based representation of the image, by breaking it into 10x20 pixel regions and approximating the coverage of pixels within one meter
        coverage = [0]*64
        for y in range(480):
            for x in range(640):
                dist = depth.get_distance(x, y)
                if 0 < dist and dist < 1:
                    coverage[x//10] += 1

            if y%20 is 19:
                line = ""
                for c in coverage:
                    line += " .:nhBXWW"[c//25]
                coverage = [0]*64
                print(line)

finally:
    pipeline.stop()
```

harshal-14, 2024-10-24:
If I installed doing 
> pip install pyrealsense2

 then it shows camera not detected. But if I uninstall this one, and then try it shows 
> pyrelasense has not attribute pipeline

MartyG-RealSense, 2024-10-24:
Having no .so files in the wrapper build folder makes it difficult to achieve a solution, unfortunately.

An installation method that has not apparently been tried yet is to use the librealsense + pyrealsense2 source code procedure at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 which was designed specifically for Jeton Nano.  

As the instructions were written a while ago, step 7 refers to Python 3.6 instead of a recent Python version such as 3.12.  If you only have one Python version installed on your computer though, you should be able to skip step 7 as setting the PYTHONPATH parameter would be unnecessary.  The PYTHONPATH is to tell the computer which Python version to use if, for example, you had Python 2 and Python 3 installed on the same computer.

MartyG-RealSense, 2024-11-04:
Hi @harshal-14   Do you require further assistance with this case, please?  Thanks!

harshal-14, 2024-11-07:
Yes I tried all above steps, but nothing seems to work. Do you have any other solution in mind?

MartyG-RealSense, 2024-11-08:
You previously succeded in building the librealsense SDK with the 'libuvc backend' procedure that you described at https://github.com/IntelRealSense/librealsense/issues/13445#issuecomment-2425488326

When editing line 46 of the libuvc_installation.sh scrpt, where you used the following CMake build instruction:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA=on -DFORCE_RSUSB_BACKEND=TRUE`

Have you tried adding the **-DBUILD_PYTHON_BINDINGS:bool=true** command for building the Python wrapper at the same time that librealsense is built?

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA=on -DBUILD_PYTHON_BINDINGS:bool=true`

You do not need to use the ** -DFORCE_RSUSB_BACKEND=TRUE** flag if you have already included  -DFORCE_LIBUVC=true

closed 2024-11-23 by MartyG-RealSense

MartyG-RealSense, 2025-01-10:
Hi @BiomechatronicsRookie  https://github.com/IntelRealSense/librealsense/issues/7540 has some additional suggestions for solutions to the `AttributeError: module 'pyrealsense2' has no attribute 'pipeline'` error.