realsenseai/librealsense #13455: Cannot build librealsense for kinfu use

Drq13112, 2024-10-23:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR305 } | 
| Operating System & Version |  Windows 11  | 
| Platform                                 | PC |
| SDK Version                            |  { 2.54.2}                          | 

Hi there, I'm trying to build librealsense to use kinfu code but i'm getting this error. It is related to GLFW lib. I built it successfully previously but It seems like it can find glfw3Targets.cmake. Did someone have a similar error??

Make Error at C:/GLFW/build/src/glfw3Config.cmake:1 (include):
  include could not find requested file:

    C:/GLFW/build/src/glfw3Targets.cmake
Call Stack (most recent call first):
  wrappers/opencv/kinfu/CMakeLists.txt:6 (find_package)

CMake Error at C:/GLFW/build/src/glfw3Config.cmake:1 (include):
  include could not find requested file:

    C:/GLFW/build/src/glfw3Targets.cmake
Call Stack (most recent call first):
  CMakeLists.txt:77 (find_package)

GLFW 3.3 not found; using internal version
CMake Deprecation Warning at third-party/glfw/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

[image]

MartyG-RealSense, 2024-10-23:
Hi @Drq13112  Do you have the flag **BUILD_CV_KINFU_EXAMPLE** ticked as required in the rs-kinfu installation instrucitons, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu#installation

If it is not in the list of flags by default then you should be able to add it to the list with the **Add Entry** button.

MartyG-RealSense, 2024-10-23:
I'm pleased to hear that you made progress!

Are you using Visual Studio in the above image, please?  If you are then a simple way to set up project linkages to librealsense in VS is to use three .props property sheet files provided by the RealSense SDK, as described at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4412175411859/comments/4412182682387

There is a guide to setting up these .props files:

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

Drq13112, 2024-10-23:
I see... It looks very promising. I'm looking forward to try it, I'll let you know as soon as I have news. Thank you very much for the info :)

Drq13112, 2024-10-24:
I am back with the windows attempt, I have tried to follow the guidelines that you have told me about: https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

[image]

However I am stuck at this point, it seems to be some problem with the Linker. On the other hand, I have tried to follow the guide that appears in the README of the rs-kinfu github and I see some inconsistencies. For example, it is indicated to use opencv 3.4, but this version is incompatible with opencv-contrib 4.x, where the kinfu files are...

Drq13112, 2024-10-24:
I am back with the windows attempt, I have tried to follow the guidelines that you have told me about. However I am stuck at this point, it seems to be some problem with the Linker:

[image]

On the other hand, I tried to follow the guide that appears in the README of the rs-kinfu github and I see some inconsistencies. For example, it indicates that you have to use opencv 3.4, but this version is incompatible with opencv-contrib 4.x, where the kinfu files are...

Drq13112, 2024-10-25:
Hello, I've made a lot of progress!!! The file compiles!!!! How exciting, I was about to give up hahahaha , but I'm not done yet, what a surprise....

When the code runs I get this error, it looks like the system is looking for the glfw3.dll file but can't find it. 

[image]

[image]

A stack guy is facing the same problem without a good response from the community:
https://stackoverflow.com/questions/70659662/visual-studio-2019-c-project-error-a-dependent-dll-was-not-found-how-to-kno
does anyone have a suggestion? I get the feeling that it's an easy bug to fix but you have to hit the nail right on the head.

MartyG-RealSense, 2024-10-25:
There is advice about adding GLFW to Visual Studio here:

http://www.mcihanozer.com/tips/setting-up-libraries/setting-up-glfw-for-visual-studio/

At the end of the guide it provides the following advice:

**********

If it says something like “glfw3.dll is missing”, just go to .lib folder you pasted in **Adding Lib Files** (a section of the guide), and copy-paste the dll into the same directory your .exe is located i.e. Debug file.

Drq13112, 2024-10-28:
Hi, the guide was helpful, but it didn't work. In the end I realized that the code uses multithreading and I didn't include the glfw_mt.lib file that includes that feature. Now the code runs, however it stops at the very beginning with this error:
[image]

[image]

Researching for a few minutes on the internet, I found out that it is because my PC does not use GPU. I mean, this code is meant to use a GPU, but my laptop doesn't have one. Does anyone have a solution for this?

In addtion. I'm running in debug mode and I checked that the process memory gets top the top while CPU keeps low during execution.

[image]

MartyG-RealSense, 2024-10-28:
Often on modern laptops the graphics capability is built into the CPU, and this is called 'integrated graphics'.  Your log image shows that 'Intel UHD Graphics 620' is detected, which is the built-in GPU.

[image]

I would always recommend building in Release mode instead of Debug if possible so that librealsense can take advantage of optimizations provided by Release mode.

Some computers have problems with the **GLSL** graphics acceleration function of librealsense and will only show depth correctly if GLSL is disabled, so unticking the flag **BUILD_GLSL_EXTENSIONS** might make a difference.

Drq13112, 2024-10-28:
I see... I will try later. I think I can build realsense again, it isn't a big deal

When I was saying GPU I meant a dedicated one hahaha like an Nvidia

MartyG-RealSense, 2024-10-28:
Thanks very much for the clarification.  

rs-kinfu can be used without needing a dedicated video card.

Drq13112, 2024-11-11:
Hi there!! I'd been a little busy and I couldn't make a lot of progress. Well, now I'm struggling with this issue, I saw some other threads similar to this error. They concluded that it is better to use anothers sources such as open3d. Does anyonen know what it is making this error??

[image]

Debugging I reliased that the code is entering into this condition:

// Run KinFu on the new frame(on GPU)
if (!kf->update(frame))
{
    kf->reset(); // If the algorithm failed, reset current state
    // Save the pointcloud obtained before failure
    export_to_ply(_points, _normals);

    // To avoid calculating pointcloud before new frames were processed, set 'after_reset' to 'true'
    after_reset = true;
    points.release();
    normals.release();
    std::cout << "reset" << std::endl;
}

Which means that the camera or the code struggles updating of the next frame.... Idk, I'll look a bit at the code .... If I don't get it, I will give up this option and focus on other alternatives.

closed 2025-01-06 by MartyG-RealSense