realsenseai/librealsense #13459: pip install pyrealsense2 Defaulting to user installation because normal site-packages is not writeable ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none) ERROR: No matching distribution found for pyrealsense2

Zazmi2, 2024-10-23:
I am using the intel realsense camera d435. I am using a windows surface laptop ARM64 Processor. Pip install pyrealsense2 doesn't work for me. Also having trouble with building from source. Any help would be greatly appreciated!

Zazmi2, 2024-10-25:
I am using 2.56.2 but I am not able to do pip install pyrealsense2 regardless on my laptop. Here is the error I get:
PS C:\Users\vagmi> pip install pyrealsense2
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
PS C:\Users\vagmi>

I don't think it is compatible with my laptop - Windows Arm64 - Qualcomm.

Is there any workaround for using pyrealsense2

MartyG-RealSense, 2024-10-25:
On a PC laptop you need a Python 3 version that is supported by pyrealsense2.  When using librealsense 2.55.1, the supported Python versions are 3.7 to 3.11.

Support for Python 3.7 is ended in librealsense 2.56.1 onwards because its age means that it is classed as End of Life (EOL).

MartyG-RealSense, 2024-11-04:
Hi @Zazmi2  Do you require further assistance with this case, please?  Thanks!

closed 2024-11-15 by MartyG-RealSense

Rasmus-UCL, 2024-11-19:
i am currently running into the same problem here 

PS C:\Users\xxxx---- pip3 install pyrealsense2
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
PS C:\Users\xxxx---- python --version 
Python 3.12.6
PS C:\Users\xxxx---- pip3 --version 
pip 24.3.1 from C:\Users\xxxx\AppData\Roaming\Python\Python312\site-packages\pip (python 3.12)

can you help me point to the right directions to make it work

MartyG-RealSense, 2024-11-19:
Hi @Rasmus-UCL  The pip install method of installing the pyrealsense2 wrapper from packages only supports up to Python 3.11 at the time of writing this as Python 3.12 pip packages have not yet been released.  This is what is causing the mismatch message.

It is also necessary to use librealsense SDK version 2.55.1, as the recent intermediate beta versions 2.56.1 and 2.56.2 do not provide pip packages.

It is possible to use Python 3.12 if the pyrealsense2 wrapper is built from source code instead of installed from packages.

ruffsl, 2025-03-09:
> The pip install method of installing the pyrealsense2 wrapper from packages only supports up to Python 3.11 at the time of writing this as Python 3.12 pip packages have not yet been released.

Is there a roadmap we can subscribe to get notified for when pyrealsense2 receives support for a modern python version?

MartyG-RealSense, 2025-03-10:
Hi @ruffsl  There is not a publicly viewable roadmap.  Support for new Python 3 versions is announced in the release notes for new librealsense SDK versions.  

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes

At the time of writing this, the most recent Python support update was for 3.12 in librealsense beta 2.56.1.

[image]