realsenseai/librealsense #13460: SR300 crashes on start

WhiteOlivierus, 2024-10-24:
| | |
|---|---|
|**librealsense**|2.56.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-10-24:
Hi @WhiteOlivierus  For the SR300 camera model you should use version **2.50.0** of the RealSense Viewer tool (Intel.RealSense.Viewer.exe) as SR300 is a retired camera model and it is no longer supported in recent Viewer and SDK versions.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

The error that you experienced usually happens when attempting to access an incompatible camera using the Viewer's 'Add Source' button.

WhiteOlivierus, 2024-10-24:
It works, thank you so much for the quick reply

MartyG-RealSense, 2024-10-24:
You are very welcome.  It's great to hear that it works for you.  I'm pleased that I could help.  :)

closed 2024-11-04 by MartyG-RealSense