realsenseai/librealsense #13463: install d455f for Nvidia Isaac_ros

bgkang7470, 2024-10-24:
Installed librealsense2 and related packages correctly, But I can't find out menu button in realsense-viewer. 
And there is no camera view. Just x,y,z axis are appear. I am developing  ros_visual_slam on Nvidia Jetson Orin nano.

MartyG-RealSense, 2024-10-24:
Hi @bgkang7470  From your description, it sounds as though your camera is not detected by realsense-viewer, so there are no camera details in the options side-panel and nothing displayed in the 3D view.  Is the image below what your realsense-viewer looks like?

[image]

Which version of librealsense are you using, please?

Also, which method did you use to install librealsense and the related packages?

bgkang7470, 2024-10-25:
Hi MartyG-RealSense!
As you described, its phenomenon is exactly the same.  I can't find the
device selection option and  "Stereo Module" in the side-panel.
The status is the same as what you attached image.png(view on web).

My realSense installation sequences are as follows:
 1. Docker-based development environment through Isaac ROS Dev
<https://nvidia-isaac-ros.github.io/concepts/docker_devenv/index.html#development-environment>
.
         mkdir -p  /mnt/nova_ssd/workspaces/isaac_ros-dev/src
         echo "export ISAAC_ROS_WS=/mnt/nova_ssd/workspaces/isaac_ros-dev/"
>> ~/.bashrc
         source ~/.bashrc
 2. Nvidia Jetson setup - Docker on SSD and checked final verification of
docker
 3. install core ROS2 packages - OK
 4. Isaac ROS RealSense Setup
     1) Clone realsense-ros repository
         cd ${ISAAC_ROS_WS}/src
        git clone -b release-3.1
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common git isaac_ros_common
        git clone https://github.com/IntelRealSense/realsense-ros.git -b
4.51.1
   2) Configure the container
       cd ${ISAAC_ROS_WS}/src/isaac_ros_common/scripts && \
       touch .isaac_ros_common-config && \
       echo CONFIG_IMAGE_KEY=ros2_humble.realsense >
.isaac_ros_common-config
  3) Launch the Docker container
      cd ${ISAAC_ROS_WS}/src/isaac_ros_common && \
.      /scripts/run_dev.sh -d ${ISAAC_ROS_WS}
  4) Build realsense_splitter and realsense2* packages:
      cd ${ISAAC_ROS_WS} && \
      colcon build --symlink-install --packages-up-to-regex realsense*
   5) realsense-viewer

    I am not sure RealSense firmware and librealsense  are  correctly
installed in the above sequences.
    In the Nvidia document note, it is noted as follows.  Do it mean that
I have to install RealSense firmware and librealsens?

[image: {D76B5322-F87C-469E-A78E-0A58F84504BD}.png]

I am looking forward to your guidance as soon as possible. Thanks in
advance.

강 배근 / Bae Guen Kang
           대표이사/CEO
경기도 화성시 와우안길 17 수원대학교 고운 첨단과학기술연구원 903호
*Mobile:* +82-10-2278-1504  *Email: ***@***.***

MartyG-RealSense, 2024-10-25:
Regarding the Nvidia document note image, it did not display on your comment as images cannot be submitted by email and have to be pasted into the comment writing box on this page.

D455f support was not introduced until librealsense **2.54.1**, so the librealsense version that you install should be at least 2.54.1.

librealsense should be installed first and the ROS wrapper installed second after librealsense has been successfully installed.

The list below shows which firmware should be used with which librealsense version.

2.55.1, 2.56.1 and 2.56.2: firmware 5.16.0.1
2.54.2: firmware 5.15.1.0
2.54.1: firmware 5.15.0.2

bgkang7470, 2024-10-25:
Thank you, I will do that.

강 배근 / Bae Guen Kang
           대표이사/CEO
경기도 화성시 와우안길 17 수원대학교 고운 첨단과학기술연구원 903호
*Mobile:* +82-10-2278-1504  *Email: ***@***.***

MartyG-RealSense, 2024-11-04:
Hi @bgkang7470  Do you require further assistance with this case, please?  Thanks!

bgkang7470, 2024-11-15:
Hi
Now, D455f is working well on Jetpack 6.1  by use of SDK's DFORCE_RSUSB_BACKEND option. For time being, This is enough.
Later, I will try again Linux native kernel driver patch for UVC.

bgkang7470, 2024-11-15:
> I'm pleased to hear that RSUSB Backend is providing a working solution for
> you. Thanks very much for the update!
>
> Please do let me know the outcome of your Linux native kernel driver patch
> test.

Yes, I will let you know.

강 배근 / Bae Guen Kang
           대표이사/CEO
경기도 화성시 와우안길 17 수원대학교 고운 첨단과학기술연구원 903호
*Mobile:* +82-10-2278-1504  *Email: ***@***.***

closed 2024-12-13 by MartyG-RealSense