realsenseai/librealsense #13464: Get raw depth data through python

purpleyoda1, 2024-10-25:
Hi!
Is it possible to access raw depth data through the python API? I followed the example code for getting depth maps, but the resulting maps seem to have been run through several filters when comparing it to the raw stream in realsense viewer. I tried accessing the depth sensor or the pipelines processing block, but its difficult to figure out what methods and options are available for the different objects. Is there a way to disable the filters in python?

I apologize if this is actually a really simple case or i have missed something obvious (im quite the novice), but i have been trying on my own for a while now. Any help is greatly appreciated!

My current function can be seen below, although it doesn't run of course 

def capture_and_save_raw_depth_map(output_path):
    # Configure realsense pipeline
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.depth, 424, 240, rs.format.z16, 30)
    
    # Start stream
    profile = pipeline.start(config)

    # Get depth sensor
    depth_sensor = profile.get_device().first_depth_sensor()
    depth_sensor.set_option(rs.filter_option, 0)

    # Disable post processing filters
    processing_blocks = profile.get_device().query_processing_blocks()
    for block in processing_blocks:
        try:
            block.set_option(rs.option.filter_option, 0)
        except:
            pass
    
    try:
        # Skip first 30 frames to stabilize
        for i in range(30):
            pipeline.wait_for_frames()
        
        # Get depth frame
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        if not depth_frame:
            raise Exception("Could not get depth frame")
        
        # Convert to numpy array and save
        depth_map = np.asanyarray(depth_frame.get_data())
        cv2.imwrite(output_path, depth_map)
        print(f"Depth image saved to {output_path}")

    finally:
        pipeline.stop()

MartyG-RealSense, 2024-10-25:
Hi @purpleyoda1  A key reason why the realsense-viewer depth images can look different to Python ones is that realsense-viewer applies a range of post-processing filters and depth colorization settings by default.  A Python script applies **no** filtering by default - filters have to be deliberately programmed into the script.

So you do no need to attempt to disable post-processing filters in Python because they are already inactive by default.

purpleyoda1, 2024-10-25:
Oh wow i see, thanks! I assumed they were on based on visual inspection of the outputs. Thank you so much for the quick response!

closed 2024-10-25 by purpleyoda1