realsenseai/librealsense #13467: orin nx 16 使用humble,跑rtabmap_ros的报错

2805651606, 2024-10-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.15                                        | 
| Platform                                 | Jetson Orin Nx  |
| Language                            |  {C/python}                          | 
| Segment			|  {Robot}                          | 

sudo -i
git clone https://github.com/jetsonhacks/installRealSenseSDK.git
cd installRealSenseSDK
# 编译方式如下，默认使用 CUDA
# ./buildLibrealsense.sh [ -v | --version <version> ] [ -j | -jobs <number of jobs> ] [ -n | --no_cuda ]
./buildLibrealsense.sh -v v2.55.1 -j 8
通过这个完成了librealsense的安装，可以正常使用，使用realsense-viewer可以正常打开，并且再安装了realsense-ros后，也可以使用d435i相机

执行下面语句
git clone https://github.com/introlab/rtabmap.git src/rtabmap
git clone --branch ros2 https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
rosdep update && rosdep install --from-paths src --ignore-src -r -y
export MAKEFLAGS="-j6" # Can be ignored if you have a lot of RAM (>16GB)
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

执行到rosdep update && rosdep install --from-paths src --ignore-src -r -y之后，发现在使用realsense-viewer就再也打不开相机了

orin@ubuntu:~/ros2_rt$ realsense-viewer
  26/10 12:29:06,526 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0
  26/10 12:29:06,732 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video1
  26/10 12:29:06,733 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video2
  26/10 12:29:06,734 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video3
  26/10 12:29:06,735 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.3/video4linux/video4
  26/10 12:29:06,737 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.3/video4linux/video5
  26/10 12:29:07,106 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0
  26/10 12:29:07,108 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video1
  26/10 12:29:07,108 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video2
  26/10 12:29:07,110 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video3
  26/10 12:29:07,110 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.3/video4linux/video4
  26/10 12:29:07,111 INFO [281472984776736] (backend-v4l2.cpp:753) Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.3/2-1.3:1.3/video4linux/video5
  26/10 12:29:07,119 INFO [281472984776736] (context.cpp:128) Found 0 RealSense devices (0xff requested & 0xff from device-mask in settings)
  请问，如何解决

2805651606, 2024-10-28:
@MartyG-RealSense  你好，并没有啊。librealsense的安装过程是
sudo -i
git clone https://github.com/jetsonhacks/installRealSenseSDK.git
cd installRealSenseSDK
./buildLibrealsense.sh [ -v | --version ] [ -j | -jobs ] [ -n | --no_cuda ]
realsense-ros的安装过程是
先建一个ros2_ws文件，在里面安装

2805651606, 2024-11-01:
那为啥执行rosdep update && rosdep install --from-paths src --ignore-src -r -y后，就无法找到相机了呢，板子可以识别到相机连接到板子上了，但是realsense-viewer却找不到了

MartyG-RealSense, 2024-11-01:
Because of the complicated installation which contains librealsense, the RealSense ROS wrapper and rtabmap_ros, I would recommend wiping everything and starting again from the beginning with a clean installation if it is possible for you to do.  Install librealsense only first and see if the realsense-viewer works again.  Then you can move on to installing the ROS wrapper and rtabmap-ros.

********

由于安装过程复杂，包含 librealsense、RealSense ROS 包装器和 rtabmap_ros，我建议清除所有内容并从头开始全新安装（如果可能）。首先仅安装 librealsense，然后查看 realsense-viewer 是否再次运行。然后，您可以继续安装 ROS 包装器和 rtabmap-ros。

2805651606, 2024-11-02:
已经干过了，重新刷机之后弄的。仅安装 librealsense，然后查看 realsense-viewer 可以运行。安装 ROS 包装器，也可以运行。就是安装 rtabmap-ros之后，realsense-viewer 不行了

MartyG-RealSense, 2024-11-02:
Is realsense-viewer necessary for your work or could you manage without it?  Thank you.

************

realsense-viewer 对你的工作来说是否必要？或者没有它你也能应付吗？谢谢。

2805651606, 2024-11-04:
realsense-viewer确实不太重要，但是我跑slam算法的时候，没有办法启动相机了。就是我没办法读到相机的imu，图像等数据了，调用时，显示找不到相机

MartyG-RealSense, 2024-11-04:
As you are using rtabmap_ros with ROS2, are you using the `rtabmap.launch.py` launch file please?

If you are using rtabmap.launch.py, have you launched the RealSense ROS launch file first to publish the RealSense camera topics before then launching the rtabmap_ros launch file afterward once the RealSense launch has completed successfully?

STEP 1
`ros2 launch realsense2_camera rs_launch.py `

STEP 2
`roslaunch rtabmap_ros rtabmap.launch`

**************

由于您正在将 rtabmap_ros 与 ROS2 结合使用，请问您是否使用了 `rtabmap.launch.py​​` 启动文件？

如果您使用的是 rtabmap.launch.py​​，您是否先启动了 RealSense ROS 启动文件以发布 RealSense 摄像头主题，然后在 RealSense 启动成功完成后再启动 rtabmap_ros 启动文件？

步骤 1
`ros2 launch realsense2_camera rs_launch.py​​ `

步骤 2
`roslaunch rtabmap_ros rtabmap.launch`

MartyG-RealSense, 2024-11-23:
Case closed due to no further comments received.

由于未收到进一步评论，案件结案。

closed 2024-11-23 by MartyG-RealSense