realsenseai/librealsense #13468: d435 with ros2

naorwaiss, 2024-10-28:
hi i try to run the d435 camera node with this command:
ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x30 pointcloud.enable:=true

then i try to convert the point_cloud2  to laserscan to build a costmap

so i get 2 question:
1) i need some imu to build the cost_map ? - if i dont have internal imu i need to add one?
2) i try to use https://github.com/ros-perception/pointcloud_to_laserscan/tree/humble 
to convert from pointcloud to laserscan but it lock like i have alot of inf at my topic of the laserscan 

thanks for the help!

MartyG-RealSense, 2024-10-28:
Hi @naorwaiss  Thanks very much for your questions.  I researched your question carefully.  A popular approach appears to be to use **costmap_2d** to generate the costmap after **depthimage_to_laserscan** has been used, as described at https://github.com/IntelRealSense/realsense-ros/issues/2442

naorwaiss, 2024-10-30:
tanks for the help --- i stack at the convert from the depth to the laser scan but i fix it

naorwaiss, 2024-11-08:
ok so i try for sometime to fix and understand some problem with laser scan and with the tf of the camera and the world but now i have problem with the nav2 

```
naor@naor-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/Desktop/naor/study/hamama$ ros2 launch camera_nav camera_nav_launch.py 
[INFO] [launch]: All log files can be found below /home/naor/.ros/log/2024-11-08-23-00-54-466274-naor-ASUS-TUF-Gaming-F15-FX507VV-FX507VV-198704
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_isolated-1]: process started with pid [198717]
[component_container_isolated-1] [INFO] [1731099655.213319178] [nav2_container]: Load Library: /opt/ros/humble/lib/libmap_server_core.so
[component_container_isolated-1] [INFO] [1731099655.241847206] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::CostmapFilterInfoServer>
[component_container_isolated-1] [INFO] [1731099655.241905363] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapSaver>
[component_container_isolated-1] [INFO] [1731099655.241911909] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated-1] [INFO] [1731099655.241916854] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated-1] [INFO] [1731099655.251611028] [map_server]: 
[component_container_isolated-1]        map_server lifecycle node launched. 
[component_container_isolated-1]        Waiting on external lifecycle transitions to activate
[component_container_isolated-1]        See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1731099655.251903268] [map_server]: Creating
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/map_server' in container '/nav2_container'
[component_container_isolated-1] [INFO] [1731099655.255103939] [nav2_container]: Load Library: /opt/ros/humble/lib/libamcl_core.so
[component_container_isolated-1] [INFO] [1731099655.261536136] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_amcl::AmclNode>
[component_container_isolated-1] [INFO] [1731099655.261574612] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_amcl::AmclNode>
[component_container_isolated-1] [INFO] [1731099655.266763365] [amcl]: 
[component_container_isolated-1]        amcl lifecycle node launched. 
[component_container_isolated-1]        Waiting on external lifecycle transitions to activate
[component_container_isolated-1]        See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1731099655.267740829] [amcl]: Creating
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/amcl' in container '/nav2_container'
[component_container_isolated-1] [INFO] [1731099655.271349913] [nav2_container]: Load Library: /opt/ros/humble/lib/libnav2_lifecycle_manager_core.so
[component_container_isolated-1] [INFO] [1731099655.272451772] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_lifecycle_manager::LifecycleManager>
[component_container_isolated-1] [INFO] [1731099655.272469229] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_lifecycle_manager::LifecycleManager>
[component_container_isolated-1] [INFO] [1731099655.276839180] [lifecycle_manager_localization]: Creating
[component_container_isolated-1] [INFO] [1731099655.279721954] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/lifecycle_manager_localization' in container '/nav2_container'
[component_container_isolated-1] [INFO] [1731099655.282156822] [lifecycle_manager_localization]: Starting managed nodes bringup...
[component_container_isolated-1] [INFO] [1731099655.282200139] [lifecycle_manager_localization]: Configuring map_server
[component_container_isolated-1] [INFO] [1731099655.282341865] [map_server]: Configuring
[component_container_isolated-1] [ERROR] [1731099655.282691227] [map_server]: Caught exception in callback for transition 10
[component_container_isolated-1] [ERROR] [1731099655.282704669] [map_server]: Original error: parameter 'yaml_filename' is not initialized
[component_container_isolated-1] [WARN] [1731099655.282720050] [map_server]: Error occurred while doing error handling.
[component_container_isolated-1] [FATAL] [1731099655.282728823] [map_server]: Lifecycle node map_server does not have error state implemented
[component_container_isolated-1] [ERROR] [1731099655.282873341] [lifecycle_manager_localization]: Failed to change state for node: map_server
[component_container_isolated-1] [ERROR] [1731099655.282890807] [lifecycle_manager_localization]: Failed to bring up all requested nodes. Aborting bringup.
[component_container_isolated-1] [INFO] [1731099655.320133881] [nav2_container]: Load Library: /opt/ros/humble/lib/libcontroller_server_core.so
[component_container_isolated-1] [INFO] [1731099655.329688183] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_controller::ControllerServer>
[component_container_isolated-1] [INFO] [1731099655.329781270] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_controller::ControllerServer>
[... cut ...]
    param_file = os.path.join(pkg_share, 'config', 'costmap_params.yaml')
    map_file = os.path.join(pkg_share, 'maps', 'map.yaml')  # Update the path to your map file

    return launch.LaunchDescription([
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(bringup_launch_file),
            launch_arguments={
                'params_file': param_file,
                'map': map_file
            }.items()
        )
    ])

```

if i need to go and ask at the nav2 fill free to say..
thanks !!

MartyG-RealSense, 2024-11-09:
I do not have knowledge about nav2 unfortunately, so I certainly recommend asking at the navigation2 support forum at the link below.  Thanks!

https://github.com/ros-navigation/navigation2/issues

MartyG-RealSense, 2024-11-15:
Hi @naorwaiss  Bearing in mind the recommendation in the comment above, do you require further assistance with this case please?  Thanks!

naorwaiss, 2024-11-17:
hi you can close ...

‫בתאריך יום ו׳, 15 בנוב׳ 2024 ב-15:59 מאת ‪MartyG-RealSense‬‏ <‪
***@***.***‬‏>:‬

closed 2024-11-17 by MartyG-RealSense