realsenseai/librealsense #13470: Initialization Delay Due to Remote Desktop Camera Devices on Windows 11

pasandal, 2024-10-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 11  |                                          | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          |                      | 

### Issue Description
Hello,
I'm experiencing a significant initialization delay with my RealSense D435 camera when running a Python script using the `pyrealsense2` library on a Windows 11 machine. The camera takes over 10 seconds to initialize, whereas on other machines, the same script initializes the camera in less than 1 second.

I'm using a simple Python script to test the camera's performance:

```
import pyrealsense2 as rs2
from time import time

def get_realsense_device():
    devices = rs2.context().query_devices()
    if len(devices) == 0:
        raise ValueError("No camera found")
    return devices[0]

rs2.log_to_console(rs2.log_severity.debug)

camera_framerate = 15
pipeline = rs2.pipeline()
config = rs2.config()
col, row = 848, 480  # Resolution

start_total = time()

# Timing get_realsense_device()
start = time()
device = get_realsense_device()
print(f"Time to get device: {time() - start:.2f}s")

serial_number = device.get_info(rs2.camera_info.serial_number)
config.enable_device(serial_number)

# Timing enable_stream()
start = time()
config.enable_stream(rs2.stream.depth, col, row, rs2.format.z16, camera_framerate)
config.enable_stream(rs2.stream.color, col, row, rs2.format.bgr8, camera_framerate)
print(f"Time to enable streams: {time() - start:.2f}s")

stream_queue = rs2.frame_queue(50, keep_frames=True)

# Timing pipeline.start()
start = time()
pipeline_profile = pipeline.start(config, stream_queue)
print(f"Time to start pipeline: {time() - start:.2f}s")

print(f"Total execution time: {time() - start_total:.2f}s")
```
**Observed Behavior:**
- On a Windows 11 laptop, the script takes approximately 11 seconds to execute.
- The delays occurs during `get_realsense_device()` and `pipeline.start()`.
- On other machines, the same script initializes in less than 1 second.
- The camera is connected via USB 3.2, and the connection has been verified.

**Debug Log Highlights:**

- Repeated entries like:
`_reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."`

**Troubleshooting Steps Taken:**
- Disabling webcam devices: In Device Manager -> disable all webcam devices, including virtual cameras and remote desktop redirected cameras

**Identified the root cause:**
- Disabling the "Remote Desktop Camera Device (HD Webcam Redirected) reduced the execution time from over 10 seconds to approx 0.5 seconds.

**Questions:**
1. Is this a known issue with the RealSense SDK on Windows 11 when remote desktop camera devices are present?
2. Are there recommended solutions or best practices to prevent such delays without manually disabling devices in Device Manager?
3. Is there a way to configure the RealSense SDK or pyrealsense2 to ignore certain devices during initialization?

MartyG-RealSense, 2024-10-28:
Hi @pasandal  It is possible for webcams to have conflicts with RealSense cameras and vice versa, though I am not aware of a past case that is specific to Windows 11.

The RealSense SDK regards non-RealSense cameras as a 'platform camera'.  The pyrealsense2 script file linked to below, a device handling script from the 'box_dimensioner_multicam.py' project, includes an If condition that only accesses the details of cameras that do not have 'platform camera' in their name, thereby ignoring the non-RealSense camera.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L44-L45

Are the other computers that you are using Windows computers too (such as Windows 10) or are they Ubuntu machines?

pasandal, 2024-10-29:
Hello, thanks for answering.
The common factors among the computers experiencing this issue are that they are laptops, running Windows 11, and have a webcam. I haven’t encountered this problem on my desktop computer, which does not have a webcam and is running either Linux or Windows 11. The most likely cause of the problems is the integrated webcam.

MartyG-RealSense, 2024-10-29:
Does it make any difference if you remove frame_queue from the pipe start line's brackets so that the pipeline activation ignores that instruction?

`pipeline_profile = pipeline.start(config)`

closed 2024-11-15 by MartyG-RealSense