realsenseai/librealsense #13474: Failed to create ros reader: Error opening file

zerostarter1998, 2024-10-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-10-30:
Hi @zerostarter1998  json camera configuration files are usually exported from the realsense-viewer tool instead of the Depth Quality Tool (DQT).  In the currently seven-year history of the RealSense 400 Series I am not aware of any RealSense users that have used a json exported from the DQT.  Would it be possible to repeat your settings in realsense-viewer and export a json from there, please?

You can also test your script's json import by downloading an official RealSense SDK json from the table at the link below.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

zerostarter1998, 2024-10-31:
Thank you for your reply. I exported a json from realsense-viewer repeating the same settings. I checked the files and found the same content, so the problem is not on that.
I downloaded a json from the mentioned link and tried to load it, leading to the same result.
Anyway the problem is unsolved and for the record the code is running in Windows 11 system.

MartyG-RealSense, 2024-10-31:
The problem may be that the `config.enable_device_from_file()` instruction is usually used for importing .bag files (recordings of camera data) and not json camera configuration files.  For importing json files, the instruction **json.load** is used.  https://github.com/IntelRealSense/librealsense/issues/6075#issuecomment-600087038 has an example of a Python script for importing a json with a specific filename.

The link below has another Python script that demonstrates use of json.load.

https://stackoverflow.com/questions/53520301/how-to-load-json-file-to-realsense-d435-camera-using-c-and-intel-api/53633791#53633791

closed 2024-11-01 by zerostarter1998