realsenseai/librealsense #13477: Realsense SDK crashes as I plug in my D435i web camera

queen0fpentacles, 2024-10-30:
### Issue Description
The Realsense SDK crashes each time I plug in my D435i web camera and try to open the viewer, I do not know why this is happening. This is librealsense 2.56.2. Any help would be much appreciated, thank you. 

|---------------------------------|------------------------------------------- |
| Camera Model                       D435i
| Firmware Version                   | (Open RealSense Viewer --> Click info)
| Operating System & Version |   MacOS
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..
| SDK Version                            |  2.56.2.                      
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                       
| Segment			|  {Robot/Smartphone/VR/AR/others }

MartyG-RealSense, 2024-10-31:
Hi @queen0fpentacles  Can you confirm which computer or computing device you are using please (PC, Mac, Raspberry Pi, Nvidia Jetson, etc).  Thanks!

Does the crash happen immediately after plugging the camera in?  And is there an error message displayed during the crash?

queen0fpentacles, 2024-10-31:
I have a Mac Apple M3 Pro, macOS sequoia version 15.1
Yes, immediately when plugged in. The application will open just fine until the camera is plugged in. 

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               realsense-viewer [3633]
Path:                  /opt/homebrew/*/realsense-viewer
Identifier:            realsense-viewer
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [2215]
Responsible:           Terminal [2213]
User ID:               501

Date/Time:             2024-10-31 10:39:49.6943 +0000
OS Version:            macOS 15.1 (24B83)
Report Version:        12
Anonymous UUID:        E84864A7-89C4-10C3-8E3F-74D55487414F

Time Awake Since Boot: 5100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Codes:       0x0000000000000001, 0x0000000000000028

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [3633]

VM Region Info: 0x28 is not in any region.  Bytes before following region: 4334600152
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1025cc000-1027b4000    [ 1952K] r-x/r-x SM=COW  /opt/homebrew/*/realsense-viewer

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_pthread.dylib       	       0x1869198e0 pthread_mutex_lock + 12
1   libusb-1.0.0.dylib            	       0x102b90f30 libusb_get_device_list + 268
2   librealsense2.2.56.2.dylib    	       0x1038ab7d4 librealsense::platform::usb_context::usb_context() + 448
3   librealsense2.2.56.2.dylib    	       0x1038b41f0 std::__1::__shared_ptr_emplace<librealsense::platform::usb_context, std::__1::allocator<librealsense::platform::usb_context>>::__shared_ptr_emplace[abi:ue170006]<>(std::__1::allocator<librealsense::platform::usb_context>) + 40
4   librealsense2.2.56.2.dylib    	       0x1038b41a0 std::__1::shared_ptr<librealsense::platform::usb_context> std::__1::allocate_shared[abi:ue170006]<librealsense::platform::usb_context, std::__1::allocator<librealsense::platform::usb_context>, void>(std::__1::allocator<librealsense::platform::usb_context> const&) + 32
5   librealsense2.2.56.2.dylib    	       0x1038b38cc librealsense::platform::usb_enumerator::query_devices_info() + 60
6   librealsense2.2.56.2.dylib    	       0x1038baa64 librealsense::platform::query_uvc_devices_info() + 48
7   librealsense2.2.56.2.dylib    	       0x1038e41bc librealsense::backend_device_factory::query_devices(unsigned int) const + 144
8   librealsense2.2.56.2.dylib    	       0x1038e93b8 librealsense::context::query_devices(int) const + 168
9   librealsense2.2.56.2.dylib    	       0x10390ea4c rs2_query_devices_ex + 52
10  realsense-viewer              	       0x10261c918 rs2::context::query_devices(int) const + 36
11  realsense-viewer              	       0x10261c57c rs2::ux_window::imgui_config_push() + 708
12  realsense-viewer              	       0x10261cc04 rs2::ux_window::operator bool() + 604
13  realsense-viewer              	       0x1026eafcc main + 1600
14  dyld                          	       0x18659c274 start + 2840

Thread 1:
0   libsystem_kernel.dylib        	       0x1868e15cc __psynch_cvwait + 8
1   libsystem_pthread.dylib       	       0x18691f894 _pthread_cond_wait + 1204
2   libc++.1.dylib                	       0x18685560c std::__1::condition_variable::__do_timed_wait(std::__1::unique_lock<std::__1::mutex>&, std::__1::chrono::time_point<std::__1::chrono::system_clock, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l>>>) + 100
3   librealsense2.2.56.2.dylib    	       0x1037a7e4c bool std::__1::condition_variable::wait_for[abi:ue170006]<long long, std::__1::ratio<1l, 1000l>, bool dispatcher::cancellable_timer::try_sleep<std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l>>>(std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l>>)::'lambda'()>(std::__1::unique_lock<std::__1::mutex>&, std::__1::chrono::duration<std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l>>, std::__1::ratio<1l, 1000l>> const&, bool dispatcher::cancellable_timer::try_sleep<std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l>>>(std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l>>)::'lambda'()) + 240
4   librealsense2.2.56.2.dylib    	       0x1037a7d10 bool dispatcher::cancellable_timer::try_sleep<std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l>>>(std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000l>>) + 88
5   librealsense2.2.56.2.dylib    	       0x1038cbe44 librealsense::polling_device_watcher::polling(dispatcher::cancellable_timer) + 60
6   librealsense2.2.56.2.dylib    	       0x1037a8b1c void std::__1::__invoke_void_return_wrapper<void, true>::__call[abi:ue170006]<active_object<std::__1::function<void (dispatcher::cancellable_timer)>>::do_loop()::'lambda'(dispatcher::cancellable_timer)&, dispatcher::cancellable_timer>(active_object<std::__1::function<void (dispatcher::cancellable_timer)>>::do_loop()::'lambda'(dispatcher::cancellable_timer)&, dispatcher::cancellable_timer&&) + 52
[... cut ...]
      "rolloutId" : "648a2601f74c42732876cb5b",
      "factorPackIds" : {
        "SIRI_TEXT_TO_SPEECH" : "67183efc59e1d560c4b6fe19"
      },
      "deploymentId" : 240000263
    },
    {
      "rolloutId" : "645c2d2f9e69a025b0a37e29",
      "factorPackIds" : {

      },
      "deploymentId" : 240000003
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}

MartyG-RealSense, 2024-10-31:
Hi @queen0fpentacles  Thanks very much for the confirmation that you have an Apple Silicon 'M' Mac.  

Other Mac users at https://github.com/IntelRealSense/librealsense/issues/12727 have reported this crash issue with realsense-viewer when inserting their camera.

'M' Macs have had problems with using RealSense with them since MacOS Monterey.  However, Intel implemented a fix for librealsense compilation on these Macs in librealsense 2.56.1 and an update for MacOS Sonoma, as described at https://github.com/IntelRealSense/librealsense/issues/11815#issuecomment-2326189604

You are the first MacOS user on this support forum to report using the new Sequoia version though, so there are not currently solutions available for that specific MacOS version.

Does it make a difference if you launch realsense-viewer in sudo admin permissions mode with the command **sudo realsense-viewer**

queen0fpentacles, 2024-10-31:
This is what it says when I do that: 

GLFW Driver Error: Cocoa: Regular windows do not have icons on macOS
2024-10-31 14:08:51.331 realsense-viewer[7114:169977] +[IMKClient subclass]: chose IMKClient_Modern
zsh: segmentation fault  sudo realsense-viewer
anjakrusell@Anjas-MacBook-Pro ~ % 

The same crash report as before also pops up.

queen0fpentacles, 2024-10-31:
Yesterday night I just updated my laptop to sequoia from macOS 14.6 as I saw somewhere that maybe my Mac had to be updated. Would you recommend going back to the previous update?

MartyG-RealSense, 2024-10-31:
Yes, librealsense 2.56.1 or 2.56.2 and MacOS 14 (Sonoma) is likely your best chance of having a successful outcome until a solution that works with Sequoia is found by the RealSense MacOS user community.

omarsamman1, 2024-11-03:
I've been experiencing the same issue when connecting my D457 on my M2 mac. I also updated to Sequoia 15.1. @queen0fpentacles any luck finding a solution?

niclimcy, 2024-11-08:
I can confirm this issue occurs after upgrading to OSX 15.1. It was working fine with sudo on OSX 15.0

MartyG-RealSense, 2024-12-01:
Hi @omarsamman1 and @niclimcy  Were either of you able to make progress with your problems on MacOS, please?  Thanks!

niclimcy, 2024-12-01:
> Hi @omarsamman1 and @niclimcy Were either of you able to make progress with your problems on MacOS, please? Thanks!

It's inconsistent but I have to run with root multiple times before the camera view launches. Usually it says "failed to set power state"

MartyG-RealSense, 2024-12-01:
Hi @niclimcy  Thanks very much for the update.  There have been improvements in the librealsense SDK recently regarding support for MacOS Apple Silicon 'm' computers, as mentioned above at https://github.com/IntelRealSense/librealsense/issues/13477#issuecomment-2449941916 but problems with accessing the camera over USB on Mac do persist.  I'm glad that you can at least get it working after several attempts.

omarsamman1, 2024-12-03:
> Hi @omarsamman1 and @niclimcy Were either of you able to make progress with your problems on MacOS, please? Thanks!

I got to the point where the viewer launches, but I can't access the footage I still get the same errors but on the viewer. Only solution was to switch to Windows lol

closed 2025-01-06 by MartyG-RealSense