realsenseai/librealsense #13483: Store metadata in the image format

captain-yoshi, 2024-11-01:
When using the `realsense-viewer` it would be great to store some metadata in the image format. E.g. for creating a dataset with some camera parameters. Is this currently possible ?

This is an example using the Nikon D300 camera:
[image]

Realsense D415:
[image]

MartyG-RealSense, 2024-11-01:
Hi @captain-yoshi  A list of the metadata parameters supported by the RealSense SDK can be found at the documentation link below.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.frame_metadata_value.html

In the RealSense Viewer, if you use the **Snapshot** camera-shaped button on the toolbar of small icons at the top of the depth stream panel then it will export a .csv format textual data file containing the metadata of the exported depth frame.  The .csv can be opened in a spreadsheet program.

[image]

[image]

MartyG-RealSense, 2024-11-09:
Hi @captain-yoshi  Do you require further assistance with this case, please?  Thanks!

captain-yoshi, 2024-11-11:
Super, thanks for the answer and the docs.

closed 2024-11-11 by captain-yoshi

MartyG-RealSense, 2024-11-11:
You are very welcome, @captain-yoshi - I'm pleased that I could help.  :)