realsenseai/librealsense #13486: The question about sdk and its corresponding FW version

dyb678, 2024-11-02:
Hi! I use the 2.19.0 sdk and I don't its corresponding FW version.Thanks!

MartyG-RealSense, 2024-11-02:
Hi @dyb678  The recommended firmware version for SDK 2.19.0 is **5.11.4**

https://github.com/IntelRealSense/librealsense/releases/tag/v2.19.0

[image]

However - and this is VERY important - only a D415 or D435 camera model could support a firmware that old.  For example, if a D455 was updated with the 5.11.4 firmware then it would become unusable and no longer detectable.

[image]

dyb678, 2024-11-02:
OO,thanks!Actually,I find when I use the imu method" linear_interpolation" in ros wrapper will cause the data unstable but "copy"method seems not suitable for SLAM task. How can solve it ?Thanks a lot!

MartyG-RealSense, 2024-11-02:
Raw RealSense IMU data is inherently "noisy".  So there may not be much that you can do to stabilize the data if the Copy option cannot be used.

Ideally, fast moving or turning of the camera should be avoided if possible as this can cause the IMU's tracking to be lost.

You could also try setting accel and gyro to their maximum speed values to see how this affects the data when using linear_interpolation.  

**ROS1**
`roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation accel_fps:=200 gyro_fps:=400`

**ROS2**
`ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=2 accel_fps:=200 gyro_fps:=400`

closed 2024-11-15 by MartyG-RealSense