realsenseai/librealsense #13487: control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

bsseo89, 2024-11-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-11-03:
Hi @bsseo89  A **control_transfer returned** warning can indicate that there is a communiction problem between the camera and the computer, typically with the USB system (the USB port or the USB cable).

Which model of RealSense camera do you have, please?  And are you using the official USB cable supplied with the camera or a cable of your own choice?

MartyG-RealSense, 2024-11-15:
Hi @bsseo89  Do you require further assistance with this case, please?  Thanks!

liuzhuangjisu, 2024-11-17:
Hi,My camera model is D435i and I use my own USB cable

MartyG-RealSense, 2024-11-17:
It is important that a self-chosen USB cable that is used with RealSense is high quality, especially if it is longer than 2 meters, and that the cable is designed for data transfer and not just device charging.