realsenseai/librealsense #13493: Librealsense dkms build failed on Ubuntu 22.04 kernel 6.0.0-1016-oem

rxng8, 2024-11-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | N/A | 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  6.0.0-1016-oem                                  | 
| Platform                                 | Dell G15 - 5530 |
| SDK Version                            |  latest                  | 
| Language                            | N/A           | 
| Segment			|  installation                | 

### Issue Description
When I installed from both (official and manual) installation guides, I both got the build failed error. Here is the log of `/var/crash/librealsense2-dkms.0.crash`
```
ProblemType: Package
DKMSBuildLog:
 DKMS make.log for librealsense2-dkms-1.3.27 for kernel 6.0.0-1016-oem (x86_64)
 Tue Nov  5 06:27:13 PM EST 2024
 make: Entering directory '/usr/src/linux-headers-6.0.0-1016-oem'
 warning: the compiler differs from the one used to build the kernel
   The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
   You are using:           gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
 scripts/Makefile.build:43: /var/lib/dkms/librealsense2-dkms/1.3.27/build/6.0.0/drivers/media/usb/uvc/Makefile: No such file or directory
 make[1]: *** No rule to make target '/var/lib/dkms/librealsense2-dkms/1.3.27/build/6.0.0/drivers/media/usb/uvc/Makefile'.  Stop.
 make: *** [Makefile:1856: /var/lib/dkms/librealsense2-dkms/1.3.27/build/6.0.0/drivers/media/usb/uvc] Error 2
 make: Leaving directory '/usr/src/linux-headers-6.0.0-1016-oem'
DKMSKernelVersion: 6.0.0-1016-oem
Date: Tue Nov  5 18:27:14 2024
Package: librealsense2-dkms 1.3.27-0ubuntu1
PackageVersion: 1.3.27-0ubuntu1
SourcePackage: librealsense2
Title: librealsense2-dkms 1.3.27-0ubuntu1: librealsense2-dkms kernel module failed to build
```

MartyG-RealSense, 2024-11-06:
Hi @rxng8  I note that you are using kernel 6.0.  When building the librealsense SDK from dkms packages, the supported kernel 6 versions are 6.2 and 6.5.

[image]

If you are unable to change your kernel version then you could instead build librealsense from source code with a pre-made build script called 'libuvc_installation.sh' using the **libuvc backend** method, which bypasses the kernel and so is not dependent on a particular Linux version or kernel version.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

MartyG-RealSense, 2024-11-15:
Hi @rxng8  Do you require further assistance with this case, please?  Thanks!

rxng8, 2024-11-16:
Yep, I got it to work using only the installation script, I also have to remove all pre-installed realsense package beforehand.
I can now visualize the camera frame by running `realsense-viewer`

Thank you very much, appreciate it!

MartyG-RealSense, 2024-11-16:
That's great news that you were successful.  I'm pleased that I could be of help.  Thanks very much for the update!