realsenseai/librealsense #13494: An error occurred while upgrading the D455 firmware using rs-fw-update on Ubuntu 20.04 focal LTS system with hardware RK3588S

diplomatist, 2024-11-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.13.0.55->5.16.0.1 | 
| Operating System & Version |   ubuntu20.04 focal LTS | 
| Kernel Version (Linux Only)    |  5.10                                        | 
| Platform                                 | rk3588s |
| SDK Version                            |  2.55.1 ,2.56.0.,2.56.2                          | 
| Language                            |  c++                        | 
| Segment			|  others                         | 

### Issue Description
I have a batch of D455 cameras with firmware version 5.13.0.55;
I compiled SDK versions 2.55.1, 2.56.0, and 2.56.2 in rk3588s;
I upgraded the camera from 5.13.0.55 to 5.16.0.1 using the compiled rs fw update, and three of the cameras encountered the following issue (two of them were normal);

[image]
The image shows that the camera with firmware version 5.16.0.1 has been re-upgraded.

The image is for reproducing the error message; Specifically, after upgrading the camera from 5.13.0.55 to 5.16.0.1, the camera entered recovery mode and could not be restored and upgraded through RS-FW-update-R. The error message is as follows:
[image]

I can only upgrade the recovery mode camera to version 5.16.0.1 using the realsense viewer tool（ 2.55.1 version） on Windows；

May I ask why rs-fw-update on Linux has this issue.

MartyG-RealSense, 2024-11-06:
Hi @diplomatist  There is a Windows version of the firmware update tool called **Intel.RealSense.FW.Update** that can be downloaded from the 'Assets' file list on the SDK Releases page.  

https://github.com/intelrealsense/librealsense/releases

So you could try that.  However, problems with updating multiple cameras have also occasionally been reported with that tool too.  

[image]

One of the set of multiple cameras may go into recovery mode if it experiences an interruption during the update.  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/7095#issuecomment-879995800 found that once it is in recovery mode then may take several tries of the -r recovery command before recovery works.

diplomatist, 2024-11-08:
Hi @MartyG-RealSense ！
I used viewer or rs-fw-update on Linux to upgrade the firmware of a single D455, and almost 100% of the attempts failed. Later, I changed the connection line for testing and found that the success rate of upgrading firmware was much higher;

An additional situation needs to be explained:
These three D455s were originally connected to three RK3588s development boards; 
Due to construction issues, RK3588s was connected to a 220V power supply, which caused the power supply of RK3588s to burn out; 
I connected these three D455s to the Linux viewer or Windows viewer with the correct firmware and they all worked properly;
But it is not possible to upgrade using these three cables in the Linux viewer or RS-FW-update program, and it can be upgraded through the Windows viewer (not yet upgraded in FW.Uupdate.exe of Windows);

Due to objective reasons, I have many D455 devices that are unable to easily upgrade firmware on Windows;

MartyG-RealSense, 2024-11-08:
By 'connection line', do you mean the USB cable please?  If you do, are you using the official 1 meter USB cables supplied with the cameras for the firmware update or a USB cable of your own choice?

If you are using your own cable and that cable is lengthy then it is recommendable to change to the official short 1 m cable when performing a firmware update and then change back to the other cable, though I can appreciate that this may not be practical to do in some situations, especially where you have many cameras that may also be difficult to access in person.

You could test whether the rate of success increases with having the micro-sized end of the cable inserted into the base of the D455 in one of two particular directions.  The micro-sized end of the cable is two-way insertion, and sometimes a camera will perform better when the micro-connector is inserted in a certain direction.

closed 2024-11-09 by diplomatist