realsenseai/librealsense #13496: librealsense-2.56.1 (source build) relocation R_X86_64_32S against symbol `malloc@@GLIBC_2.2.5' can not be used when making a PIE object error

HariKrishnan06082k, 2024-11-06:
### Issue Description
Building error when trying to install librealsense SDK from source code when building with graphical examples enabled. I'm able to build successfully with RSUSB backend method but I would like to ideally use realsense-viewer for my ROS2 wrapper. The error occurs at a point when it has to generate executable for the realsense-viewer. I have had success building realsense-viewer before from other versions of SDK. The other issue I have is when using D455 with ROS2 wrapper version 4.56.1. For my work I would require the infra stream to be enabled at 60Hz for both IR1 and IR2 along with metadata when I do that I get no frames received error. But that I would create another issue ig for now I just need the rs-viewer to work because that would give me and idea about what resolutions and what streams can be enable simultaneously. Cuz recently the default profile resolution is 848x480 or something .

Camera Model - Realsense D455 camera
Firmware Version - 5.16.0.1
OS - Linux 22.04 LTS
Platform - PC AMD CPU (cuda 11.8)
Kernel Version - 6.8.0-48-generic
Librealsense version - 2.56.1 (Build from source code)
ldd --version = 2.35
cmake --version = 3.22.1
Kernel Patch Done - Yes

HariKrishnan06082k, 2024-11-06:
Device info: 
    Name                          : 	Intel RealSense D455
    Serial Number                 : 	246322300557
    Firmware Version              : 	5.16.0.1
    Recommended Firmware Version  : 	5.16.0.1
    Physical Port                 : 	/sys/devices/pci0000:80/0000:80:01.3/0000:82:00.0/0000:83:0c.0/0000:84:00.0/usb4/4-2/4-2.1/4-2.1:1.0/video4linux/video24
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B5C
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	322743060552
    Firmware Update Id            : 	322743060552

MartyG-RealSense, 2024-11-07:
Hi @HariKrishnan06082k  Rather than building realsense-viewer just for checking resolutions, it may be easier if I provide D455-relevant information below.

You can enable depth, RGB and both infrared streams simultaneously.  Each stream enabled will add a further processing burden to the computer's CPU though.

Whilst RealSense should be able to work with any Intel or ARM processor, there are no guarantees made about AMD.  Past support cases have shown that RealSense cameras should be able to work with AMD Ryzen CPUs but non-Ryzen AMD processors may have a lesser chance of working correctly.

The default depth and infrared resolution is indeed 848x480.  The default RGB resolution is 1280x720.

Whilst depth and infrared resolution can be set to 1280x720, it is limited to 30 FPS rather than 60 and so would not be suitable for your 60 FPS / Hz requirement.  60 FPS for depth and infrared is supported on 848x480.

Infrared 2 (the right-side infrared) will be unavailable on USB 2.1 connections and requires full-speed USB 3.

Building librealsense with the RSUSB backend method automatically enables metadata support.  I note though that you applied a kernel patch.  You do not need to patch the kernel on an RSUSB build and I would recommend against doing so because of the risk of breaking the RSUSB = true librealsense build.

MartyG-RealSense, 2024-11-08:
Multiple questions are no problem at all.  :)

1.  It is not necessary to enable the depth topic in order to publish the infra1 and infra2 topics.    

The infra topics can be enabled in the launch instruction with `enable_infra1:=true enable_infra2:=true` or by setting infra1_enabled and infra2_enabled to true in the launch file.

https://github.com/IntelRealSense/realsense-ros/issues/2760#issuecomment-1569750779 may be a helpful reference for your interest in emitter_on_off and metadata.

2.  At https://github.com/IntelRealSense/realsense-ros/issues/2561#issuecomment-1325743680 one of my Intel RealSense colleagues suggests putting " " quotes around the .yaml path in the config_file instruction.

3.  Changing to a different kernel version on your computer should clear the kernel patching.  So you could perhaps update to a slightly newer kernel and then reinstall the previous kernel version.

4.  High quality, lengthy USB cables that are RealSense-validated are supplied by a company called Newnex.  They also offer USB cables with 'active repeater' signal booster components for 10 meters + use and fiber-optic cabling solutions that will work with RealSense cameras up to 100 meters.

https://newnex.com/realsense-3d-camera-connectivity.php

https://www.youtube.com/watch?v=GLQgR1jT04M

HariKrishnan06082k, 2024-11-08:
Thank you, I managed to solve the Composable node problem by modifying the launch file and using a custom .yaml file to load it into a runtime container. As for the librealsense 2.56.1 the working solution for me is to use RSUSB method to make it work with current latest kernel of Ubuntu Kernel Version - 6.8.0-48-generic. I think a lot of hassle around this kernel issue would be solved with binary release of librealsense but ig I'll have to wait for the official release. Is there any way to build rs-viewer separately because It would be helpful for my application. For now , in Ubuntu system it looks like you build it only along with the entire source code with DBUILD_GRAPHICAL_EXAMPLES set to true in cmake flags. Or if there's a way to fix the error caused for the specific .o file which builds the graphical tools in source code that also would help. Pretty sure my computer supports X11 forwarding and OpenGL as I've had success before building rs-viewer with other versions of librealsense. 

I saw another relevant issue for arm based processor here : [https://github.com/IntelRealSense/librealsense/issues/11807](url)
but the thread was closed with making the DBUILD_GRAPHICAL_EXAMPLES to false. I would ideally want to solve this issue so I don't compromise on losing out graphical tools when your processor actually supports it. Even though this error is probably specific to cmake I would like to keep the thread open to get a solution for making changes to respective CmakeFiles or flags that would enable the successful build of graphical tools. Thanks so much for your response. 

Attaching a sample nodefactory_plugin ros launch file which worked seamlessly for testing to make it easy for others to use.

```
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

# Make sure required packages can be found
rs_node_class=  'RealSenseNodeFactory'

realsense_node_params = [{'name': 'serial_no',              'default': "''", 'description': 'choose device by serial number'},
                         {'name': 'usb_port_id',            'default': "''", 'description': 'choose device by usb port id'},
                         {'name': 'device_type',            'default': "''", 'description': 'choose device by type'},
                         {'name': 'log_level',              'default': 'info', 'description': 'debug log level [DEBUG|INFO|WARN|ERROR|FATAL]'},
                         {'name': 'rgb_camera.profile',     'default': '0,0,0', 'description': 'color image width'},
                         {'name': 'rgb_camera.color_profile',     'default': '640,480,15', 'description': 'color stream profile'}, #modified
                         {'name': 'depth_module.depth_profile',   'default': '640,480,60', 'description': 'depth stream profile'}, #modified
                         {'name': 'depth_module.depth_format',    'default': 'Z16', 'description': 'depth stream format'},#modified
                         {'name': 'depth_module.emitter_enabled',  'default': '1', 'description': 'enable/disable emitter'}, #modified
                         {'name': 'depth_module.emitter_on_off',  'default': 'true', 'description': 'enable/disable emitter'}, #modified
                         {'name': 'depth_module.infra_profile',   'default': '640,480,60', 'description': 'infra streams (0/1/2) profile'}, #modified
                         {'name': 'enable_color',           'default': 'true', 'description': 'enable color stream'},
                         {'name': 'enable_depth',           'default': 'true', 'description': 'enable depth stream'},
                         {'name': 'enable_infra',           'default': 'false', 'description': 'enable infra stream'},
                         {'name': 'enable_infra1',          'default': 'true', 'description': 'enable infra1 stream'},
                         {'name': 'enable_infra2',          'default': 'true', 'description': 'enable infra2 stream'},
                         {'name': 'enable_gyro',            'default': 'false', 'description': "enable gyro stream"},
                         {'name': 'enable_accel',           'default': 'false', 'description': "enable accel stream"},
                         {'name': 'unite_imu_method',       'default': "0", 'description': '[0-None, 1-copy, 2-linear_interpolation]'},
                         {'name': 'intra_process_comms',    'default': 'true', 'description': "enable intra-process communication"},
                         {'name': 'enable_sync',            'default': 'false', 'description': "'enable sync mode'"},
                         {'name': 'pointcloud.enable',      'default': 'false', 'description': ''},
                         {'name': 'enable_rgbd',            'default': 'false', 'description': "'enable rgbd topic'"},
                         {'name': 'align_depth.enable',     'default': 'false', 'description': "'enable align depth filter'"},
                         {'name': 'publish_tf',             'default': 'true', 'description': '[bool] enable/disable publishing static & dynamic TF'},
                         {'name': 'tf_publish_rate',        'default': '1.0', 'description': '[double] rate in HZ for publishing dynamic TF'},
                        ]

def declare_configurable_parameters(parameters):
[... cut ...]
```
One warning which I couldn't figure out how to fix is this , 
**" Failed to set value to sensor: depth_module.emitter_on_off = 0[0, 1]
[component_container-1] hwmon command 0x7b( 17 0 0 0 ) failed (response -9= No expected user action)"** 

coming from the configurable ros2 param depth_module.emitter_on_off true.

Also if you could provide a quick timeline update for upcoming dkms binary release for latest kernel of Noble and Jammy (Kernel Version - 6.8.0-48-generic) that would be appreciated.

MartyG-RealSense, 2024-11-09:
I'm pleased to hear that you have made significant progress.

The source code of the realsense-viewer tool is located at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer

I do not have information about a time estimate for release of a librealsense version that has dkms packages.  Usually, there is a gap of around 6 months between major releases, and 2.55.1 was released in April 2024.  So if the roughly 6 month gap is maintained then it might be November or December when the next 'final' release equipped with packages is released, though that is only speculation and not official advice.

Regarding the DBUILD_GRAPHICAL_EXAMPLES=true parameter not working on your computer: as you have an AMD CPU rather than an Intel or Arm one, there are not guarantees that I can provide about how well librealsense will work with an AMD chip even if changes were made to CMake, unfortunately.

It is possible to build the SDK examples individually with CMake if a specific example's C++ script has its code copied into a new project and then you create a special CMakeLists.txt file that contains the necessary linkages to librealsense.  The link below has a project that demonstrates this.  The example's .cpp C++ script file will be unchanged from the original when copied into the new project and only the CMakeLists.txt file of the project needs to be changed to enable it to run on its own as a standalone application.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

Regarding setting emitter_on_off in ROS, does entering the command below into the ROS terminal work for you?

`ros2 param set /camera/camera depth_module.emitter_on_off true`

HariKrishnan06082k, 2024-11-12:
Hi, yes the command ros2 param set seems to work on command line after the node is launched. But was wondering if there's a way to launch it through launch file or while starting a ros driver node class. As for the rs-viewer tool if you think AMD might be the issue then don't worry about it but I find it weird that it is able to build inside a docker container without any error , must be using the same kernel modules of my host system in that case.

MartyG-RealSense, 2024-11-12:
You could try adding it to the launch file as a **rosparam** code block.  Something like this:

```
<rosparam> 
/camera/camera/stereo_module/emitter_on_off: true
</rosparam>
```

HariKrishnan06082k, 2024-11-12:
Hi, thanks for replying patiently to all the questions. Feel free to close the issue if you are sure it might be an AMD issue for building graphical tools. Have a nice rest of your day. I will figure out the ros2 param issue :)

MartyG-RealSense, 2024-11-13:
There is no way to confirm that the AMD CPU is related to the graphical tools issue.  I can only speculate about the possibility of problems occurring based on AMD CPUs not being officially validated for RealSense cameras.

HariKrishnan06082k, 2024-11-14:
Hi sorry to open this issue again but it looks like the emitter_on and emitter_on_off is not taking effect in config file also as it pretty much does the same as rs_launch.py there is no concept of rosparam tag in ros2 I think. My intention is to get infrared images without projector on (basically dots) . To do this the only thing I figured out which worked is to launch the IR streams then once the realsense_node is up then do runtime ros2 param set /camera/camera depth_module.emitter_enabled 1 followed by ros2 param set /camera/camera depth_module.emitter_on_off true . Now i get clean IR images without projector (to emulate stereo pair).

Is there a way to make it work in Launch Description as in rs_launch.launch.py

MartyG-RealSense, 2024-11-15:
Another way to turn the projector off is to set the `laser_power` parameter to a value of '0'.

MartyG-RealSense, 2024-11-23:
Hi @HariKrishnan06082k  Do you require further assistance with this case, please?  Thanks!

HariKrishnan06082k, 2024-11-25:
Hi thanks for the help I found the param to be working only during run time as I wasn't able to make it work at launch config level, I timed the launch so that my realsense node is up and running first and then I set this separately as rosparam. If it can be provided as part of LaunchConfiguration that would be ideal for future use but for now everything is working so you can close the issue thanks for your help.

closed 2024-11-26 by MartyG-RealSense