realsenseai/librealsense #13500: Ubuntu 24.04 Unable to locate package librealsense2-utils

gmmyung, 2024-11-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu 24.04  | 
| Kernel Version (Linux Only)    | 6.8.0-48-generic                                      | 
| Platform                                 | PC (x86 intel) |
| SDK Version                            |          | 
| Language                            |            | 
| Segment			|        | 

### Issue Description
I have followed the instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) to install the Realsense sdk. I was able to install `librealsense2-dkms`, but after that, when running `sudo apt-get install librealsense2-utils`, it says: `E: Unable to locate package librealsense2-utils`

MartyG-RealSense, 2024-11-08:
Hi @gmmyung   There is not currently support for installing DKMS packages for kernel 6.8.  The current newest supported kernel for package installation at the time of writing this is 6.5 in librealsense 2.55.1 (SDK beta versions 2.56.1 and 2.56.2 do not have DKMS packages).

[image]

SDK beta version 2.56.2 supports manual patching of a source code build of the SDK for kernel 6.8.  Alternatively, you could try building the SDK from source code with the **RSUSB backend** installation method, which bypasses the kernel and so is not dependent on a particular Linux version or kernel version.

gmmyung, 2024-11-08:
Ok, I guess I should go reinstall Ubuntu 22.04 :(
Is there any timeframe/PR I can keep track of until this is resolved?

MartyG-RealSense, 2024-11-08:
I recommend keeping an eye on the SDK 'Releases' page for the release of a non-beta version that has DKMS packages provided for it.  I do not currently have information regarding a time estimate for when that will occur.

https://github.com/intelrealsense/librealsense/releases

closed 2024-11-15 by gmmyung

martinakos, 2025-03-09:
Just following the linux instructions, I successfully install librealsense2-dkms in ubuntu 24.04 with kernel 6.8.0-1020-oem. However, it can't locate the librealsense2-utils package. Is that coming soon, if the dkms is available already?