realsenseai/librealsense #13503: Error

users-9706, 2024-11-09:
| | |
|---|---|
|**librealsense**|2.56.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```

MartyG-RealSense, 2024-11-09:
Hi @users-9706  In the past a solution for this error has been to download and run the very old **2.21.0** version of the RealSense Viewer, and then close it and launch the latest Viewer version again.

You can obtain the 2.21.0 Viewer by visiting the link below and clicking on the **Intel.RealSense.Viewer.exe** link in the 'Assets' file list at the bottom of the page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.21.0

When launching the 2.21.0 Viewer, if you are prompted to click on a button to update then it is very important that you **do not** click the update button, as it could install an old firmware update that makes the camera unusable.  It should be enough simply to open the 2.21.0 Viewer and then immediately close it again and launch the 2.56.2 Viewer.

[image]

users-9706, 2024-11-10:
After uninstalling the software, the realsense-config.json file in the C: /Users/XXX/AppData/Roaming folder is deleted, and then reinstalled,  then it works.

closed 2024-11-23 by MartyG-RealSense