realsenseai/librealsense #13506: 403 forbiident

shatarajt, 2024-11-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 } | 
| Firmware Version                   | (Open RealSense Viewer (newest version) | 
| Operating System & Version |  Linux (Ubuntu 22.04)  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |  {  2}                          | 
| Language                            |  { }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
im currently running mobaxterm with wsl2 ubuntu 22.04
jamesshatara@DESKTOP-VNKT0G1:~/librealsense/build$ echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | \
sudo tee /etc/apt/sources.list.d/librealsense.list
sudo apt-get update
deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo jammy main
Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease
Hit:3 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:4 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:5 http://packages.ros.org/ros2/ubuntu jammy InRelease
Hit:6 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:7 https://librealsense.intel.com/Debian/apt-repo jammy InRelease
Err:1 https://librealsense.intel.com/Debian/ubuntu jammy InRelease
  403  Forbidden [IP: 18.238.132.115 443]
Reading package lists... Done
E: Failed to fetch http://librealsense.intel.com/Debian/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 18.238.132.115 443]
E: The repository 'http://librealsense.intel.com/Debian/ubuntu jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
i cant make any progress any help would be greatly appreciated

MartyG-RealSense, 2024-11-12:
Hi @shatarajt  Building librealsense from dkms packages in WSL2 tends to have problems, as discussed at https://github.com/IntelRealSense/librealsense/issues/12810 which highlights that RealSense users have been able to successfully build from source code instead of packages on WSL2.

shatarajt, 2024-11-12:
jamesshatara@DESKTOP-VNKT0G1:~$ ./libuvc_installation.sh
+ exec
+ exec
+ '[' 0 -ne 0 ']'
+ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy
++ uname -r
+ echo 'Kernel version 5.15.153.1-microsoft-standard-WSL2'
Kernel version 5.15.153.1-microsoft-standard-WSL2
+ sudo apt-get update
[sudo] password for jamesshatara:
Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease
Hit:3 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:4 http://packages.ros.org/ros2/ubuntu jammy InRelease
Hit:5 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:7 https://librealsense.intel.com/Debian/apt-repo jammy InRelease
Err:6 https://librealsense.intel.com/Debian/ubuntu jammy InRelease
  403  Forbidden [IP: 18.238.132.9 443]
Reading package lists... Done
E: Failed to fetch http://librealsense.intel.com/Debian/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 18.238.132.9 443]
E: The repository 'http://librealsense.intel.com/Debian/ubuntu jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
 so when i get to the last line of script i get this error... it also seems my pc cant detect the camera and i know the camera works. any recommendations?

MartyG-RealSense, 2024-11-13:
Is using Linux a requirement for your project, please?  The SDK can easily be installed on a Windows computer without using WSL2 by downloading and running an installer program file called **Intel.RealSense.SDK-WIN10** in the "Assets" file list at the bottom of the page linked to below.  It works with both Windows 10 and Windows 11.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

[image]

closed 2024-12-01 by MartyG-RealSense