realsenseai/librealsense #13509: Detailed 3D Scanning using the D405

tedtheturtlenz, 2024-11-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | Any | 
| Operating System & Version |   Windows 11  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  Any                          | 
| Language                            |  C/Python                          | 
| Segment			|  3D Scanning                        | 

### Issue Description
Hey there,
I just had a question with regards to the possible detail level of the D405.
I am looking at using a pair of D405 stereo cameras to 3D scan Bonsai trees (small house plants) with the goal of being able to extract information about the Bonsai trees from the acquired point cloud / processed mesh (e.g. trunk diameter, branch diameter, leaf coverage etc). Bonsai trees can have very small branches, down to just over a mm in diameter. Would it be feasible to get an accurate model of a Bonsai tree even down to its small branches using the D405? I am currently planning on taking several 3D scans of the plant and then combining the point clouds in software (potentially CloudCompare) as to get a full 3D scan of the plant.
So essentially my question boils down to this:
Given the D405 is being used to scan an object within 30cm of the camera, would it be possible to recover sufficient detail to reproduce a 3D model down to around 1mm diameter branches?
Apologies if this is not the right place to be asking this question, I just saw similar questions about other RealSense cameras, so I thought to ask here.
Thanks for your time,
Tim

MartyG-RealSense, 2024-11-13:
Hi @tedtheturtlenz  The camera will have difficulty with detecting depth for very thin objects (thinner than a toothpick).  

I used a D405 to depth-scan a plant that had some leaves with 1 mm thick stems (highlighted in white boxes on the image below).  The depth image was not able to render those stems.

[image]

tedtheturtlenz, 2024-11-17:
> Hi @tedtheturtlenz The camera will have difficulty with detecting depth for very thin objects (thinner than a toothpick).
>
> I used a D405 to depth-scan a plant that had some leaves with 1 mm thick [...]

Hey Marty,
Thanks so much for getting back to me, I really appreciate it. And thanks for the example with the plant, that helps clarify things a lot.
Thanks so much for your time,
Cheers

closed 2024-11-17 by tedtheturtlenz

MartyG-RealSense, 2024-11-17:
You are very welcome.  I'm pleased that I could help.  :)