realsenseai/librealsense #13513: Realsense viewer and librealsene failing with "failed to convert special folder: errno=42"

warmist, 2024-11-14:
Source of error is this line:
https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/third-party/rsutils/src/special-folder.cpp#L81

wcstombs_s returns value 42 which is EILSEQ i.e. it fails to convert wide string to multibyte string.

There are numerous issues closed with this but nobody dug down to the source. It seems that this only happens if your username has unicode characters.

MartyG-RealSense, 2024-11-14:
Hi @warmist  Have you visited the advice about the **errno=42** error at https://github.com/IntelRealSense/librealsense/issues/13037 please?

https://github.com/IntelRealSense/librealsense/issues/13457#issuecomment-2432900837 has instructions for finding on a Windows computer and deleting the **realsense-config.json** configuration file mentioned in that link.

warmist, 2024-11-14:
Hi @MartyG-RealSense. Thanks for answering.

Yes, i've seen both of them. Also i've looked into all the duplicates.

#13037 does not apply because "problem due to an uninitialized json variable" cannot happen as it's before file loading. It's an error when resolving (and transforming) the path to appdata folder. There is no "realsense-config.json" file anywhere in appdata (also related: apps shouldn't write to appdata and should create a folder).

https://github.com/IntelRealSense/librealsense/issues/13457#issuecomment-2432900837 yes, i've looked into hidden files as well. Also you can access the folder faster if you type "%appdata%" into explorer path (i.e. the part where it has "C:\etc...")

I'll try to setup a minimal repro as i feel this issue might be hard to reproduce without setting up a windows machine with username with unicode chars in it.

warmist, 2024-11-18:
Here is minimal reproduction: https://gist.github.com/warmist/e0b1927387864566db2804c50d0cf512

In my environment `wcstombs_s` fails with `error=42`. Content of path:
[image]
(see unicode letter in path)

warmist, 2024-11-18:
Further research shows that `wcstombs_s` uses current locale and current locale is "en-us" and fails to convert unicode character. It should use `_wcstombs_s_l` with new locale (i.e. `_create_locale(LC_ALL,"");`)

MartyG-RealSense, 2024-11-18:
I had a discussion with my Intel RealSense colleagues about the 'failed to convert special folder: errno=42' error earlier in 2024, so I will follow up on it and pass your kindly provided information to them for their consideration.

MartyG-RealSense, 2024-12-01:
Hi @warmist  Do you require further assistance with this case, please?  Thanks!

warmist, 2024-12-01:
> Hi @warmist Do you require further assistance with this case, please? Thanks!

Thank you for asking. I don't need assistance, however i wonder if (and when) this will be fixed?

MartyG-RealSense, 2024-12-01:
I have sent a follow-up message to my Intel RealSense colleagues about this issue.

MartyG-RealSense, 2024-12-04:
My Intel RealSense colleagues have provided some further advice about the errno=42 error.

Given that the problem seems to be related to certain charcters being used in a Windows user account name, I asked whether creating a secondary Windows user account with a name that has no special characters might help.  This was their response. 

***********

You can try to run it as Administrator instead of switching the account.  On Windows 11, it also has a feature to run the app as a different user.  That option can be enabled with the instructions at the link below:
 
https://winaero.com/how-to-run-apps-as-different-user-in-windows-11/

warmist, 2024-12-04:
@MartyG-RealSense Thank you for the reply.

Sadly we cannot do that as we are using realsense as a lib and we cannot ask users of your software to create secondary accounts or run as administrator. Also running programs as adminstrator on windows does not change the appdata folder as it still uses same `c:\users\<username>\AppData\Roaming` folder. You can test that by running a commandline as admin and typing `echo %appdata%`

Nir-Az, 2024-12-04:
Hi, 
We will add this issue to our backlog,
If you find a quick solution and needs it ASAP, feel free to contribute with a PR :)
I will update once we get to it

warmist, 2024-12-05:
@Nir-Az Thanks. I will do a PR if we fix it for ourselves before you get to it.

Nir-Az, 2024-12-12:
@warmist Fixed on PR #13597 by @AviaAv 
You are welcome to try it out and update :)

warmist, 2025-01-06:
> Hi @warmist Do you require further assistance with this case, please? Thanks!

Ah thanks for asking, i've moved on from that job position. Couldn't test if the fix helps sorry.

MartyG-RealSense, 2025-01-06:
It's no trouble at all.  Thanks very much for the update.  As you have moved jobs, I will close this issue.  Thanks again!

closed 2025-01-06 by MartyG-RealSense